espressif/esp-idf #496: Interrupt wdt timeout on CPU0

jaracil, 2017-04-06:
When I compile a very trivial project with "FREERTOS_UNICORE=y", the esp32 console yields this message:
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

This is the code that fails when setting "FREERTOS_UNICORE=y" 

```c
void test_task(void *pvParameter) {
    while (true) {
        printf("Hola Mundo\r\n");
        vTaskDelay(300 / portTICK_PERIOD_MS);
    }
}

void app_main(void)
{
    nvs_flash_init();
    xTaskCreate(&test_task, "test_task", 4096, NULL, 5, NULL);
}
```

igrr, 2017-04-07:
This is a regression introduced in 45581db when we have started using software interrupt to trigger a yield from a critical section. Unfortunately initialization of the said interrupt is only done for the case of !FREERTOS_UNICORE.

This patch should help:

```c++
diff --git a/components/esp32/cpu_start.c b/components/esp32/cpu_start.c
index 495a178..7bc487b 100644
--- a/components/esp32/cpu_start.c
+++ b/components/esp32/cpu_start.c
@@ -206,9 +206,7 @@ void start_cpu0_default(void)
 #if CONFIG_TASK_WDT
     esp_task_wdt_init();
 #endif
-#if !CONFIG_FREERTOS_UNICORE
     esp_crosscore_int_init();
-#endif
     esp_ipc_init();
     spi_flash_init();
     /* init default OS-aware flash access critical section */
```
We'll do more tests here and fix this in master soon.

phieber, 2017-04-08:
@igrr Yes, this patch helped also in my case...

closed 2017-04-10 by igrr