espressif/esp-idf #499: STA disconnect caused ESP32 crashed when BLE and WIFI are coexistence

wangqi0120850, 2017-04-07:
[0;32mI (1011731) ANTS_BLE: adv_data_process [50:9b:0d:a1:0d:5e][0m
I (1017681) wifi: bcn_timout,ap_probe_send_start
I (1020181) wifi: ap_probe_send over, resett wifi status to disassoc
I (1020181) wifi: state: run -> init (1)
I (1020181) wifi: pm stop, total sleep time: 0/1007317484

I (1020181) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:1
[0;32mI (1020191) ANTS_WIFI: ... done reading from socket. Last read return=0 errno=128

[0m
[0;32mI (1020191) ANTS_WIFI: SYSTEM_EVENT_STA_DISCONNECTED
[0m
[0;32mI (1020201) ANTS_WIFI: 3... [0m
[0;32mI (1021211) ANTS_WIFI: 2... [0m
[0;32mI (1022211) ANTS_WIFI: 1... [0m
[0;32mI (1022311) ANTS_WIFI: SYSTEM_EVENT_STA_DISCONNECTED
[0m
[0;32mI (1023211) ANTS_WIFI: 0... [0m
[0;32mI (1024211) ANTS_WIFI: Starting again![0m
[0;32mI (1024421) ANTS_WIFI: SYSTEM_EVENT_STA_DISCONNECTED
[0m
[0;31mE (1024601) BT: btc_search_callback  BLE observe complete. Num Resp 10
[0m
[0;32mI (1026521) ANTS_WIFI: SYSTEM_EVENT_STA_DISCONNECTED
[0m
[0;32mI (1028631) ANTS_WIFI: SYSTEM_EVENT_STA_DISCONNECTED
[0m
[0;32mI (1028631) ANTS_WIFI: wifi deinit[0m
I (1029631) wifi: Deinit lldesc rx mblock:10
I (1029631) wifi: Deinit lldesc rx ampdu len mblock:7
I (1029631) wifi: Deinit lldesc rx ampdu entry mblock:4
I (1029631) wifi: eb_mgmt_s_free_list i = 0

I (1029631) wifi: eb_tx_bar_free_list i = 0

I (1029641) wifi: eb_rx_block_free_list i = 0

[0;32mI (1029641) ANTS_WIFI: wifi init[0m
[0;32mI (1029641) ANTS_WIFI: Enter initialise_wifi.[0m
[0;32mI (1029651) ANTS_WIFI: After WIFI_INIT_CONFIG_DEFAULT.[0m
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400d8919  PS      : 0x00060e30  A0      : 0x800d1d9d  A1      : 0x3ffd76b0  
A2      : 0x3ffd77b8  A3      : 0x00000000  A4      : 0x0000000a  A5      : 0x3ffd0a34  
A6      : 0x3ffd317c  A7      : 0x3ffd3144  A8      : 0x800d88ee  A9      : 0x3ffd7690  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffd959c  
A14     : 0x3ffd9578  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

This is my log info for this crash. ESP32 was working BLE and WIFI coexistence, after long time(about several hours), STA has disconnected from AP and during reconnecting, BLE was didn't work. Then I tried to deinit the wifi module and reinit it. however, it caused crash. Sometime it will show the log:
BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x200c

I think the key point is that STA reonncet make the BLE Stack blocked and caused crash. Someone can give me any suggestion?

Thanks.

TimXia, 2017-04-12:
Please call esp_wifi_stop() before esp_wifi_deinit(). Actually, you do not need to call esp_wifi_deinit() and then call esp_wifi_init() again, just call esp_wifi_stop() and esp_wifi_start().

TimXia, 2017-04-12:
We will add judgement when esp_wifi_deinit() is called.

wangqi0120850, 2017-04-13:
I (2153481) wifi: bcn_timout,ap_probe_send_start
I (2155981) wifi: ap_probe_send over, resett wifi status to disassoc
I (2155981) wifi: state: run -> init (1)
I (2155981) wifi: pm stop, total sleep time: 0/2136016994

I (2155981) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:1

Guru Meditation Error of type StoreProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x401483d2  PS      : 0x00060730  A0      : 0x801483f6  A1      : 0x3fff0b80  
A2      : 0x000002a4  A3      : 0x3ffff174  A4      : 0x3ffeaf84  A5      : 0x3ffd4670  
A6      : 0x00000014  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x3fff0b60  
A10     : 0x00000000  A11     : 0x000002b0  A12     : 0x40081c24  A13     : 0x00000000  
A14     : 0x3ffed274  A15     : 0x3ffed23c  SAR     : 0x00000017  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000005  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x401483d2:0x3fff0b80 0x401483f6:0x3fff0ba0 0x40135008:0x3fff0bc0 0x401170eb:0x3fff0be0 0x40115d34:0x3fff0c00 0x40115d74:0x3fff0c20 0x40116d96:0x3fff0c40 0x40127769:0x3fff0c60 0x401277e7:0x3fff0c90 0x4016af8a:0x3fff0cb0 0x401279ff:0x3fff0cd0

**BT: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x200c**

I have used esp_wifi_stop()-->esp_wifi_start() instead of esp_wifi_deinit()->esp_wifi_init() after ESP32 disconnected from AP. The crash was still exist.
I got this log when esp32 disconnected from AP, BLE and WIFI both didn't work. Looks like the RF has stopped.  I can't find the 0x200c in IDF source code. Maybe the crash wasn't caused by the esp_wifi_deinit(). I wonder why the RF didn't work?

Thanks.

TimXia, 2017-04-13:
Did you enable software controls WiFi/BT coexistence? It is in manuconfig --> WiFi --> Software controls WiFi/Bluetooth coexistence.

TimXia, 2017-04-13:
@TianHao-Espressif Please help to take a look at this problem.

TianHao-Espressif, 2017-04-13:
This crash is due to freertos timer task stack size is not enough. The release/v2.0 fix this bug or you can modify `configTIMER_TASK_STACK_DEPTH` manually.

wangqi0120850, 2017-04-13:
@TimXia  I didn't enable the 'software controls WiFi/BT coexistence' in menuconfig. The BLE and WIFI both work fine before crashed.

@TianHao-Espressif I have disable the Task Watchdog in menuconfig. Did it would make any effect?
I will modify the TIMER_TASK_STACK_DEPTH and try it again. 

Thanks for your attention.

wangqi0120850, 2017-04-13:
@TianHao-Espressif    I have modify the configTIMER_TASK_STACK_DEPTH to 4096, the default value was 2048. However the crash was still exist.

TianHao-Espressif, 2017-04-19:
You use unique core or two? I mean use only 1 cpu, or use 2 cpus.

wangqi0120850, 2017-05-18:
use 2 cpus

TimXia, 2017-06-12:
@wangqi0120850 Does this problem still exist? If so, could you please provide the disassembly file or source code to us to find the cause?

gicek, 2017-06-30:
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: ADC_Check_433Re
CPU 1: IDLE
I (7568811) wifi: bcn_timout,ap_probe_send_start
I (7571311) wifi: ap_probe_send over, resett wifi status to disassoc
I (7571311) wifi: state: run -> init (1)
I (7571311) wifi: pm stop, total sleep time: 0/-1029795978

I (7571311) wifi: n:1 0, o:1 1, ap:255 255, sta:1 1, prof:1
[0;32mI (7571321) PLATFORM: SYSTEM_EVENT_STA_DISCONNECTED
[0m
I (7571441) wifi: n:1 1, o:1 0, ap:255 255, sta:1 1, prof:1
I (7571441) wifi: state: init -> auth (b0)
I (7571451) wifi: state: auth -> assoc (0)
I (7571451) wifi: state: assoc -> run (10)
I (7581451) wifi: pm start, type:0

I (7581451) wifi: state: run -> init (2)
I (7581451) wifi: pm stop, total sleep time: 0/651

I (7581451) wifi: n:1 0, o:1 1, ap:255 255, sta:1 1, prof:1
[0;32mI (7581461) PLATFORM: SYSTEM_EVENT_STA_DISCONNECTED
[0m
I (7581591) wifi: n:1 1, o:1 0, ap:255 255, sta:1 1, prof:1
I (7581591) wifi: state: init -> auth (b0)
I (7581591) wifi: state: auth -> assoc (0)
W (7581591) wifi: alloc eb len=208 type=4 fail

I also met the same problem，it‘s normally after power on

kzyapkov, 2017-07-04:
Possibly related to #675

jstar88, 2017-07-10:
Same problem: both BLE and Wifi alone work fine, however when using them at same time i got this errors:
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 7 - NOT_ASSOCED
after a while(15minuts), and the IP is 0.0.0.0.

Im using Arduino with PubSubClient lib and xTaskCreatePinnedToCore to configure the passive BLE scan on core 0(otherwise do not work). 
Im also using xQueue to share data, anyway i got this error from first time without these things.
I tried to
- increase wifi task priority
- increase configTIMER_TASK_STACK_DEPTH to 4096
- remove interrupt on UART frame error
- switching MQTT publish from core 0 to 1(maybe is it blocking wifi?)

Sermus, 2017-07-24:
I have similar symptoms. With passive ble scanning configured like this 
`	static esp_ble_scan_params_t ble_scan_params = {
		.scan_type              = BLE_SCAN_TYPE_PASSIVE,
		.own_addr_type          = BLE_ADDR_TYPE_PUBLIC,
		.scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
		.scan_interval          = 0x10,
		.scan_window            = 0x10
	};`
I have STA interface disconnected from AP. But i got all the BT adverisements.

If i configure BLE scanning like this:
`	static esp_ble_scan_params_t ble_scan_params = {
		.scan_type              = BLE_SCAN_TYPE_PASSIVE,
		.own_addr_type          = BLE_ADDR_TYPE_PUBLIC,
		.scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
		.scan_interval          = 0x100,
		.scan_window            = 0x30
	};
`
I have STA connected to the AP but it misses almost all the advertisements. Beside this i usually have a trace like:
E (64006) BT: btc_search_callback  BLE observe complete. Num Resp 0

SW coexistance is off in my case.

FayeY, 2017-11-21:
Since this is a very old issue, I'm going to close it. Please feel free to reopen it, if this problem is still unsolved on the latest IDF.

closed 2017-11-21 by FayeY

GianlucaLoi, 2020-09-15:
Hello everyone,

I'm having the same error message "**BT_HCI: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x200c**". I'm using the ESP-IDF v4.3-dev-907-g6c17e3a64.
My application works as follows:

1. Connect to an AP
2. Periodically run ble scan (scan period = 5seconds) and send a MQTT message

After some time, the ble stack stucks with this error, but the device remain active. 

Did you find any solution about this problem?

Gianluca.

TimXia, 2020-12-15:
@blueMoodBHD Please help.

GianlucaLoi, 2020-12-15:
> @blueMoodBHD Please help.

Hello @TimXia 

I "solved" this problem, I think. You can find a possible solution in this other github issue: 
https://github.com/espressif/esp-idf/issues/4001

@blueMoodBHD could you validate this solution?

Regards,

Gianluca

blueMoods, 2020-12-15:
@GianlucaLoi  We have solved this bug in commit https://github.com/espressif/esp-idf/commit/e5af56db411daf9bdabbf4d8dc020e2fc30f84d7
Could you please test with it?