espressif/esp-idf #501: ESP32 Possible bug: ADC output values incorrect (12bit mode)

stevenvdr, 2017-04-07:
I have the following setup: the on chip DAC generates a sine wave (between 0.1V and 0.9V) on GPIO26 using the I2S DMA. This sine wave is send through a buffer to make sure the input and output impedances are within range for the chip. Then the output of the buffer is fed back to pin 33, which has the ADC connected to it. This setup is illustrated in the figure.

[image]

I have prepared a small test case that can be used to reproduce the same environment. The buffer in between the ADC and DAC is not actually necessary, just connect pin26 and pin33. Data coming from the adc is send back to the pc over the COM port using following protocol:
[image]

The test case can be found in this repository: https://github.com/stevenvdr/esp-adc-bug

The expected behaviour: this behaviour can be seen when putting the ADC in 11 bit precision mode, the sine wave can easily be identified in the output samples (the steps are due to the frequency of the DAC).
[image]

The problem: The problem occurs when the adc is set to 12 bit mode. The sine is still identifiable, but errors occur on the output value of the samples. (The peaks around output value 3400).
[image]

The data shown here can be found in the github project. 

Following things have been checked:
- Adjusted attenuator settings. Whenever the 12 bit  value is around output value 3400, the error occurs.
- Different sine frequencies
- Waveforms have been checked with oscilloscope

projectgus, 2017-04-10:
Hi @stevenvdr ,

Thanks for providing such good quality data and a comprehensive report.

I think this may be related to #164, the ESP32 ADC has some linearity issues. There is work underway to better characterise it, so that glitches like this can be corrected.

Angus

stevenvdr, 2017-04-13:
The ESP-IDF has a feature that adds a carriage return '\c' before every '\n' character that is printed on stdout. This caused the bytes to shift and the data to become invalid. By not defining CONFIG_NEWLIB_STDOUT_ADDCR the issue seems to be resolved.

closed 2017-04-13 by stevenvdr

picpic020960, 2017-05-25:
bonjour,
i test your program. compil OK. but when run , ko.
with 'flash monitor' i see "E (204) I2S: Invalid bits per sample" (see below)
please help.
Regards

I (1117) heap_alloc_caps: At 40093178 len 0000CE88 (51 KiB): IRAM
I (1138) cpu_start: Pro cpu start user code
I (1195) cpu_start: Starting scheduler on PRO CPU.
I (204) cpu_start: Starting scheduler on APP CPU.
E (204) I2S: Invalid bits per sample
I (204) ADC: Done initialising DAC
I (204) ADC:Guru Meditation Error of type LoadProhibited occurred on cor
andled.
Register dump:
PC      : 0x40084349  PS      : 0x00060031  A0      : 0x80084d12  A1
0x40084349: xTaskIncrementTick at D:/ESP32/esp32_win32_msys2_environment
/esp-idf/components/freertos/tasks.c:4415

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5
A6      : 0x00000012  A7      : 0x00000001  A8      : 0x800842b7  A9
A10     : 0x3ffb1c68  A11     : 0x00060021  A12     : 0x3ffb8d8c  A13
A14     : 0x00000000  A15     : 0x3ffb8ad0  SAR     : 0x00000015  EXCCAU
EXCVADDR: 0x00000008  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT

Backtrace: 0x40084349:0x3ffb05b0 0x40084d12:0x3ffb05d0 0x40085868:0x3ffb
600

stevenvdr, 2017-05-26:
@picpic020960 Some changes were recently made to the I2S driver of ESP-IDF. This has probably caused your error. I recommend changing the .bits_per_sample property in the i2s_config struct to 16. (Note: I haven't checked this yet)

stevenvdr, 2017-05-26:
I've updated the code in the repository to work with the new version of ESP-IDF.

picpic020960, 2017-06-04:
OK
works fine
thanks

picpic020960, 2017-06-05:
One question : 
"Waveforms have been checked with oscilloscope"
How to ?  I2S is not numeric ?
Thanks