espressif/esp-idf #505: Remove pthread.h in newlib please.

BernardXiong, 2017-04-09:
Because there is no pthreads implementation in idf, please remove this file in newlib.

BernardXiong, 2017-04-09:
also remove the "sched.h" file.

loboris, 2017-04-09:
[Lua-RTOS-ESP32](https://github.com/whitecatboard/Lua-RTOS-ESP32) uses _pthread.h_ in thread implementation, possible some other projects too, why ask for removal just because you don't need it?

BernardXiong, 2017-04-10:
@loboris because we implement a standalone pthread.

igrr, 2017-04-10:
@jolivepetrus if we remove pthread.h here, can you add it on your side? Will this cause any issues?

jolivepetrus, 2017-04-10:
In our side we use our own pthread.h. I think that there is not problem if you remove pthread.h from newlib.

madscientist42, 2017-04-11:
The question would be if anyone else is impacted by that distinction.  

@BernardXiong 

Just because you're using things that way, doesn't lead to it going the way you're asking for.  If you're implementing a standalone pthread instead of working with the one provided- WHY?

If it's not capable, would it NOT be better to hand-fix your problem and ask for the gap there to be fixed by the IDF team?

I'm...leery...of making arbitrary requests.  I've got a partial implementation of lwBT built and in the process of debugging to see how well Classic works at least until we get a UART HCI passthrough mode for the BT made available in the IDF.  I don't ask for them to rip the current incomplete Bluedroid out because there's a conflict with namespace in the Bluedroid for SDP.

BernardXiong, 2017-04-12:
@madscientist42 
I don't know what's your meaning.

The `pthread.h` file in idf is from newlib. Mostly, there is no this `pthread.h` file in toolchains, for example in code sourcery g++ lite. Anyway, I can modify our implementation to make a stand pthread.h file with newlib.

igrr, 2017-09-19:
Closing this as "wontfix" because we now have a minimal pthread implementation, hence pthread.h is actually needed.

closed 2017-09-19 by igrr