espressif/esp-idf #509: uart_set_pin does not take care of peripheral muxing

rojer, 2017-04-11:
...at least that's my suspicion.
default RX and TX pins for UART1 are GPIO9 and 10 respectively, which are also muxed to SD2 and SD3.
if one tries to use GPIO9/SD2 as RX, it sort of works, but causes random crashes after justa  few bytes received.
the suspicion is that the SPI peripheral remains muxed to GPIO9 and it interferes with data transfers.
note: it's strange that this is the case, give that DIO flash mode is used so SD2 and SD3 are supposed to be unused.

anyway, even if the above analysis is incorrect: try using GPIO9 for U1 RX -> BOOM, use some other GPIO (e.g. 18) -> ok.

igrr, 2017-04-11:
> if one tries to use GPIO9/SD2 as RX

Do you mean that you connect some external UART TX to GPIO9?

On most boards, GPIO9 is connected to the SPI flash chip. So I don't see how UART can possibly be used on GPIO9 if SPI flash chip is also connected to this pin.

rojer, 2017-04-11:
in DIO mode SD2 and SD3 should be unused by the flash chip, i was able to free up 2 GPIOs that way on ESP8266

igrr, 2017-04-11:
This is true if these pins are not connected to the flash chip. On ESP-WROOM32, for example, these two pins are still connected to the flash chip, and can not be manipulated arbitrarily. If you disconnect these pins from the flash chip, you should be able to use GPIO 9 and 11.

rojer, 2017-04-11:
ok, i'll just stay clear of gpio 7-11 then. thanks!

closed 2017-04-11 by rojer