espressif/esp-idf #512: I2S Driver Pin Assignment

blade121, 2017-04-12:
The I2S Driver only works if the pins assigned in the i2s_pin_config_t default to GPIO functions.
To repro, try assigning I2S_BCLK_GPIO_NUM to GPIO12 or 14 an calling i2s_set_pin (BCLK will not output).

I worked around this by assigning the direction and function of each pin before calling i2s_set_pin, as shown below.

	PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[I2S_DI_GPIO_NUM], PIN_FUNC_GPIO);
	gpio_set_direction(I2S_DI_GPIO_NUM, GPIO_MODE_INPUT);

	PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[I2S_DO_GPIO_NUM], PIN_FUNC_GPIO);
	gpio_set_direction(I2S_DO_GPIO_NUM, GPIO_MODE_OUTPUT);

	PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[I2S_BCLK_GPIO_NUM], PIN_FUNC_GPIO);
	gpio_set_direction(I2S_BCLK_GPIO_NUM, GPIO_MODE_OUTPUT);

	PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[I2S_WS_GPIO_NUM], PIN_FUNC_GPIO);
	gpio_set_direction(I2S_WS_GPIO_NUM, GPIO_MODE_OUTPUT);

Seems like this should be included in the i2s_set_pin function - the i2c driver equivalent function (i2c_set_pin) sets the pin functions.

tuanpmt, 2017-04-12:
Because GPIO12..15 are connected to JTAG using IOMUX. You need to mux them back to GPIO to use with GPIO matrix.
Please add `PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[12], PIN_FUNC_GPIO);` before call `i2s_set_pin`

FayeY, 2017-05-03:
Hi blade121, is this problem solved?

closed 2017-07-07 by FayeY