espressif/esp-idf #513: BLE stop advartasing after several connect/disconnect

AlexNikTaran, 2017-04-12:
HI,
I used all BLE examples from SDK, but in the any examples i fix same problems.

BLE start advartasing process
Connect to ESP32
Disconnect from ESP32
After 5-6 cyrcles steps 1-3 BLE don't advartasing.

igrr, 2017-04-14:
@TianHao-Espressif please advise

tushaar, 2017-04-17:
Same issue here.

TianHao-Espressif, 2017-04-17:
We will reproduce and debug it .

YelongHuang, 2017-05-23:
@AlexNikTaran I think this bug has been fixed in the latest idf sdk, can you update the sdk and try again?

meriororen, 2018-11-25:
Hi, @chegewara thanks for taking time to re-test the stack. However, my test was a little different from yours, the steps are:

1 connect
2 set MTU to 255
3 subscribe to NOTIFY characteristics
4 send 1 packet in full MTU size
5 wait for notification
6 send another packet
7 go through 4 to 6 about 3 to 5 times
8 disconnect

chegewara, 2018-11-25:
I done similar to yours test and i have to disappoint you, it works on my side. I dont have exact numbers as your:
- mtu=255, 
- size of write about 200 bytes, 
- every connection 4 write request,
- each write request is followed by notification,
- 100ms delay between each write request,
My test still running (about 750 loops so far), and logs(after 420-ish loops):
```
I (529241) SampleServer: onConnect
W (529242) SampleServer: custom gatts event handler, event: 14
W (529389) SampleServer: custom gap event handler, event: 20
W (529569) SampleServer: custom gatts event handler, event: 4
W (529616) SampleServer: custom gatts event handler, event: 2
I (529620) SampleServer: write to char:
W (529620) SampleServer: custom gatts event handler, event: 2
W (529621) task: send notification 1801

I (529716) SampleServer: write to char:
W (529716) SampleServer: custom gatts event handler, event: 2
W (529717) task: send notification 1802

I (529813) SampleServer: write to char:
W (529814) SampleServer: custom gatts event handler, event: 2
W (529814) task: send notification 1803

I (529911) SampleServer: write to char:
W (529911) SampleServer: custom gatts event handler, event: 2
W (529912) task: send notification 1804

I (530057) SampleServer: onDisconnect
W (530059) SampleServer: custom gatts event handler, event: 15
W (530065) SampleServer: custom gap event handler, event: 0
W (530066) SampleServer: custom gap event handler, event: 1
W (530070) SampleServer: custom gap event handler, event: 6
I (530337) SampleServer: onConnect
W (530337) SampleServer: custom gatts event handler, event: 14
W (530513) SampleServer: custom gap event handler, event: 20
W (530672) SampleServer: custom gatts event handler, event: 4
W (530719) SampleServer: custom gatts event handler, event: 2
I (530722) SampleServer: write to char:
W (530723) SampleServer: custom gatts event handler, event: 2
W (530723) task: send notification 1805

I (530818) SampleServer: write to char:
W (530819) SampleServer: custom gatts event handler, event: 2
W (530819) task: send notification 1806

I (530916) SampleServer: write to char:
W (530916) SampleServer: custom gatts event handler, event: 2
W (530917) task: send notification 1807

I (531013) SampleServer: write to char:
W (531014) SampleServer: custom gatts event handler, event: 2
W (531014) task: send notification 1808

I (531160) SampleServer: onDisconnect
```

Im rather having poblems with my client app (scan sometimes stuck), but its my fault i think. That good stress test for Kolban's library, thanks.

meriororen, 2018-11-26:
@chegewara thanks for trying to reproduce. I'm not familiar with "Kolban's Library" you are referring to, are you referring to Arduino library?

chegewara, 2018-11-26:
@meriororen 
Its both, esp-idf C++ and arduino. https://github.com/nkolban/esp32-snippets

My test is still running, i still got problems with client code, but server and advertising works good after 7500 cycles.

chegewara, 2018-11-27:
@meriororen I am persistent and i am still running this test. Over 26000 connections without issue:
```
I (92029371) SampleServer: onConnect
W (92029372) SampleServer: custom gatts event handler, event: 14
W (92029609) SampleServer: custom gap event handler, event: 20
W (92030134) SampleServer: custom gatts event handler, event: 4
W (92030456) SampleServer: custom gatts event handler, event: 2
I (92030477) SampleServer: write to char:
W (92030477) SampleServer: custom gatts event handler, event: 2
W (92030478) task: send notification 104418

I (92030566) SampleServer: write to char:
W (92030567) SampleServer: custom gatts event handler, event: 2
W (92030568) task: send notification 104419

I (92030678) SampleServer: write to char:
W (92030678) SampleServer: custom gatts event handler, event: 2
W (92030679) task: send notification 104420

I (92030769) SampleServer: write to char:
W (92030770) SampleServer: custom gatts event handler, event: 2
W (92030770) task: send notification 104421

I (92030982) SampleServer: onDisconnect
W (92030984) SampleServer: custom gatts event handler, event: 15
W (92030990) SampleServer: custom gap event handler, event: 0
W (92030991) SampleServer: custom gap event handler, event: 1
W (92030995) SampleServer: custom gap event handler, event: 6
```

jaracil, 2018-12-27:
I'm having this issue in IDF versions 3.1.x and later. In IDF versions 3.0.x and earlier I have found a solution which is to reactivate the BLE Advertising every 30 seconds.

The problem appears when there are failed connections because long distance or low coverage. Then bluetooth stack stops sending advertisings.

In IDF versions 3.1.x and latter even triggering a esp_ble_gap_start_advertising returning ESP_OK  every 30 seconds the issue isn't fixed.

@chegewara Can you launch your connection test in circumstances where connections are not clean? (put the BLE distance to the limit) 

Thanks.

Weijian-Espressif, 2018-12-29:
[image]

@jaracil After the connection is established, the device will automatically stop scan or stop adv. 
It is defined by the Bluetooth spec.

Weijian-Espressif, 2019-01-14:
@jaracil ,  please start advertisings when get connection event or disconnection event.

jaracil, 2019-01-16:
@Weijian-Espressif 
Activating the advertisings after each connection / disconnection was the usual procedure, but in some occasions the advertising was not activated. The workaround I used, was to reactivate the advertisings every 30 seconds. This workaround works in versions 3.0.x. In 3.1.x  Advertisings are deactivated forever until the next reset.

Weijian-Espressif, 2019-02-15:
@jaracil please update to the commit id `3aa5a1e5a236c3ccdc01a8809964132ebbd45bcc` and try, thanks.

jaracil, 2019-02-19:
> @jaracil please update to the commit id `3aa5a1e5a236c3ccdc01a8809964132ebbd45bcc` and try, thanks.

@Weijian-Espressif sorry but this hash don't match with any commit.

Weijian-Espressif, 2019-02-21:
@jaracil please try again.

Weijian-Espressif, 2019-02-21:
@jaracil After the ESP32 receives the connection request, the controller will automatically stop the Advertisings. This is defined by the Bluetooth spec, I think it is not a bug. If you want to use multiple connections, after the device is connected, please call the API to start Advertisings again.

jaracil, 2019-02-21:
> @jaracil After the ESP32 receives the connection request, the controller will automatically stop the advertisings. This is defined by the Bluetooth spec, I think it is not a bug.

Of course it's not a bug.

> If you want to use multiple connections, after the device is connected, please call the API to start advertisings again.

This is what I'm doing.

The bug is sometimes the ADV don't start again after esp_ble_gap_start_advertising even when returning ESP_OK.

I'm going to test NimBLE and check if the bug is in Bluedroid or in VHCI.

Thanks.

Weijian-Espressif, 2019-02-21:
@jaracil, please update to the commit id `3aa5a1e5a236c3ccdc01a8809964132ebbd45bcc` and add some printing in btm_ble_conn_complete() and btm_ble_write_adv_enable_complete().  

we need check :
    The adv did not start successfully  or the Bluetooth connection request was received after the adv was successfully started, and the controller stopped adv.

please help test, thanks.

Weijian-Espressif, 2019-03-15:
@jaracil, any update ?

Weijian-Espressif, 2019-04-01:
@jaracil, Can you help test latest v3.2, we have fixed a bug about BT sleep.

jaracil, 2019-08-03:
> @jaracil, Can you help test latest v3.2, we have fixed a bug about BT sleep.

After a month of testing with IDF V3.2.2 in production, I have not detected the problem again.

Alvin1Zhang, 2019-10-12:
@AlexNikTaran @jaracil Thanks for reporting and the updates. Would you please help confirm whether the issue still happens or not? Could this ticket be closed? Thanks.

Alvin1Zhang, 2020-06-11:
Thanks for reporting and updates. Feel free to reopen if the issue still happens. Thanks.

closed 2020-06-11 by Alvin1Zhang

kovid-dozee, 2022-11-29:
Hello All,
I am using ESP32 for GATTC, GATTS and WiFi. While I am using Bluetooth functionality I don't work on WiFi. I am facing the same issue in my device as well, whenever the device is in the gattc mode and takes the data from server on Notification. After taking few values the bluetooth advertising stops and Bluetooth stops working. I need to restart the device to make it work again. I am using IDF v4.3-dirty as the toolchain.