espressif/esp-idf #514: RMT support does not provide a way to test if transmist is done

Makuna, 2017-04-12:
rmt_wait_tx_done() will wait for ever.

I suggest exposing a function like...
bool rmt_is_tx_done() 

less interesting but a viable alternative is to modify rmt_wait_tx_done() to take an optional param for the timeout with a default value of portMAX_DELAY.  Further modify it to return the value from the xSemaphoreTake if it fails.

jeremyherbert, 2017-06-07:
Just tripped on this problem today. Any timeframe on a fix?

Makuna, 2017-08-07:
Any updates on this?

igrr, 2017-08-07:
We have a PR which adds the timeout value to `rmt_wait_tx_done` and fixes #274. Will give it a nudge, thanks for the reminder.

EBZ-Krisemendt, 2017-09-15:
@iggr does closing https://github.com/espressif/esp-idf/issues/274 mean, that this one can be closed, too?

igrr, 2017-09-15:
Yes, thanks for reminder.

closed 2017-09-15 by igrr