espressif/esp-idf #516: UDP transfer issue

long585, 2017-04-13:
my test case use  sendto() to send 1200bytes data to my udp server on pc. 
if the send speed too fast, then cause a issue ,I found the problem in this function by tracking the code

```
static err_t
low_level_output(struct netif *netif, struct pbuf *p)
{
  wifi_interface_t wifi_if = tcpip_adapter_get_esp_if(netif);
  struct pbuf *q = p;
  err_t ret;

  if (wifi_if >= ESP_IF_MAX) {
    return ERR_IF;
  }

  if(q->next == NULL) {
    ret = esp_wifi_internal_tx(wifi_if, q->payload, q->len);
  } else {
    LWIP_DEBUGF(PBUF_DEBUG, ("low_level_output: pbuf is a list, application may has bug"));
    q = pbuf_alloc(PBUF_RAW_TX, p->tot_len, PBUF_RAM);
    if (q != NULL) {
      pbuf_copy(q, p);
    } else {
        
      return ERR_MEM;
    }
    ret = esp_wifi_internal_tx(wifi_if, q->payload, q->len);
    pbuf_free(q);
  }
    printf("e3:%d\n",ret);
  return ret;
}

```
the debug info is "e3:-1"

igrr, 2017-04-14:
-1 means `ERR_MEM`, this error is returned by `esp_wifi_internal_tx` when all the transmit buffers have been used by the application. This is not an unrecoverable error — the application can detect this by checking `errno` value, and retry transmission after some delay. You may also check if increasing the value of `Max number of WiFi dynamic TX buffers` configuration in menuconfig (under Component config > Wifi) has any impact.

If the rate at which your application is sending data over UDP is higher than the actual data transfer rate of the WiFi link, you will get ERR_MEM sooner or later. Application needs to implement rate control measures to handle such cases (i.e. by degrading sampling rates, increasing the compression ratio of data you are trying to send).

closed 2017-04-14 by igrr

long585, 2017-04-14:
Your official documents which write UDP transmission up to 150Mbps, How did you reach the speed?

igrr, 2017-04-14:
That is the throughput the hardware is capable of; various software overheads related to task switching, queue/mutex operations time, and so on, reduce the highest achievable throughput in an RTOS environment. Also the value is given for the case of testing in a shield box. In most practical cases (such as testing in air instead of the shield box) the throughput is limited due to external factors.

long585, 2017-04-14:
what  is  the udp transfer speed of your practical cases ? my test case probably only 10Mbps