espressif/esp-idf #520: Invalid cache access somewhere in BT

rojer, 2017-04-13:
updated to latest master (9050307dfe600c915adaccf8076220d6474876db) and got the following exception:

```
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x40086ae1  PS      : 0x00060034  A0      : 0x80004160  A1      : 0x3ffc0740  
A2      : 0x00000062  A3      : 0x00000001  A4      : 0x00000005  A5      : 0x3ffd3c60  
A6      : 0x001b0000  A7      : 0x0000d200  A8      : 0xbad00bad  A9      : 0x3f400f6c  
A10     : 0x00000002  A11     : 0xffffffff  A12     : 0x80111ee8  A13     : 0x3ffd3c00  
A14     : 0x0019d200  A15     : 0x3ffd3ca0  SAR     : 0x00000020  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4009ad74  LEND    : 0x4009ad7f  LCOUNT  : 0x00000000  

Backtrace: 0x40086ae1:0x3ffc0740 0x40004160:0x3ffc0760 0x400041dd:0x3ffc0780 0x40087e79:0x3ffc07a0 0x40087f70:0x3ffc07c0 0x40083906:0x3ffc07f0 0x40081678:0x3ffc0810
```

backtrace is: `ram_chip_i2c_readReg <- ? <- ? <- pll_correct_dcap <- bt_track_pll_cap <- btdm_bb_isr <- _xt_lowint1`

igrr, 2017-04-14:
Thanks, we have a fix for this, currently in testing.

tushaar, 2017-04-17:
@igrr 
Same issue here. Looking forward to fix.

chiragvaya, 2017-04-22:
@igrr @projectgus Same issue here.
Can you please give us an approximate date when this will be fixed? 
Am not able to use Bluetooth as ESP keeps rebooting.

linusaranha, 2017-04-23:
@igrr, Would it be possible to provide a patch for this in the mean time?

Regards,
Linus

igrr, 2017-04-25:
The temporary fix for this has been merged in https://github.com/espressif/esp-idf/commit/9d8425bd72aa2fd530370ad77998ac29a865de83, this change moves about 1kb of data from DROM to DRAM. We will optimize this so that only the data which gets used in ISRs is placed into DRAM.

closed 2017-04-25 by igrr

neilyoung, 2017-04-25:
@igrr Confirming, the crash is gone. Strange scattered scan behaviour. RadBeacons still not recognized.

jt040-jetnet, 2017-11-21:
hello all i am getting similar kind of error 
please suggest how to fix the issue 
i am using esp32 in arduino core 
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x400d13f0  PS      : 0x00060034  A0      : 0x40082388  A1      : 0x3ffc0bf0  
A2      : 0x00000000  A3      : 0x3ffc358c  A4      : 0x00000001  A5      : 0x40086408  
A6      : 0x00000003  A7      : 0x00060423  A8      : 0x80081142  A9      : 0x04000c01  
A10     : 0x00060423  A11     : 0x00000000  A12     : 0x00060023  A13     : 0x3ffc0bd0  
A14     : 0x3ffc0c18  A15     : 0x00000001  SAR     : 0x00000018  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d13f0:0x3ffc0bf0 0x40082385:0x3ffc0c10 0x4008f551:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:716
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14

projectgus, 2017-11-23:
Hi @jt040-jetnet ,

This is usually caused by an interrupt handler which should be marked as "not IRAM safe" but was marked as IRAM safe. You can get more information by decoding the Backtrace line with this tool:
https://github.com/me-no-dev/EspExceptionDecoder

For further discussion please open this as an issue on the [arduino-esp32 github](https://github.com/espressif/arduino-esp32/issues/) (or post on [the Arduino for ESP32 forum ](https://esp32.com/viewforum.php?f=19) as the issues related to Arduino rather than ESP-IDF itself.