espressif/esp-idf #521: components/bootloader: Build fails for CONFIG_FLASHMODE_QIO=y

devsaurus, 2017-04-13:
When I enable QIO flash mode with `CONFIG_FLASHMODE_QIO=y` the linker fails with
```
build/bootloader/main/libmain.a(flash_qio_mode.o):(.literal.enable_qio_mode+0x8): undefined reference to `esp_rom_spiflash_master_config_readmode'
build/bootloader/main/libmain.a(flash_qio_mode.o): In function `enable_qio_mode':
sdk/esp32-esp-idf/components/bootloader/src/main/./flash_qio_mode.c:167: undefined reference to `esp_rom_spiflash_master_config_readmode'
```

A recursive grep for `esp_rom_spiflash_master_config_readmode` shows the call, the prototype definition but no implementation of this function (9050307dfe600c915adaccf8076220d6474876db).

projectgus, 2017-04-14:
Hi @devsaurus ,

Thanks for reporting this. This was a regression merged earlier this week (1132ef7554). We are currently in the process of merging a fix, it should land in the next few days.

Angus

igrr, 2017-04-17:
Fixed in 85e76a7.

closed 2017-04-17 by igrr