espressif/esp-idf #522: How to disable the logs of libnet80211.a

young-mu, 2017-04-14:
Logs like these:

I (14863) wifi: frc2_timer_task_hdl:3ffed988, prio:22, stack:2048
I (14868) wifi: Init lldesc rx mblock:25
I (14868) wifi: Init lldesc rx ampdu len mblock:7
I (14868) wifi: Init lldesc rx ampdu entry mblock:4
I (14871) wifi: pp_task_hdl : 3fffa7fc, prio:23, stack:8192
I (14877) wifi: mode : sta (24:0a:c4:01:a6:60)
I (15603) wifi: n:5 0, o:1 0, ap:255 255, sta:5 0, prof:1
I (16164) wifi: state: init -> auth (b0)
I (16201) wifi: state: auth -> assoc (0)
I (16208) wifi: state: assoc -> run (10)

How to disable them?

igrr, 2017-04-14:
```c++
#include "esp_log.h"

esp_log_level_set("*", ESP_LOG_WARNING);
```

http://esp-idf.readthedocs.io/en/latest/api/system/log.html

closed 2017-04-14 by igrr

young-mu, 2017-04-14:
OK, it works for me. Thank you!
(BTW, it should be **ESP_LOG_WARN**, not **ESP_LOG_WARNING**)