espressif/esp-idf #523: deep sleep issue

long585, 2017-04-14:
use the esp-idf 2.0 the deep sleep example had happen this issue:

```
rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3388
ho 0 tail 12 room 4
load:0x40078000,len:7504
load:0x40080000,len:252
entry 0x40080034
I (47) boot: ESP-IDF v2.0-rc1-30-gc61fdff 2nd stage bootloader
I (48) boot: compile time 09:18:46
I (48) boot: Enabling RNG early entropy source...
I (68) boot: SPI Speed      : 40MHz
I (80) boot: SPI Mode       : DIO
I (93) boot: SPI Flash Size : 4MB
I (105) boot: Partition Table:
I (116) boot: ## Label            Usage          Type ST Offset   Length
I (139) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (162) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (186) boot:  2 factory          factory app      00 00 00010000 00100000
I (209) boot: End of partition table
I (222) boot: Disabling RNG early entropy source...
I (239) boot: Loading app partition at offset 00010000
I (830) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
I (831) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x074c4 ( 29892) map
I (847) boot: segment 2: paddr=0x000274d4 vaddr=0x3ffb0000 size=0x024bc (  9404) load
I (877) boot: segment 3: paddr=0x00029998 vaddr=0x40080000 size=0x00400 (  1024) load
I (900) boot: segment 4: paddr=0x00029da0 vaddr=0x40080400 size=0x17900 ( 96512) load
I (972) boot: segment 5: paddr=0x000416a8 vaddr=0x400c0000 size=0x00034 (    52) load
I (973) boot: segment 6: paddr=0x000416e4 vaddr=0x00000000 size=0x0e924 ( 59684) 
I (989) boot: segment 7: paddr=0x00050010 vaddr=0x400d0018 size=0x24030 (147504) map
I (1015) boot: segment 8: paddr=0x00074048 vaddr=0x50000200 size=0x00008 (     8) load
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.

Backtrace: 0x4008433a:0x3ffe3c00 0x400847c8:0x3ffe3c20 0x40081a18:0x3ffe3c40 0x40080f68:0x3ffe3c60 0x40078429:0x3ffe3ca0 0x40078630:0x3ffe3cd0 0x40078b90:0x3ffe3d50 0x400800f9:0x3ffe3e70 0x40007c34:0x3ffe3eb0 0x40000740:0x3ffe3f20

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3388
ho 0 tail 12 room 4
load:0x40078000,len:7504
load:0x40080000,len:252
entry 0x40080034
I (47) boot: ESP-IDF v2.0-rc1-30-gc61fdff 2nd stage bootloader
I (47) boot: compile time 09:18:46
I (47) boot: Enabling RNG early entropy source...
I (67) boot: SPI Speed      : 40MHz
I (80) boot: SPI Mode       : DIO
I (92) boot: SPI Flash Size : 4MB
I (105) boot: Partition Table:
I (116) boot: ## Label            Usage          Type ST Offset   Length
I (139) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (162) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (185) boot:  2 factory          factory app      00 00 00010000 00100000
I (209) boot: End of partition table
I (221) boot: Disabling RNG early entropy source...
I (239) boot: Loading app partition at offset 00010000
I (830) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
I (830) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x074c4 ( 29892) map
I (846) boot: segment 2: paddr=0x000274d4 vaddr=0x3ffb0000 size=0x024bc (  9404) load
I (877) boot: segment 3: paddr=0x00029998 vaddr=0x40080000 size=0x00400 (  1024) load
I (899) boot: segment 4: paddr=0x00029da0 vaddr=0x40080400 size=0x17900 ( 96512) load
I (971) boot: segment 5: paddr=0x000416a8 vaddr=0x400c0000 size=0x00034 (    52) load
I (972) boot: segment 6: paddr=0x000416e4 vaddr=0x00000000 size=0x0e924 ( 59684) 
I (988) boot: segment 7: paddr=0x00050010 vaddr=0x400d0018 size=0x24030 (147504) map
I (1014) boot: segment 8: paddr=0x00074048 vaddr=0x50000200 size=0x00008 (     8) load
I (1041) cpu_start: Pro cpu up.
I (1052) cpu_start: Starting app cpu, entry point is 0x40080e70
I (0) cpu_start: App cpu up.
I (1085) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1107) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (1128) heap_alloc_caps: At 3FFB5A70 len 0002A590 (169 KiB): DRAM
I (1149) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1170) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1192) heap_alloc_caps: At 40097D00 len 00008300 (32 KiB): IRAM
I (1213) cpu_start: Pro cpu start user code
I (1273) cpu_start: Starting scheduler on PRO CPU.
Not a deep sleep reset
I (209) cpu_start: Starting scheduler on APP CPU.
Enabling timer wakeup, 20s
Entering deep sleep
```

igrr, 2017-04-14:
Could you please run `xtensa-esp32-elf-addr2line -pfia -e build/your-app-name.elf` and feed it the "Backtrace" line from the log above? The elf file has to match the binary which produced this output.
Thanks

igrr, 2017-04-14:
Also, the output `ESP-IDF v2.0-rc1-30-gc61fdff 2nd stage bootloader` doesn't indicate that you are using v2.0. This seems to be some version between v2.0-rc1 and v2.0-rc2. Could you please double check the version of IDF using `git describe` command, running it withing IDF_PATH?

long585, 2017-04-14:
my esp-idf  version is `v2.0-rc1-443-g3cad00f`

long585, 2017-04-14:
Oh,i konw my bootloader is old，but the example is use new idf to compiled .

long585, 2017-04-14:
I had replace the old bootloader,but the issue still exist.

long585, 2017-04-14:
```
rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:4124
ho 0 tail 12 room 4
load:0x40078000,len:9228
load:0x40080000,len:252
entry 0x40080034
I (48) boot: ESP-IDF v2.0-rc1-443-g3cad00f-dirty 2nd stage bootloader
I (48) boot: compile time 14:20:43
I (52) boot: Enabling RNG early entropy source...
I (69) boot: SPI Speed      : 40MHz
I (82) boot: SPI Mode       : DIO
I (94) boot: SPI Flash Size : 4MB
I (106) boot: Partition Table:
I (118) boot: ## Label            Usage          Type ST Offset   Length
I (140) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (163) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (187) boot:  2 factory          factory app      00 00 00010000 00100000
I (210) boot: End of partition table
I (223) boot: Disabling RNG early entropy source...
I (240) boot: Loading app partition at offset 00010000
I (831) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
I (832) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x074c4 ( 29892) map
I (848) boot: segment 2: paddr=0x000274d4 vaddr=0x3ffb0000 size=0x024bc (  9404) load
I (878) boot: segment 3: paddr=0x00029998 vaddr=0x40080000 size=0x00400 (  1024) load
I (901) boot: segment 4: paddr=0x00029da0 vaddr=0x40080400 size=0x17900 ( 96512) load
I (973) boot: segment 5: paddr=0x000416a8 vaddr=0x400c0000 size=0x00034 (    52) load
I (974) boot: segment 6: paddr=0x000416e4 vaddr=0x00000000 size=0x0e924 ( 59684) 
I (990) boot: segment 7: paddr=0x00050010 vaddr=0x400d0018 size=0x24030 (147504) map
I (1016) boot: segment 8: paddr=0x00074048 vaddr=0x50000200 size=0x00008 (     8) load
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.

Backtrace: 0x4008433a:0x3ffe3bf0 0x400847c8:0x3ffe3c10 0x40081a18:0x3ffe3c30 0x40080f68:0x3ffe3c50 0x40078545:0x3ffe3c90 0x4007874c:0x3ffe3cc0 0x40078d1d:0x3ffe3d40 0x400800f9:0x3ffe3e70 0x40007c34:0x3ffe3eb0 0x40000740:0x3ffe3f20

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:4124
ho 0 tail 12 room 4
load:0x40078000,len:9228
load:0x40080000,len:252
entry 0x40080034
I (48) boot: ESP-IDF v2.0-rc1-443-g3cad00f-dirty 2nd stage bootloader
I (48) boot: compile time 14:20:43
I (52) boot: Enabling RNG early entropy source...
I (68) boot: SPI Speed      : 40MHz
I (81) boot: SPI Mode       : DIO
I (93) boot: SPI Flash Size : 4MB
I (106) boot: Partition Table:
I (117) boot: ## Label            Usage          Type ST Offset   Length
I (140) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (163) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (186) boot:  2 factory          factory app      00 00 00010000 00100000
I (210) boot: End of partition table
I (223) boot: Disabling RNG early entropy source...
I (240) boot: Loading app partition at offset 00010000
I (831) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
I (831) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x074c4 ( 29892) map
I (847) boot: segment 2: paddr=0x000274d4 vaddr=0x3ffb0000 size=0x024bc (  9404) load
I (878) boot: segment 3: paddr=0x00029998 vaddr=0x40080000 size=0x00400 (  1024) load
I (901) boot: segment 4: paddr=0x00029da0 vaddr=0x40080400 size=0x17900 ( 96512) load
I (972) boot: segment 5: paddr=0x000416a8 vaddr=0x400c0000 size=0x00034 (    52) load
I (973) boot: segment 6: paddr=0x000416e4 vaddr=0x00000000 size=0x0e924 ( 59684) 
I (989) boot: segment 7: paddr=0x00050010 vaddr=0x400d0018 size=0x24030 (147504) map
I (1015) boot: segment 8: paddr=0x00074048 vaddr=0x50000200 size=0x00008 (     8) load
I (1042) cpu_start: Pro cpu up.
I (1053) cpu_start: Starting app cpu, entry point is 0x40080e70
I (0) cpu_start: App cpu up.
I (1086) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1108) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (1129) heap_alloc_caps: At 3FFB5A70 len 0002A590 (169 KiB): DRAM
I (1150) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1171) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1193) heap_alloc_caps: At 40097D00 len 00008300 (32 KiB): IRAM
I (1214) cpu_start: Pro cpu start user code
I (1274) cpu_start: Starting scheduler on PRO CPU.
Not a deep sleep reset
I (209) cpu_start: Starting scheduler on APP CPU.
Enabling timer wakeup, 20s
Entering deep sleep
```
@igrr

igrr, 2017-04-14:
3cad00f is not v2.0 release, this is a fairly recent version of the master branch.

I have checked out commit 3cad00f and tested it on two boards:
- ESP_WROVER_KIT
- widora air

I get expected output (reset by timer, or ULP) on both of them.

Could you please decode the backtrace as I have instructed above?

long585, 2017-04-14:
my board is esp-wroom-32

closed 2017-04-18 by igrr