espressif/esp-idf #525: Is there any method to get time for general purpose?

asukiaaa, 2017-04-16:
I tried the following method to get time and succeeded in getting time with using the first one.
However, it may not a method to get time for general purpose.
Is there any good method to get time?

Method | Status
--- | ---
esp_log_early_timestamp() |  Succeeded in building without any warning but it may not be used except for logging.
system_get_time() | Succeeded in building with duplication warning.
system_get_rtc_time() | Cannot build.

There was a similar Issue https://github.com/espressif/esp-idf/issues/196 but it was closed by announcing that `system_get_rtc_time()` is removed.

Thanks.

igrr, 2017-04-16:
[`gettimeofday`](http://man7.org/linux/man-pages/man2/gettimeofday.2.html) function should be used to get current time.

You may also use `time` and `clock` functions, which have lower resolution.

asukiaaa, 2017-04-17:
Thank you for the information.
Oh, esp-idf can call c time library.

I can get milliseconds with using `clock`.
```c
#include <time.h>

uint32_t clock_millis;

static void somefunc() {
    clock_millis = (uint32_t) (clock() * 1000 / CLOCKS_PER_SEC);
    printf("clock_millis: %d\n", clock_millis);
}
```

closed 2017-04-17 by asukiaaa