espressif/esp-idf #526: Gatts Example - Guru Meditation Error: Core  0 panic'ed

tushaar, 2017-04-16:
Im using the Gatts Example. Esp keeps rebooting.
Is these come config that needs to be done.

Rebooting...
ets Jun  8 2016 00:22:57
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3152
load:0x40078000,len:8568
load:0x40080000,len:252
0x40080000: _iram_start at ??:?

entry 0x40080034
0x40080034: _iram_start at ??:?

I (1356) cpu_start: Pro cpu up.
I (1356) cpu_start: Single core mode
I (1358) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1370) heap_alloc_caps: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1391) heap_alloc_caps: At 3FFCE740 len 000118C0 (70 KiB): DRAM
I (1412) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1433) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1455) heap_alloc_caps: At 40098884 len 0000777C (29 KiB): IRAM
I (1476) cpu_start: Pro cpu start user code
I (1535) cpu_start: Starting scheduler on PRO CPU.
BTDM CONTROLLER VERSION: 010101
btip start
copy .data from 4000d890 to 3ffae6e0, len 00001830
set .bss 0x0 from 3ffb8000 to 3ffbff70, len 00007f70
BTDM ROM VERSION 0101
BD_ADDR: 24:0A:C4:05:37:CA
NVDS MAGIC FAILED
RF Init OK with coex
I (3551) phy: phy_version: 350, Mar 22 2017, 15:02:06, 0, 0
Enable Classic BT
Enable Low Energy
E (3555) BT: Startup BTU

E (3557) BT: GATTS_CreateService:  handle of service handle1
E (3572) BT: bta_dm_co_ble_load_local_keys: func not ported

I (3583) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 0

I (3584) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 1
E (3587) BT: Partial data write into ADV
I (3591) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 40
I (3597) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 44
I (3606) GATTS_DEMO: Stop adv successfully
I (3609) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40
I (3616) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40
I (3625) GATTS_DEMO: the gatts demo char length = 3
I (3630) GATTS_DEMO: prf_char[0] =11
I (3635) GATTS_DEMO: prf_char[1] =22
I (3639) GATTS_DEMO: prf_char[2] =33
I (3644) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 44
I (3651) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 46, service_handle 44
I (3659) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40
I (3667) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 47, service_handle 44

E (36333) BT: btm_ble_resolve_random_addr_on_conn_cmpl unable to match and resolve random address
E (36710) BT: smp_br_connect_callback is called on unexpected transport 2
I (36713) GATTS_DEMO: SERVICE_START_EVT, conn_id 0, remote 76:40:33:50:c6:cc:, is_conn 1
I (36716) GATTS_DEMO: SERVICE_START_EVT, conn_id 0, remote 76:40:33:50:c6:cc:, is_conn 1

**Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Register dump:**
PC      : 0x400869f9  PS      : 0x00060034  A0      : 0x80004160  A1      : 0x3ffc0530  
0x400869f9: ram_chip_i2c_readReg at ??:?

A2      : 0x00000062  A3      : 0x00000001  A4      : 0x00000005  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x3ffb8360  A8      : 0xbad00bad  A9      : 0x3f400f28  
A10     : 0x00000002  A11     : 0xffffffff  A12     : 0x8009813b  A13     : 0x3ffd3b70  
A14     : 0x3ffae270  A15     : 0x3ffd3b90  SAR     : 0x0000001a  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  

Backtrace: 0x400869f9:0x3ffc0530 0x40004160:0x3ffc0550 0x400041dd:0x3ffc0570 0x40087d91:0x3ffc0590 0x40087e88:0x3ffc05b0 0x4008367a:0x3ffc05e0 0x40081664:0x3ffc0600
0x400869f9: ram_chip_i2c_readReg at ??:?
0x40087d91: pll_correct_dcap at ??:?
0x40087e88: bt_track_pll_cap at ??:?
0x4008367a: btdm_bb_isr at intc.c:?
0x40081664: _xt_lowint1 at xtensa_vectors.o:?
Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

eclipsevl, 2017-04-16:
Known issue, have a look at #520

tushaar, 2017-04-16:
Thanks @eclipsevl

projectgus, 2017-04-20:
I'm going to close this as a duplicate of #520. @tushaar , please feel free to click Subscribe on the other issue in order to receive updates when they are available.

closed 2017-04-20 by projectgus

jt040-jetnet, 2017-11-21:
hello every one 
i am working of touch keys of esp32 arduino core
getting below error when interrupt occurs.
can any one help me fix the issue

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x400d13f0  PS      : 0x00060034  A0      : 0x40082388  A1      : 0x3ffc0bf0  
A2      : 0x00000000  A3      : 0x3ffc358c  A4      : 0x00000001  A5      : 0x40086408  
A6      : 0x00000003  A7      : 0x00060223  A8      : 0x80081142  A9      : 0x04000c01  
A10     : 0x00060223  A11     : 0x00000000  A12     : 0x00060023  A13     : 0x3ffc0bd0  
A14     : 0x3ffc0c18  A15     : 0x00000001  SAR     : 0x0000001a  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400d13f0:0x3ffc0bf0 0x40082385:0x3ffc0c10 0x4008f54c:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:716
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14