espressif/esp-idf #528: A issue about socket transfer

long585, 2017-04-18:
My test case only one task,if recved the udp client date then start udp server transfer.
```

static void my_udp_send()
{
    int send_size = 0;
    do{
        send_size = sendto(udp_cocket,test_buf,TEST_BUF_SIZE,0,(struct sockaddr*)&udp_client,addrsize);
        
        vTaskDelay(1/ portTICK_PERIOD_MS);
        
        
    }while(send_size == -1);

}
void create_udp_socket()
{
    char tbuf[10] ={0};
    int len=0,ret=0;
    
    
    printf("waitbit1\n");
    xEventGroupWaitBits(wifi_event_group, STA_CONNECTED_AP_BIT,
                            false, true, portMAX_DELAY);
    printf("waitbit2\n");                       
    udp_cocket = socket(AF_INET, SOCK_DGRAM, 0);
    memset(&udp_server_addr, 0, sizeof(udp_server_addr));
    udp_server_addr.sin_family = AF_INET;
    udp_server_addr.sin_addr.s_addr = 0;
    udp_server_addr.sin_port = htons(DEMO_LOCAL_UDP_PORT);
    
    ret = bind(udp_cocket, (struct sockaddr*)&udp_server_addr, sizeof(udp_server_addr));
    if (ret) {
        ESP_LOGE(TAG, " bind failed");
        //goto failed3;
    }
    addrsize = sizeof(udp_client);
    while(1)
    {
        vTaskDelay(10 / portTICK_PERIOD_MS);
        len = recvfrom(udp_cocket,tbuf,sizeof(tbuf),0,(struct sockaddr*)&udp_client,&addrsize);
        if(len > 0)
        {
            printf("start send MUT_THREAD\n");
            udp_can_transfer = 1;
#if 1            
            while(1)
            {
                my_udp_send();
            }
#endif            
        }
    }

}
```
After running for some time there was a issue:
```
start send MUT_THREAD
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x4008563b  PS      : 0x00060933  A0      : 0x80084878  A1      : 0x3ffdabb0  
A2      : 0x0000001d  A3      : 0x3ffdac34  A4      : 0x00000004  A5      : 0x3ffc6f5c  
A6      : 0x3ffdac60  A7      : 0x00000000  A8      : 0x800833f4  A9      : 0x3ffdaba0  
A10     : 0x00000003  A11     : 0x00060923  A12     : 0x00060923  A13     : 0x3ffb625c  
A14     : 0x3ffc1e8c  A15     : 0xfffffff8  SAR     : 0x00000015  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002d  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
```
By looking at the compilation file,i found the exception is：
```
 .text.vListInsert
                0x40085608       0x2f /home/long/esp/esp-idf-v2.0-rc1/examples/zq_wcb_project/build/freertos/libfreertos.a(list.o)
                0x40085608                vListInsert
 *fill*         0x40085637        0x1 
```

projectgus, 2017-04-20:
Did the crash dump include a line with a backtrace?

You can use the IDF monitor tool to automatically decode this information from the crash dumps:
http://esp-idf.readthedocs.io/en/latest/idf-monitor.html

projectgus, 2017-04-20:
Also, can you please post the full source code? You can post it over at https://gist.github.com and paste a link here.

long585, 2017-04-20:
My source full code was been posted，[https://gist.github.com/long585/6efe1ee248b19b6dc927a63dbab89a9c]

long585, 2017-04-20:
I posted the code modified to two tasks so that this issue will soon be able to appear

projectgus, 2017-04-21:
Thanks. As per the other comment I left, can you please run this code with "make monitor" and post the full output from the crash?

long585, 2017-04-21:
one issue:
```
I (46) boot: ESP-IDF v2.0-rc1-30-gc61fdff 2nd stage bootloader
I (46) boot: compile time 17:13:03
I (46) boot: Enabling RNG early entropy source...
I (66) boot: SPI Speed      : 40MHz
I (79) boot: SPI Mode       : DIO
I (91) boot: SPI Flash Size : 4MB
I (104) boot: Partition Table:
I (115) boot: ## Label            Usage          Type ST Offset   Length
I (138) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (161) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (184) boot:  2 factory          factory app      00 00 00010000 00100000
I (208) boot: End of partition table
I (220) boot: Disabling RNG early entropy source...
I (238) boot: Loading app partition at offset 00010000
I (956) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
I (956) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x089b4 ( 35252) map
I (972) boot: segment 2: paddr=0x000289c4 vaddr=0x3ffb0000 size=0x02408 (  9224) load
I (1002) boot: segment 3: paddr=0x0002add4 vaddr=0x40080000 size=0x00400 (  1024) load
I (1026) boot: segment 4: paddr=0x0002b1dc vaddr=0x40080400 size=0x1c98c (117132) load
I (1107) boot: segment 5: paddr=0x00047b70 vaddr=0x400c0000 size=0x00000 (     0) load
I (1108) boot: segment 6: paddr=0x00047b78 vaddr=0x00000000 size=0x08490 ( 33936) 
I (1125) boot: segment 7: paddr=0x00050010 vaddr=0x400d0018 size=0x3a498 (238744) map
I (1153) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1175) heap_alloc_caps: At 3FFC2068 len 0001DF98 (119 KiB): DRAM
I (1196) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (1217) heap_alloc_caps: At 4009CD8C len 00003274 (12 KiB): IRAM
I (1238) cpu_start: Pro cpu up.
I (1250) cpu_start: Starting app cpu, entry point is 0x40080d64
I (0) cpu_start: App cpu up.
I (1281) cpu_start: Pro cpu start user code
I (1530) phy: phy_version: 258, Nov 29 2016, 15:51:07, 0, 0
I (2122) cpu_start: Starting scheduler on PRO CPU.
I (861) cpu_start: Starting scheduler on APP CPU.
I (861) wifi: frc2_timer_task_hdl:3ffc9f04, prio:22, stack:2048
I (871) wifi: Init lldesc rx mblock:25
I (871) wifi: Init lldesc rx ampdu len mblock:7
I (871) wifi: Init lldesc rx ampdu entry mblock:4
I (871) wifi: pp_task_hdl : 3ffd6dbc, prio:23, stack:8192
ap_ssid:ZONCARE_03ae65
ap.ssid size:14
I (881) zq_wifi_test: Setting WiFi configuration AP SSID:ZONCARE_03ae65,STA SSID:TP-LINK_47E300
I (1551) wifi: mode : sta (24:0a:c4:03:ae:64) + softAP (24:0a:c4:03:ae:65)
I (1551) zq_wifi_test: SYSTEM_EVENT_STA_START

I (1551) zq_wifi_test: SYSTEM_EVENT_AP_START

I (1551) zq_wifi_test: socket_data_task......

waitbit1
I (1671) wifi: ap channel adjust o:6,1 n:1,1
I (1671) wifi: n:1 1, o:6 0, ap:1 1, sta:1 1, prof:6
I (2331) wifi: state: init -> auth (b0)
I (2331) wifi: state: auth -> assoc (0)
I (2331) wifi: mgtframe > 256 !

I (2341) wifi: state: assoc -> run (10)
I (2491) wifi: connected with TP-LINK_47E300, channel 1
I (2491) zq_wifi_test: SYSTEM_EVENT_STA_CONNECTED

I (3361) event: ip: 192.168.1.100 , mask: 255.255.255.0 , gw: 192.168.1.253 
I (3371) zq_wifi_test: SYSTEM_EVENT_STA_GOT_IP

ip:192.168.1.100 mask:255.255.255.0 gw:192.168.1.253 
waitbit2
I (1549391) wifi: n:1 0, o:1 1, ap:1 1, sta:1 0, prof:1
I (1551441) wifi: n:1 1, o:1 0, ap:1 1, sta:1 1, prof:1
start send
I (3813541) wifi: n:1 0, o:1 1, ap:1 1, sta:1 0, prof:1
[... 60 lines cut ...]
I (2305) wifi: state: assoc -> run (10)
I (2455) wifi: connected with TP-LINK_47E300, channel 1
I (2455) zq_wifi_test: SYSTEM_EVENT_STA_CONNECTED

I (3315) event: ip: 192.168.1.101 , mask: 255.255.255.0 , gw: 192.168.1.253 
I (3315) zq_wifi_test: SYSTEM_EVENT_STA_GOT_IP

ip:192.168.1.101 mask:255.255.255.0 gw:192.168.1.253 
waitbit2
start send
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400855c3  PS      : 0x00060b33  A0      : 0x80084800  A1      : 0x3ffd09d0  
A2      : 0x0000001d  A3      : 0x3ffd0a54  A4      : 0x00000004  A5      : 0x3ffbcce0  
A6      : 0x3ffd0a80  A7      : 0x00000000  A8      : 0x800833e0  A9      : 0x3ffd09c0  
A10     : 0x00000003  A11     : 0x00060b23  A12     : 0x00060b23  A13     : 0x3ffb6418  
A14     : 0x3ffb7c20  A15     : 0xfffffff8  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002d  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400855c3:0x3ffd09d0 0x40084800:0x3ffd09f0 0x40082d83:0x3ffd0a10 0x400edc71:0x3ffd0a50 0x400ee065:0x3ffd0a80 0x40107a58:0x3ffd0ab0 0x40107db5:0x3ffd0ad0 0x40107581:0x3ffd0b10 0x40107754:0x3ffd0b70 0x40104f0b:0x3ffd0ba0 0x40104f51:0x3ffd0bc0

Rebooting...
ets Jun  8 2016 00:22:57

```

projectgus, 2017-04-21:
Hi,

If you use "make monitor" then the full backtrace will be decoded automatically.

If using addr2line manually, paste the entire line after "Backtrace:" in the debug output and it will print the full trace.

Angus

long585, 2017-04-21:
How to use "make monitor"? 
```
long@long-virtual-machine:~/esp/esp-idf-v2.0-rc1/examples/zq_wcb_project$ make monitor
Makefile:8: /make/project.mk: 没有那个文件或目录
make: *** No rule to make target '/make/project.mk'。 停止。
```

projectgus, 2017-04-21:
How are you building your project at the moment? 

If make monitor isn't working, you can run:

```
xtensa-esp32-elf-addr2line -pfia -e zq_wifi_test.elf 0x400855c3:0x3ffd09d0 0x40084800:0x3ffd09f0 0x40082d83:0x3ffd0a10 0x400edc71:0x3ffd0a50 0x400ee065:0x3ffd0a80 0x40107a58:0x3ffd0ab0 0x40107db5:0x3ffd0ad0 0x40107581:0x3ffd0b10 0x40107754:0x3ffd0b70 0x40104f0b:0x3ffd0ba0 0x40104f51:0x3ffd0bc0
```

(All on one line)

Also, I just noticed you seem to be using IDF v2.0-rc1. A number of bugs were fixed between v2.0-rc1 and v2.0 final. Please upgrade your IDF to V2.0.

long585, 2017-04-21:
```
long@long-virtual-machine:~/esp/esp-idf-v2.0-rc1/examples/zq_wcb_project/build$ xtensa-esp32-elf-addr2line -pfia -e zq_wifi_test.elf 0x400855c3:0x3ffd09e0 0x40084800:0x3ffd0a00 0x40082d83:0x3ffd0a20 0x400edc71:0x3ffd0a60 0x400ee065:0x3ffd0a90 0x40107a58:0x3ffd0ac0 0x40107db5:0x3ffd0ae0 0x40107581:0x3ffd0b20 0x40107754:0x3ffd0b80 0x40104f0b:0x3ffd0bb0 0x40104f51:0x3ffd0bd0
0x400855c3: uxListRemove at /home/long/esp/esp-idf-v2.0-rc1/components/freertos/./list.c:212
0x40084800: xPortGetCoreID at /home/long/esp/esp-idf-v2.0-rc1/components/freertos/./tasks.c:3414
 (inlined by) xTaskRemoveFromEventList at /home/long/esp/esp-idf-v2.0-rc1/components/freertos/./tasks.c:3019
0x40082d83: xQueueGenericSend at /home/long/esp/esp-idf-v2.0-rc1/components/freertos/./queue.c:2040
0x400edc71: sys_mbox_post at /home/long/esp/esp-idf-v2.0-rc1/components/lwip/port/freertos/sys_arch.c:563
0x400ee065: tcpip_send_api_msg at /home/long/esp/esp-idf-v2.0-rc1/components/lwip/api/tcpip.c:476
0x40107a58: tcpip_apimsg at /home/long/esp/esp-idf-v2.0-rc1/components/lwip/api/api_lib.c:659
0x40107db5: netconn_send at /home/long/esp/esp-idf-v2.0-rc1/components/lwip/api/api_lib.c:659
0x40107581: lwip_sendto at /home/long/esp/esp-idf-v2.0-rc1/components/lwip/api/sockets.c:3174
0x40107754: lwip_sendto_r at /home/long/esp/esp-idf-v2.0-rc1/components/lwip/api/sockets.c:3174
0x40104f0b: my_udp_send at /home/long/esp/esp-idf-v2.0-rc1/examples/zq_wcb_project/main/./test.c:70
0x40104f51: socket_data_task_test at /home/long/esp/esp-idf-v2.0-rc1/examples/zq_wcb_project/main/./test.c:126 (discriminator 1)
```

projectgus, 2017-04-21:
Thanks.

Can you confirm if you're still using esp-idf v2.0 RC1?

long585, 2017-04-21:
The new verison is ESP-IDF v2.0-rc1-597-gf73c6f8 ?

projectgus, 2017-04-21:
Try `git checkout v2.0` - https://github.com/espressif/esp-idf/releases/tag/v2.0

long585, 2017-04-21:
This new version of the code is currently no issue, I would like to further continuous testing.
Thanks, @projectgus .

FayeY, 2017-05-03:
Hi long585, is this problem solved? Can we close this issue?

long585, 2017-05-03:
This problem has been solved,you can close it.

closed 2017-05-03 by FayeY