espressif/esp-idf #529: eclipse : project in .cpp build project also with error

sfranchin63, 2017-04-18:
Hi,
I tried to build a project in cpp. Eclipse tells me that there are some mistakes, but the project is ok and I can download it. How can these errors disappear?
[image]

projectgus, 2017-04-20:
Hi @sfranchin63 ,

It looks like Eclipse may not be configured correctly to find all of the header paths in your system. When this is the case, the build succeeds but the code editor shows errors like this. If you scroll to the top of the file, does it mark the `#include` statements at the top as being errors, also?

If this looks like the problem, make sure to very carefully follow the steps to set up the Eclipse project in Windows, as shown here:
http://esp-idf.readthedocs.io/en/latest/eclipse-setup-windows.html

Some third party tutorials are based on an older version of these steps, and following their steps will not correctly find all search paths.

After checking all the project properties, run Project -> Clean then Project -> Build to allow Eclipse to regenerate its internal cache of search paths.

igrr, 2017-05-03:
Hi @sfranchin63, did the above suggestion help you resolve the issue?

closed 2017-06-19 by FayeY