espressif/esp-idf #530: Crash in WiFI RX path

rojer, 2017-04-19:
```
mg_call              0x3ffb45fc after user flags=0 rmbl=86 smbl=1024
mg_call              0x3ffb45fc after proto flags=0 rmbl=0 smbl=1024
mg_mgr_handle_conn   0x3ffb45fc after fd=7 nc_flags=0 rmbl=0 smbl=1024
mg_mgr_handle_conn   0x3ffb45fc fd=7 fd_flags=2 nc_flags=0 rmbl=0 smbl=1024
mg_write_to_socket   0x3ffb45fc 1024 bytes -> 7
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400dd1d8  PS      : 0x00060b30  A0      : 0x800dd634  A1      : 0x3ffcc130  
A2      : 0x3ffd1bb4  A3      : 0x3ffc25a0  A4      : 0x3ffb3ce0  A5      : 0x3ffb3bd0  
A6      : 0x3ffc25a0  A7      : 0x0000101c  A8      : 0x00000000  A9      : 0x3ffcc100  
A10     : 0x3ffb48fc  A11     : 0x3ffcc126  A12     : 0x00000004  A13     : 0x3ffb4908  
A14     : 0x01c0344f  A15     : 0x01c0344f  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

Backtrace: 0x400dd1d8:0x3ffcc130 0x400dd634:0x3ffcc150 0x400ddf4c:0x3ffcc220 0x400925c8:0x3ffcc240 0x40093f6a:0x3ffcc290
...
```

```
(gdb) bt
#0  0x400dd1d8 in sta_rx_eapol ()
#1  0x400dd634 in sta_input ()
#2  0x400ddf4c in sta_rx_cb ()
#3  0x400925c8 in ppRxPkt ()
#4  0x40093f6a in ppTask ()

Dump of assembler code for function sta_rx_eapol:
   0x400dd1cc <+0>:     entry   a1, 32
   0x400dd1cf <+3>:     l32r    a8, 0x400d27ec
   0x400dd1d2 <+6>:     l32i.n  a10, a4, 4
   0x400dd1d4 <+8>:     l32i.n  a8, a8, 0
   0x400dd1d6 <+10>:    l32i.n  a10, a10, 4
   0x400dd1d8 <+12>:    l32i.n  a9, a8, 4     <=========================== 
   0x400dd1da <+14>:    beqz.n  a9, 0x400dd204 <sta_rx_eapol+56>
   0x400dd1dc <+16>:    l32i.n  a8, a8, 8
```

whatever `a8` is supposed to point at, is `NULL` and the code does not expect it.

projectgus, 2017-04-20:
Hi @rojer,

Do you have any more details about the circumstances where this bug is triggered?

g_misc_nvs gets set by `esp_wifi_init()`, and it shouldn't ever be set back to NULL unless `esp_wifi_deinit()` is called and successfully deinitialises WiFi. Is it possible this `esp_wifi_deinit` was called before the bug triggered?

If not, this might be memory corruption (something like a buffer overrun writing to whatever is before g_misc_nvs, causing it to be overwritten). It may be worth checking what is being linked directly before g_misc_nvs in `.bss`.

If you are using a JTAG debugger, you can set a store watchpoint on g_misc_nvs (ideally after esp_wifi_init() completes) to have the system halt when it is being set to NULL.

If you don't have a JTAG setup, you may be able to get away with this:

```
esp_wifi_init();
...
extern void *g_misc_nvs;
esp_set_watchpoint(0, g_misc_nvs, 4, ESP_WATCHPOINT_STORE);
```

Although this approach has the disadvantage that it will probably change the linker order, so g_misc_nvs may no longer be linked in the same place relative to other symbols.

(Watchpoints are per-CPU, so in both cases have to be set on both CPUs to be sure of catching the errant write.)

Angus

rojer, 2017-04-20:
hm, it definitely gets nulled out. i don't see obvious damage around it... will figure out something to set watchpoints with.

rojer, 2017-04-20:
it turned out to be task's stack overflow.
sorry, false alarm!

closed 2017-04-20 by rojer

rojer, 2017-04-20:
fwiw, it was #534 - allocated a 4K stack variable and BOOM.

projectgus, 2017-04-20:
Aha, glad you worked that one out.

There is also a useful option in menuconfig, it's turned off by default, under FreeRTOS "Set a debug watchpoint as a stack overflow check". This will cause an immediate unhandled debug exception if the task overflows its stack size. (The difference between this and the canary checker is that it's instant, whereas the canary checker won't see the crash until the next context switch happens on the that core.)

Angus

projectgus, 2017-04-20:
Out of interest, was it statically allocated task memory?

rojer, 2017-04-20:
how many watchpoints does esp32 support? there are about 10 tasks already in the system, if i'm not mistaken.

it was [this line](https://github.com/cesanta/mongoose/blob/7c0493071f182b890f4e01ece84ab2523858cc76/mongoose.c#L6691)

projectgus, 2017-04-20:
Only 2 watchpoints per CPU. The feature I mentioned uses watchpoint 1 in all cases, during a context switch the watchpont is moved to the end of the new task's stack.

Reason I was asking about static vs dynamically allocated tasks is that when a task's memory is dynamically allocated (the default/normal way via xTaskCreate()) the task's stack is allocated from the heap via malloc. It seemed unusual to me that an overflow in heap memory corrupted something in .bss (which is before heap in the address space.) But it's not impossible of course, once you corrupt one bit of memory all bets are off!

rojer, 2017-04-20:
ah, so watchpoint moves with the context. clever! i should definitely enable that.

bss is before heap, if task stack was allocated near the start of the heap, then smashing through its bottom would clobber bss. in this case overrun was also pretty large - by more than 1K.

projectgus, 2017-04-21:
Ah yeah, of course! My mistake.