espressif/esp-idf #533: SPI 3 Wire Mode and Half Duplex Do Not Work

ccceeefff, 2017-04-20:
When we set a SPI device with SPI_DEVICE_HALFDUPLEX and SPI_DEVICE_3WIRE, it still transmits command, address and tx_data on the MOSI pin. Meanwhile, the TX and RX phase happens on the same clock cycles rather than in TX then RX as expected for half-duplex mode.

`
// setting device config for 3-wire and half-duplex
spi_device_interface_config_t cfg = {
   .flags = (SPI_DEVICE_3WIRE | SPI_DEVICE_HALFDUPLEX)
};

// creating a fixed transaction for send/recv
spi_transaction_t trans = {
   .flags = (SPI_TRANS_USE_RXDATA | SPI_TRANS_USE_TXDATA),
   .length = 8,
   .rxlength = 8
};
trans.tx_data[0] = 0x0F;

// send out and watch pins on oscilloscope
spi_device_transmit(handle, &trans);
`

Is this not expected to work as of now? Any timeline when it should work or reason why its not supported yet?

ccceeefff, 2017-04-21:
I was able to get 3-wire SPI to work with two small fixes in the spi_master driver:

1. Use `spi_d` for both Tx and Rx. This is documented in the Technical Reference Manual but mistakenly noted in spi_master.h. According to the header, `spi_q` is used for 3-wire SPI.
2. In `spi_bus_initialize`, initializing the MOSI pin with the GPIO matrix should set the direction to `GPIO_MODE_INPUT_OUTPUT` such that slave responses are properly routed back to the internal SPI line.

Extra documentation or examples using the command/address bits would also help. For MSB transactions, it wasn't apparent that you have to shift all values to the top bits of the register for them to shift out properly.

Spritetm, 2017-04-24:
Gotcha. Working on SPI stuff now, I'll add the fix as well.

closed 2017-04-29 by igrr

nsamala, 2019-02-18:
@ccceeefff do you have an example of 3 wire SPI that you'd be able to share? Thanks for any help!