espressif/esp-idf #535: Crash in iBeacon scanner, didn't happen with previous versions

neilyoung, 2017-04-21:
I'm appearing a repeatable crash with a little script I wrote a couple of weeks ago and which didn't crash before. After updating to the latest release it crashes.

```

I (31) boot: ESP-IDF v2.0-rc1-597-gf73c6f8 2nd stage bootloader
I (31) boot: compile time 13:06:17
I (58) boot: Enabling RNG early entropy source...
I (58) boot: SPI Speed      : 40MHz
I (59) boot: SPI Mode       : DIO
I (59) boot: SPI Flash Size : 4MB
I (69) boot: Partition Table:
I (77) boot: ## Label            Usage          Type ST Offset   Length
I (96) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (116) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (137) boot:  2 factory          factory app      00 00 00010000 00100000
I (157) boot: End of partition table
I (167) boot: Disabling RNG early entropy source...
I (181) boot: Loading app partition at offset 00010000
I (1448) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
I (1449) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x294d0 (169168) map
I (1460) boot: segment 2: paddr=0x000494e0 vaddr=0x3ffc0000 size=0x02754 ( 10068) load
I (1488) boot: segment 3: paddr=0x0004bc3c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at ??:?

I (1507) boot: segment 4: paddr=0x0004c044 vaddr=0x40080400 size=0x18940 (100672) load
I (1578) boot: segment 5: paddr=0x0006498c vaddr=0x400c0000 size=0x00000 (     0) load
I (1579) boot: segment 6: paddr=0x00064994 vaddr=0x00000000 size=0x0b674 ( 46708) 
I (1590) boot: segment 7: paddr=0x00070010 vaddr=0x400d0018 size=0x7ae74 (503412) map
0x400d0018: _stext at ??:?

I (1614) cpu_start: Pro cpu up.
I (1622) cpu_start: Starting app cpu, entry point is 0x40080df0
0x40080df0: call_start_cpu1 at /Users/me/Documents/ESP-32/esp-idf/components/esp32/./cpu_start.c:171

I (0) cpu_start: App cpu up.
I (1649) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1668) heap_alloc_caps: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1686) heap_alloc_caps: At 3FFCEE90 len 00011170 (68 KiB): DRAM
I (1704) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1722) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1741) heap_alloc_caps: At 40098D40 len 000072C0 (28 KiB): IRAM
I (1759) cpu_start: Pro cpu start user code
I (1816) cpu_start: Starting scheduler on PRO CPU.
I (186) cpu_start: Starting scheduler on APP CPU.
BTDM CONTROLLER VERSION: 010101
btip start
copy .data from 4000d890 to 3ffae6e0, len 00001830
set .bss 0x0 from 3ffb8000 to 3ffbff70, len 00007f70
BTDM ROM VERSION 0101
BD_ADDR: 24:0A:C4:05:98:8E
NVDS MAGIC FAILED
RF Init OK with coex
I (386) phy: phy_version: 350, Mar 22 2017, 15:02:06, 1, 0
Enable Classic BT
Enable Low Energy
E (406) BT: Startup BTU

E (406) BT: GATTS_CreateService:  handle of service handle1
E (426) BT: bta_dm_co_ble_load_local_keys: func not ported

I (436) BLE: Wait for scans...
I (1636) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -74, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D, MAJ 997, MIN 64763, TXPWR -60
I (2246) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -68, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D, MAJ 997, MIN 64763, TXPWR -60
I (2846) BLE: BDA C1:3F:FC:FB:03:E5, RSSI -67, UUID B9407F30-F5F8-466E-AFF9-25556B57FE6D, MAJ 997, MIN 64763, TXPWR -60
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x40086eb9  PS      : 0x00060034  A0      : 0x80004160  A1      : 0x3ffc0530  
0x40086eb9: ram_chip_i2c_readReg at ??:?

A2      : 0x00000062  A3      : 0x00000001  A4      : 0x00000005  A5      : 0x3ffd59f0  
A6      : 0x3ffae270  A7      : 0x0000c260  A8      : 0xbad00bad  A9      : 0x3f400f28  
A10     : 0x00000002  A11     : 0xffffffff  A12     : 0x800987f3  A13     : 0x3ffd59c0  
A14     : 0x3ffae270  A15     : 0x00000028  SAR     : 0x0000001b  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x40086eb9:0x3ffc0530 0x40004160:0x3ffc0550 0x400041dd:0x3ffc0570 0x40088251:0x3ffc0590 0x40088348:0x3ffc05b0 0x40083a0a:0x3ffc05e0 0x4008181d:0x3ffc0600
0x40086eb9: ram_chip_i2c_readReg at ??:?

0x40088251: pll_correct_dcap at ??:?

0x40088348: bt_track_pll_cap at ??:?

0x40083a0a: btdm_bb_isr at intc.c:?

0x4008181d: _xt_lowint1 at xtensa_vectors.o:?

Rebooting...
ets Jun  8 2016 00:22:57

```

Here is the script. It starts a 60 s scans and explicitly filters iBeacons. If I use the default scan_interval/scan_window 0x50/0x30, it works somehow, but it doesn't crash. Changing both to anything else crashes.

```
#include <stdio.h>

#include <stdlib.h>
#include <string.h>

#include "bt.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "freertos/FreeRTOS.h"
#include "freertos/event_groups.h"
#include "freertos/task.h"
#include "nvs_flash.h"
#include "sdkconfig.h"

#include "esp_blufi_api.h"
#include "esp_bt_defs.h"
#include "esp_bt_device.h"
#include "esp_bt_main.h"
#include "esp_gap_ble_api.h"

extern void bt_task(void *ignore);

static const char tag[] = "BLE";

static uint8_t ibeacon_prefix[] = {
    0x02, 0x01, 0x00, 0x1A, 0xFF, 0x4C, 0x00, 0x02, 0x15};

void gap_event_handler(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param) {
    esp_ble_gap_cb_param_t *p = (esp_ble_gap_cb_param_t *)param;

    if (p->scan_rst.search_evt == ESP_GAP_SEARCH_INQ_RES_EVT) {
        // Check for iBeacon adv prefix. Ignore 3rd byte, this varies from beacon type to beacon type
        p->scan_rst.ble_adv[2] = 0x00;
        for (int i = 0; i < sizeof(ibeacon_prefix); i++) {
            if (p->scan_rst.ble_adv[i] != ibeacon_prefix[i]) {
                return;
            }
        }

        ESP_LOGI(tag, "BDA %02X:%02X:%02X:%02X:%02X:%02X, RSSI %d, UUID %02X%02X%02X%02X-%02X%02X-%02X%02X-%02X%02X-%02X%02X%02X%02X%02X%02X, MAJ %d, MIN %d, TXPWR %d",
                 p->scan_rst.bda[0],
                 p->scan_rst.bda[1],
                 p->scan_rst.bda[2],
                 p->scan_rst.bda[3],
                 p->scan_rst.bda[4],
                 p->scan_rst.bda[5],

                 p->scan_rst.rssi,

                 p->scan_rst.ble_adv[9],
                 p->scan_rst.ble_adv[10],
                 p->scan_rst.ble_adv[11],
                 p->scan_rst.ble_adv[12],

                 p->scan_rst.ble_adv[13],
                 p->scan_rst.ble_adv[14],

                 p->scan_rst.ble_adv[15],
                 p->scan_rst.ble_adv[16],

                 p->scan_rst.ble_adv[17],
                 p->scan_rst.ble_adv[18],

                 p->scan_rst.ble_adv[19],
                 p->scan_rst.ble_adv[20],
                 p->scan_rst.ble_adv[21],
                 p->scan_rst.ble_adv[22],
                 p->scan_rst.ble_adv[23],
                 p->scan_rst.ble_adv[24],

                 (p->scan_rst.ble_adv[25] << 8) | p->scan_rst.ble_adv[26],

                 (p->scan_rst.ble_adv[27] << 8) | p->scan_rst.ble_adv[28],

                 p->scan_rst.ble_adv[29] - 256

                 );
    }
}

void bt_task(void *ignore) {
    esp_err_t ret;
    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    esp_bt_controller_init(&bt_cfg);

    ret = esp_bt_controller_enable(ESP_BT_MODE_BTDM);
    if (ret) {
        ESP_LOGE(tag, "%s enable bt controller failed\n", __func__);
        goto end;
    }

    ret = esp_bluedroid_init();
    if (ret != ESP_OK) {
        ESP_LOGE(tag, "%s init bluedroid failed\n", __func__);
        goto end;
    }

    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(tag, "%s enable bluedroid failed\n", __func__);
        goto end;
    }
    ret = esp_ble_gap_register_callback(gap_event_handler);
    if (ret != ESP_OK) {
        ESP_LOGE(tag, "esp_ble_gap_register_callback: rc=%d", ret);
        goto end;
    }

    static esp_ble_scan_params_t ble_scan_params = {
        .scan_type = BLE_SCAN_TYPE_ACTIVE,
        .own_addr_type = ESP_PUBLIC_ADDR,
        .scan_filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
        .scan_interval = 0x100,//0x50,
        .scan_window = 0x100};//0x30};

    ret = esp_ble_gap_set_scan_params(&ble_scan_params);
    if (ret != ESP_OK) {
        ESP_LOGE(tag, "esp_ble_gap_set_scan_params: rc=%d", ret);
        goto end;
    }

    ret = esp_ble_gap_start_scanning(60);
    if (ret != ESP_OK) {
        ESP_LOGE(tag, "esp_ble_gap_start_scanning: rc=%d", ret);
        goto end;
    }
    ESP_LOGI(tag, "Wait for scans...");
end:
    vTaskDelete(NULL);
}

void app_main(void) {
	nvs_flash_init();
    xTaskCreatePinnedToCore(&bt_task, "btTask", 2048, NULL, 5, NULL, 0);
}

```

neilyoung, 2017-04-21:
Uhm... I think the active scan was the problem. Changed to passive. No crash

neilyoung, 2017-04-21:
No, that doesn't make the difference. The crash is persistent.

negativekelvin, 2017-04-21:
https://github.com/espressif/esp-idf/issues/520

neilyoung, 2017-04-22:
@negativekelvin Yep, looks similar.

closed 2017-04-25 by igrr