espressif/esp-idf #536: Max Rate of UDP packets

filipeaguiar22, 2017-04-22:
Hi everyone,
 I am trying to send real time audio through WiFi Router, using UDP.
    Disabling encryption in the Access Point doubled the rate of my package rate.
    Currently I can send one package every 20ms before I get an error saying I am out of memory.
    The pseudo code is:
      
     while 1:
           send UDP package (8 bytes payload)
           wait (20ms)
    
I found a reference (below) that the ESP8266 is limited and the ES32 is an evolution of it.
     Any information or direction would be greatly appreciated.
    Kind regards, 
   
Filipe

> In the official documentation 4A-AT-Espressif AT Instruction Set_020.pdf under the CIPSEND command they claim:
Enters unvarnished transmission, 20ms interval between each packet, maximum 2048 bytes per packet. . -from http://www.esp8266.com/viewtopic.php?f=14&t=2861

liuzfesp, 2017-05-10:
Hi @filipeaguiar22 do you mean send one UDP packet every 20ms?  Moreover, could you provide following information:
1. How much free memory is left after your application is startup?
2. Do you use buffer cache in application, e.g.  taskA keeps sending 1 UDP per 20ms to taskB, taskB buffered the packet and try it's best to send the UDP packets, once the sending socket api detects out of memory error, then taskB start the flow control, such as delay 1ms etc and resend the dropped packets.
3. What the maximum real time audio rate?
In IDF2.0, all tx buffer is allocated dynamically, the maximum dynamically allocated tx buffer number is 32 (you can configure it via menuconfig), for UDP, if sending too fast, it will have out of memory memory, of course here in your example, 1 packet per 20ms it not fast, however, it maybe out of memory in some scenario, such as when the channel is busying etc

FayeY, 2017-06-19:
Hi @filipeaguiar22 , is there any update? Can we close this issue now?

closed 2017-07-05 by FayeY

filipeaguiar22, 2017-07-05:
Hi FayeY. 
  Sorry for the absence. I had changed for C  to achieve higher speeds. 
  Thanks for the help.