espressif/esp-idf #538: I2S RX 32bit mode

jaracil, 2017-04-23:
Hi.
I2S Driver is broken for 32 bit mode.
See technical book (section 8.4.5 page 163) :
 https://espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf

You can see four posible RX_I2S_FIFO_MOD:

0 16-bit dual channel data
1 16-bit single channel data
2 32-bit dual channel data
3 32-bit single channel data

... but I2S Driver only uses 0 and 1 modes.
`I2S[i2s_num]->fifo_conf.rx_fifo_mod = i2s_config->channel_format < I2S_CHANNEL_FMT_ONLY_RIGHT ? 0 : 1;`

projectgus, 2017-04-25:
Hi @jaracil ,

Thanks for bringing this up. A fix for this in review now, and should be merged soon. Will let you know.

Angus

igrr, 2017-05-03:
Fixed in 7db85a38.

closed 2017-05-03 by igrr