espressif/esp-idf #539: Expected toolchain crosstool-ng-1.22.0-61-gab8375a not available in Espressif crosstool-NG repo

jmattsson, 2017-04-24:
Trying to build on the ESP-IDF currently gets me:
```
WARNING: Toolchain version is not supported: crosstool-NG crosstool-ng-1.22.0-61-gab8375a9
Expected to see version: crosstool-NG crosstool-ng-1.22.0-61-gab8375a
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
```
and then the build promptly fails.

This despite doing a fresh clone and build of https://github.com/espressif/crosstool-NG.git. Presumably I could download the precompiled binaries, but my build pipeline is all set up do to it from source, and the docs still say that's a valid option. Could someone pretty please either update the docs or, ideally, the ct-ng repo?

igrr, 2017-04-24:
What commit ID are you getting instead of the one which IDF expects? Are you using xtensa-1.22.x branch?

jmattsson, 2017-04-24:
I still get `xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-61-gab8375a9) 5.2.0` on a freshly built toolchain.

Ah, wait, the expected version is just missing the trailing `9`!

jmattsson, 2017-04-24:
(Or, alternatively, the toolchain has one character too many in its build id)

jmattsson, 2017-04-24:
Apologies for the noise, things do build with the freshly built toolchain. I must've gotten my PATHs crossed somehow.

closed 2017-04-24 by jmattsson