espressif/esp-idf #542: Wake Up from Touch

andriyadi, 2017-04-26:
I tried [this example](https://github.com/espressif/esp-idf/blob/b540322dc10ee2a0ce773086da42627468897325/examples/system/deep_sleep/main/deep_sleep_example_main.c) without any changes, using some devboards I have, but with no luck. 
Wake up from timer and GPIO, all work, but no luck with touch.

Any advice?

igrr, 2017-04-26:
What kind of circuit are you using to connect the touch pad?
The example will also print some touch readings during calibration, can you paste the relevant part of the log here?

andriyadi, 2017-04-26:
No specific circuit. Direct connection from touch pin/pad to ESP-WROOM-32. 

I saw the code is doing some calibration. CMIIW, it expects the average touch reading to be above 300. In my case, average touch reading (when not touched) on the pin I use is above 1200. And when touched the reading is under 200. So I believe it meets the threshold.

igrr, 2017-04-26:
This indeed is a regression. Looking into this.

andriyadi, 2017-04-26:
I  just had access to my computer, here's the log requested:

```
Enabling timer wakeup, 30s
Enabling EXT1 wakeup on pins GPIO25, GPIO26
Touch pad #8 average: 1174, wakeup threshold set to 1074.
Touch pad #9 average: 897, wakeup threshold set to 797.
Enabling touch pad wakeup
Entering deep sleep
```

None of touch #8 or #9 that causes wake up from deep sleep.
Please take a look. Thanks!

igrr, 2017-04-26:
As a temporary workaround, you may roll back to v2.0 release.

We're looking into this, but the issue looks pretty odd, may take some time to solve.

igrr, 2017-04-27:
Ok, found the issue, will have the fix in master shortly. For now you can fix it locally:

```patch
From 1c3b40adeb1de0b18575d5419bfc61264b6ec733 Mon Sep 17 00:00:00 2001
From: Ivan Grokhotkov <ivan@espressif.com>
Date: Thu, 27 Apr 2017 12:42:56 +0800
Subject: [PATCH] esp32: select 8M clock as RTC_FAST_CLK on startup

Even though RTC_CLK_CONFIG_DEFAULT correctly had RTC_FAST_FREQ_8M as the
fast clock, the bootloader modified fast_freq field to the currently
selected RTC_FAST_CLK (so that the clock choice is not affected by the
bootloader). The application startup code never switched to 8M clock,
which caused the default (XTAL/4) to be used as RTC_FAST_CLK. This had
caused a number of issues, such as touch pads not working in deep sleep.

Fixes https://github.com/espressif/esp-idf/issues/542.
Ref TW12053.
---
 components/esp32/clk.c | 1 +
 1 file changed, 1 insertion(+)

diff --git a/components/esp32/clk.c b/components/esp32/clk.c
index 449ea39..ce7b580 100644
--- a/components/esp32/clk.c
+++ b/components/esp32/clk.c
@@ -43,6 +43,7 @@ void esp_clk_init(void)
 {
     rtc_config_t cfg = RTC_CONFIG_DEFAULT();
     rtc_init(cfg);
+    rtc_clk_fast_freq_set(RTC_FAST_FREQ_8M);
 
 #ifdef CONFIG_ESP32_RTC_CLOCK_SOURCE_EXTERNAL_CRYSTAL
     select_rtc_slow_clk(RTC_SLOW_FREQ_32K_XTAL);
-- 
2.10.1 (Apple Git-78)
```

andriyadi, 2017-04-28:
@igrr Superb! I can confirm it working flawlessly. Super thanks!

Side note, previously touchRead on each pad is around 50-100. Now, above 800. Is it affected by that change?

igrr, 2017-04-28:
Your previous comment said

> average touch reading to be above 300. In my case, average touch reading (when not touched) on the pin I use is above 1200. And when touched the reading is under 200.

Could you clarify in which case you saw 50-100? For 1cm^2 sized touch pads I'm getting ~1200 both with old and new code.

andriyadi, 2017-04-28:
That's before applying your patch. My touch pad is like the picture below, about 2x1.5cm. I have another custom board, with similar size pad, the reading is around the same, again that's before the patch. 

[image],

andriyadi, 2017-04-28:
By the way, that reading difference that I experienced is not really a concern, though. You could close this issue. Again, thanks for the fix.

closed 2017-04-29 by igrr