espressif/esp-idf #543: Unit Test App on arduino-esp32

pacm93, 2017-04-26:
Hi! I develop for the ESP32 using the Arduino core. I wanted to take the unit test component from esp-idf and use it there. Is this feasible? I don't have much idea from where I should start.

igrr, 2017-04-27:
There are two issues I see with this.

First, the unit test component is not being compiled along with other ESP-IDF components when @me-no-dev builds the binary libraries for the Arduino core. So there is no straightforward way.
If @me-no-dev would add tools/unit-test-app/components/ to the list of component directories in the Arduino project, then you would essentially need to replicate the contents of tools/unit-test-app/components/main in your Arduino sketch.

Second, our unit test component relies on some linker flags to ensure that all individual test files are included into the final application. Because individual test functions are not referenced directly from the main program, linker will not include corresponding object files into the final output. In IDF, we solve this problem by adding `--whole-archive`/ `-no-whole-archive` linker flags around each archive file which contains unit tests ([example](https://github.com/espressif/esp-idf/blob/master/components/app_update/test/component.mk#L5)). I don't know how to do something equivalent in Arduino.
A workaround would be to define some symbol in each file with tests (e.g. an empty function, `void include_test_file_xxx() { }`) and then reference each of these functions from the main sketch:
```c++
extern "C" void include_test_file_xxx();
extern "C" void include_test_file_yyy();
extern "C" void include_test_file_zzz();

void setup() {
  include_test_file_xxx();
  include_test_file_yyy();
  include_test_file_zzz();
}
```
This will force linker to include each test object file into the final application, along with the actual tests.

If you go for an option of using Arduino core as an ESP-IDF component, you don't have either of these issues — just add $IDF_PATH/tools/unit-test-app/components to the list of component directories of your project, and write unit tests in the same way this is done in ESP-IDF components.

pacm93, 2017-04-27:
@igrr Thank you very much! I think I will go for your suggestion.

closed 2017-04-28 by igrr