espressif/esp-idf #544: CAN Driver

saeugetier, 2017-04-26:
Hi devs,

are there any plans to support a CAN driver in the ESP-IDF. I saw there is no describtion of the CAN interface in the ESP32 technical refence. But for example Simba RTOS has implemented the CAN interface.

zelll, 2017-04-26:
There isn't a CAN driver in official ESP-IDF, but it seems someone has made one: https://www.esp32.com/viewtopic.php?f=12&t=380

igrr, 2017-04-27:
@ThomasBarth has written a CAN driver for the ESP32, it is available as a component here:
https://github.com/ThomasBarth/ESP32-CAN-Driver/

I'll keep this issue open, but for now we don't have near term plans for a CAN driver.

jolivepetrus, 2017-05-11:
A minimal CAN driver is available in Lua RTOS:

https://github.com/whitecatboard/Lua-RTOS-ESP32/blob/master/components/lua_rtos/drivers/can.c
https://github.com/whitecatboard/Lua-RTOS-ESP32/tree/master/components/can

https://github.com/whitecatboard/Lua-RTOS-ESP32/wiki/CAN-module

closed 2018-07-12 by igrr

snahmad, 2018-07-28:
is can driver is available now officially?

commit 1d2727f4 (2018-07-12): CAN Driver

The following commit contains the first version of the ESP32 CAN Driver.

closes #544