espressif/esp-idf #546: Wifi PA

lmagalhaes1337, 2017-04-27:
Is it possible to turn off the PA while scanning for APs? Better yet, is it possible to turn off the PA completly before starting the Wifi module?

Thanks,

TimXia, 2017-05-03:
@lmagalhaes1337 Before starting the WiFi module, the PA is off. It is turned on when calling esp_wifi_start(). You can call esp_wifi_stop() to turn off the PA while scanning for APs.

lmagalhaes1337, 2017-05-04:
Can you please explain the sequence of commands to do that? I've tried a couple of things and they haven't worked.

```
wifi_scan_config_t scanConfig;
	scanConfig.ssid = 0;
		scanConfig.bssid = 0;
		scanConfig.channel = 0;
		scanConfig.show_hidden = true;
		//scanConfig.scan_type = WIFI_SCAN_TYPE_PASSIVE;
```
Using the passive scan type crashes the CPU.

```
ESP_ERROR_CHECK(esp_wifi_scan_start(&scanConfig, false));
		ESP_ERROR_CHECK(esp_wifi_stop());
		vTaskDelay(1000/portTICK_RATE_MS); // free the CPU for 100ms

		// get the list of APs found in the last scan
			ESP_ERROR_CHECK(esp_wifi_scan_get_ap_records(&apNum, apRecords));
```
This will give an error saying that the wifi is not enabled to get the ap records. Adding a ESP_ERROR_CHECK(esp_wifi_start()); line before the get ap records routine will make the code work but not output any AP. Care to help?

Thanks,

Luis

TimXia, 2017-05-05:
@lmagalhaes1337 Sorry for some explaining mistake. The PA can not be turned off when scanning for APs. If you want to do something with WiFi like scaning and connecting, you must call esp_wifi_start() first. If you do not need WiFi any more, call esp_wifi_stop() please.

TimXia, 2017-05-08:
@lmagalhaes1337 Do you mean that esp32 does not send probe request when scan for APs?

lmagalhaes1337, 2017-05-08:
No, I was just asking if it was possible to turn off the PA while scanning for APs.

TimXia, 2017-05-08:
@lmagalhaes1337 If esp32 does not send frame, the PA is not turned on.

lmagalhaes1337, 2017-05-08:
But to send the frame, is it mandatory for the PA to be turned on?

TimXia, 2017-05-08:
@lmagalhaes1337 Yes.

jack0c, 2017-05-18:
@lmagalhaes1337 We study this problem again, we can provide this API, but not in idf2.1

TimXia, 2018-03-29:
@lmagalhaes1337 If ESP32 is sending data, PA is turned on mandatorily. After data is sent out, PA is turned off mandatorily.

closed 2018-04-04 by FayeY