espressif/esp-idf #550: Crash with SPI Master

berkutta, 2017-04-28:
I'm currently facing the issue that my ESP32 run's crazy after pressing the Reset 3-4times (even happens after uploading my code 3-4times and thus resetting the ESP). 

My hardware consists of no external Pullups. I've used the Sparkfun ESP32 Thing together with a RFM95.

You find the whole IDF project under the following URL: 
https://github.com/berkutta/esp32_lmic.git

"Bad Boot"
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3420
ho 0 tail 12 room 4
load:0x40078000,len:10376
ho 0 tail 12 room 4
load:0x40080000,len:252
entry 0x40080034
I (648) cpu_start: Pro cpu up.
I (648) cpu_start: Single core mode
I (649) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (661) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (682) heap_alloc_caps: At 3FFB5B70 len 0002A490 (169 KiB): DRAM
I (702) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (723) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (745) heap_alloc_caps: At 400933F8 len 0000CC08 (51 KiB): IRAM
I (765) cpu_start: Pro cpu start user code
D (803) clk: RTC_SLOW_CLK calibration value: 3519527
V (833) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (833) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (881) intr_alloc: Connected src 56 to int 2 (cpu 0)
V (929) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (985) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (1058) intr_alloc: Connected src 16 to int 3 (cpu 0)
V (1106) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1166) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (1242) intr_alloc: Connected src 24 to int 12 (cpu 0)
I (1291) cpu_start: Starting scheduler on PRO CPU.
D (1345) nvs: nvs_flash_init_custom start=9 count=6
Task
Fired IRQ1
E (1361) spi_master: spi_device_queue_trans(700): invalid dev handle
assertion "ret==ESP_OK" failed: file "/home/bmarty/esp/esp32_lmic/main/./hal.c", line 208, function: hal_spi
abort() was called at PC 0x400e5a1b on core 0
Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x40008155:0x3ffb7f90 0x40007d16:0x3ffb7fb0 0x400f9c30:0x3ffb7fe0 0x400f93d7:0x3ffb8030 0x400f98bf:0x3ffb8050 0x400f9bc5:0x3ffb8070 0x400f9d06:0x3ffb8090 0x400f9d1c:0x3ffb80b0 0x400f71e8:0x3ffb80d0 0x400f6f9e:0x3ffb80f0

Rebooting...
```

"Good Boot"
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3420
ho 0 tail 12 room 4
load:0x40078000,len:10376
ho 0 tail 12 room 4
load:0x40080000,len:252
entry 0x40080034
I (648) cpu_start: Pro cpu up.
I (648) cpu_start: Single core mode
I (649) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (661) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (681) heap_alloc_caps: At 3FFB5B70 len 0002A490 (169 KiB): DRAM
I (702) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (723) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (745) heap_alloc_caps: At 400933F8 len 0000CC08 (51 KiB): IRAM
I (765) cpu_start: Pro cpu start user code
D (802) clk: RTC_SLOW_CLK calibration value: 3494144
V (832) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (833) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (880) intr_alloc: Connected src 56 to int 2 (cpu 0)
V (929) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (985) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (1058) intr_alloc: Connected src 16 to int 3 (cpu 0)
V (1106) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1166) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (1242) intr_alloc: Connected src 24 to int 12 (cpu 0)
I (1291) cpu_start: Starting scheduler on PRO CPU.
D (1345) nvs: nvs_flash_init_custom start=9 count=6
Task
hal io init
I (1361) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1361) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1370) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1379) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1388) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Init SPI Bus
V (1399) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1405) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x80E
D (1414) intr_alloc: Connected src 30 to int 13 (cpu 0)
Add device to Bus
I (1421) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1431) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Read register
Got following Reg Version: 0x12whatLMIC RESET19186: Unknown event: 17Packet queuedrunTask
runFired IRQ0
Task
runTask
runTask
runFired IRQ1
Task
runTask
runFired IRQ1
Task
runTask
run269905: EV_TXCOMPLETE (includes waiting for RX windows)Task
run301165: Unknown event: 17Packet queuedTask
```

I'm not 100% sure if the issue is in IDF or my App Code. But because it runs first and after some resets not (without changing the SW) I'm quite sure that there is something really broken.

igrr, 2017-04-29:
Seems like SPI driver is returning some error code, and your application does an `assert` on the error code, which fails. Could you log the error code returned by the SPI driver?

```
assertion "ret==ESP_OK" failed: file "/home/bmarty/esp/esp32_lmic/main/./hal.c", line 208, function: hal_spi
```

negativekelvin, 2017-04-29:
Your bad log doesn't say Init SPI Bus or Add device to Bus so spi is not ready but something is trying to spi_device_queue_trans anyway.

igrr, 2017-05-03:
I don't see `hal_disableIRQs` calling `hal_io_check`, only `hal_enableIRQs` does. Or maybe @berkutta has updated the code already?

@berkutta is the issue still present?

berkutta, 2017-05-03:
@igrr At the end it was an issue with crude compiler options on my side. It now works, no issue from ESP-IDF. I will now close this.

closed 2017-05-03 by berkutta