espressif/esp-idf #551: SPI master sending wrong data to ILI9341-based display

inxonic, 2017-04-30:
Hi,

my application with an ILI9341-based display is broken when compiled with IDF after e9c372bc2dfec648fd3d06a5f96502b77d07bd58.

This can be reproduced with the spi_master example. I could capture this with an oscilloscope (at 100 kHz): 

Before: This is the [`CF 00 83 30`](https://github.com/espressif/esp-idf/blob/07b61d54f7e502bb12527e4b582de121ffa6fa02/examples/peripherals/spi_master/main/spi_master_example_main.c#L54):
[image]

After: This seems to be `CF 00 AD 6E`, don't know where it's coming from:
[image]

Kind regards
Daniel

negativekelvin, 2017-04-30:
Just a theory new spi driver can use DMA for smaller transfers so you may have to be careful where your buffers are placed so DMA can access them.

inxonic, 2017-05-01:
@negativekelvin your guess is correct. The initialization sequence (`static const ili_init_cmd_t ili_init_cmds`) probably gets placed into DROM. I could make it work again by defining the data as `DRAM_ATTR`.

negativekelvin, 2017-05-01:
@Spritetm could spi driver check ptr address before deciding to use DMA?

closed 2017-05-08 by igrr