espressif/esp-idf #557: attp_build_sr_msg() crashed when p_msg null

easion, 2017-05-02:
> **crash point:**

/*******************************************************************************
**
** Function         attp_build_sr_msg
**
** Description      Build ATT Server PDUs.
**
*******************************************************************************/
BT_HDR *attp_build_sr_msg(tGATT_TCB *p_tcb, UINT8 op_code, tGATT_SR_MSG *p_msg)
{
    BT_HDR          *p_cmd = NULL;
    UINT16          offset = 0;	

    switch (op_code) {
    case GATT_RSP_READ_BLOB:
    case GATT_RSP_PREPARE_WRITE:
        GATT_TRACE_EVENT ("ATT_RSP_READ_BLOB/GATT_RSP_PREPARE_WRITE: len = %d offset = %d",
                          p_msg->attr_value.len, p_msg->attr_value.offset);		
			offset = p_msg->attr_value.offset;
....
}

> **Call with:**
void bta_gatts_send_rsp (tBTA_GATTS_CB *p_cb, tBTA_GATTS_DATA *p_msg)
{
    UNUSED(p_cb);

    if (GATTS_SendRsp (p_msg->api_rsp.hdr.layer_specific,
                       p_msg->api_rsp.trans_id,
                       p_msg->api_rsp.status,
                       (tGATTS_RSP *)p_msg->api_rsp.p_rsp) != GATT_SUCCESS) {
        APPL_TRACE_ERROR("Sending response failed\n");
    }

}

xcguang, 2017-05-04:
Hi @easion ,
Does it crashed when you handler ESP_GATTS_WRITE_EVT?Maybe you should check param->write.is_prep,and you can see https://github.com/espressif/esp-idf/issues/418

heyinling, 2017-05-25:
@easion plz try on latest master. this bug is fixed now on my side.

FayeY, 2017-06-15:
Hi @easion , is this problem solved? Can we close this issue?

closed 2017-07-05 by FayeY