espressif/esp-idf #562: Crash in timer ISR

simon-jouet, 2017-05-02:
Hi Everyone,

So I received a few days ago some wroom32 modules and I've been playing a bit with them. I started porting the 3d printer firmware marlin to the ESP32 as I think it might be a great application, however I'm running in some problems (which isn't very unexpected).

Okay so I've modified the HAL of marlin and added the code to support the timers, this looks to be working okay, however, after a few ticks of the interrupt the ESP32 crashes, I'm pretty sure the issue is fairly straightforward but I'm struggling to find the root cause.

So the first things I did was to use the exception decoder to figure out where the issue was but the reported location doesn't make sense (could it be because it's an ISR?). I spent most of the evening setting up openocd-esp32 and a raspberry pi as a JTAG debugger but I can't get the breakpoints working, getting the same errors as described here https://github.com/espressif/openocd-esp32/issues/7. I spent the last hour or so setting up esp-idf and arduino as a component for it to try to enable more debug logs and get all the latest fixes but the issue remain the same.

Anyway below is the error I'm getting as reported by *make monitor*, would anybody have insight on how I can go about debugging these issues in the future and what I could be doing wrong with OpenOCD?

```
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: BREAK instr
Register dump:
PC      : 0x400ffb07  PS      : 0x00060036  A0      : 0x80083bf8  A1      : 0x3ffc0b00
0x400ffb07: _ZN11Temperature3isrEv at C:/msys32/home/simon/esp/marlin/main/temperature.cpp:442

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3ffc560c  A5      : 0x00000000
A6      : 0x3ffb1270  A7      : 0x3ffb1f14  A8      : 0xffffc001  A9      : 0x3ffc55b4
A10     : 0x00000020  A11     : 0x3ff44000  A12     : 0x800d11d2  A13     : 0x3ffb1160
A14     : 0x3ffc1088  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x400ffb07:0x3ffc0b00 0x40083bf8:0x3ffc0be0 0x40081e45:0x3ffc0c00
0x400ffb07: _ZN11Temperature3isrEv at C:/msys32/home/simon/esp/marlin/main/temperature.cpp:442

0x40083bf8: __timerISR at C:/msys32/home/simon/esp/marlin/components/arduino/cores/esp32/esp32-hal-timer.c:179

0x40081e45: _xt_lowint1 at xtensa_vectors.o:?

Rebooting...
```

Cheers,
Simon

igrr, 2017-05-02:
What is the contents of `Temperature::isr()`, could you post a gist? My guess is that you are calling a some libc function which tries to acquire a lock, which easily fails in an ISR. Another possible reason is an assert failing, although that normally results in an error message to be printed as well.

simon-jouet, 2017-05-03:
Hi @igrr 

Thanks might help with the debugging

So the code for temperature::isr() is here https://github.com/simon-jouet/Marlin/blob/HAL_ESP32/Marlin/temperature.cpp#L1479  as you can see it's not the prettiest code, that's why I was hoping to get OpenOCD or some traces to help figuring out the issue (and i'm sure, the many more to come).

The timer and interrupt handler calling this is defined in https://github.com/simon-jouet/Marlin/blob/HAL_ESP32/Marlin/src/HAL/HAL_ESP/HAL_timers_ESP.cpp (in my local version I moved the timerAttachInterrupt() from and associated code from HAL_timer_enable_interrupt into HAL_timer_start.

Anyway I'm sure you have better things to do that look into this, I just wanted to know if getting the *BREAK instr* exception and struggling with OpenOCD for debugging was reasonable :)
I also tried to enable the gdbstub on crash but surprisingly it didn't work out, the chip just rebooted.

I guess I will have to go the old way toggling a pin.

Nonetheless great job on the ESP32, there are still few things to iron out but the it's looking like a great chip!

Spritetm, 2017-05-03:
Whoa, that ISR indeed is... well, it is something. Can I suggest splitting up the ISRs up in a top half and a bottom half? Essentially, make the ISR just set a semaphore, then have a task that blocks on that semaphore and does stuff when needed. (That would mean you're required to add proper multi-thread locking to the needed variables, however.)

simon-jouet, 2017-05-03:
@Spritetm It is indeed something ... (unfortunately)

All of this code is part of the core of Marlin which I would have wanted to modify as little as possible. I'm not one of the Marlin dev and making this kind of modifications to the code will likely result in the code never being merged into master. Anyway you're right the way it's right now is plain horrible.

Might be better porting a different firmware that has a more sane codebase than Marlin (maybe smoothie?). Anyway thanks for the feedbacks!

simon-jouet, 2017-05-03:
Okay for those interested (which is probably none), I tracked down the issue to just fairly poor code

The comparison at line 1894 https://github.com/simon-jouet/Marlin/blob/334ad3ed9ec9a267511f74b086b73cdd0b1e82bd/Marlin/temperature.cpp#L1894 was comparing int with a float which it didn't like and resulting in the crash.

Still don't get why I couldn't debug with OpenOCD or why the gdbstub wasn't starting on the crash, but I will just blame the isr :)

closed 2017-05-03 by simon-jouet