espressif/esp-idf #563: esp_wifi_scan_start doesn't block until the end of the scan

CristiCimpianu, 2017-05-03:
Hi,

I'm using the master branch:
```
cristic@cucu:~/work/hw/esp-idf$ git branch -v
* master 07b61d54 Merge branch 'bugfix/rtc_fast_freq_init' into 'master'
```
I'm trying a synchronous scenario to scan the wifi channels and then switch to AP mode and it seems to me that esp_wifi_scan_start returns after scanning the first channel instead of waiting for all the channels.
I'm printing the scan results on the SYSTEM_EVENT_SCAN_DONE event and the code I'm using is:
```
  wifi_scan_config_t scan_config = {0};
  scan_config.show_hidden = false;
  scan_config.scan_time.passive = 1000;

  esp_wifi_stop();
  ESP_ERROR_CHECK( esp_wifi_set_mode (WIFI_MODE_STA) );
  ESP_ERROR_CHECK( esp_wifi_start() );
  ESP_ERROR_CHECK( esp_wifi_scan_start (&scan_config, true) );
  esp_wifi_stop();
```
If the final esp_wifi_stop is commented out I get 12 scan results from all channels, if it's not commented, I get 2 scan results from channel 1.

Is it a problem with the API or am I using it wrong?

TimXia, 2017-05-08:
@CristiCimpianu Scan results are processed in event task. The priority of event task is higher than app task. If you call esp_wifi_stop() here, esp32 will stop before getting scan results. So please call esp_wifi_stop() after getting scan results.

CristiCimpianu, 2017-05-08:
@TimXia That's sort of a workaround. Please note that the [docs](http://esp-idf.readthedocs.io/en/latest/api-reference/wifi/esp_wifi.html#_CPPv219esp_wifi_scan_startP18wifi_scan_config_tb) say `if block is true, this API will block the caller until the scan is done, otherwise it will return immediately`.

TimXia, 2017-05-09:
@CristiCimpianu Yes. It blocks until scan is done, not including getting scan results. In fact, we can get scan results whenever after the API returns.

CristiCimpianu, 2017-05-09:
@TimXia I think I don't understand you. I'm retrieving the scan results on the SYSTEM_EVENT_SCAN_DONE event by calling esp_wifi_scan_get_ap_records. The problem is that I'm getting SOME but not ALL scan results if I call esp_wifi_stop. 
Are you telling me that esp_wifi_scan_start does block until the scan is done, but the results are not yet ready to be retrieved when the function returns? If I wait 5 seconds after the function returns will I get the same number of results as in the case when I try to retrieve them immediately after esp_wifi_scan_start returns?
Or are you telling me that esp_wifi_stop frees resources and affects the scan results? In this case why do I get SYSTEM_EVENT_SCAN_DONE with 2 results and I am able to retrieve them?

TimXia, 2017-05-10:
@CristiCimpianu I understand you. It's our problem. The API returns before the scan results are retrived. We will fix it. Thank you for your question.

Pratikhyadav, 2018-02-07:
by default which scanning mode will be set? Active or Passive ?

xentec, 2018-06-07:
Hi, it's been a while. Are you still going to fix this?

XinDeng11, 2018-06-18:
@xentec @CristiCimpianu Yeah, we are on the go

milamber-ls, 2018-11-23:
any update on that?

liuzfesp, 2018-11-23:
Really sorry about this issue. I think this issue still exists in latest IDF, we will give it higher priority.

HudsonProdigy, 2019-08-07:
has this been resolved yet?

espxiehang, 2019-11-06:
@CristiCimpianu @HudsonProdigy I use the master branch for testing and will print all scan results before executing esp_wifi_stop().
```
              esp_wifi_stop();
              ESP_ERROR_CHECK( esp_wifi_set_mode (WIFI_MODE_STA) );
              ESP_ERROR_CHECK( esp_wifi_start() );
              ESP_ERROR_CHECK( esp_wifi_scan_start (config, true) );
              esp_wifi_stop();
```
:>sta -S -k 1000 -h false

:>I (18387) wifi: flush txq
I (18387) wifi: stop sw txq
I (18387) wifi: lmac stop hw txq
I (18407) wifi: mode : sta (3c:71:bf:ce:fe:54)

+WIFI:STA_STOP

+WIFI:STA_START

+SCANSTART

+SCAN:ESP_C119E9,24:0a:c4:c1:19:e9,0,1,-12,0,1,0,0,0,7

+SCAN:mi_mesh,50:64:2b:b7:0c:43,0,4,-27,0,0,0,0,1,7

+SCAN:udp2,74:c3:30:f5:b4:f4,0,9,-34,0,2,0,0,0,7

+SCAN:NETGEAR88,a4:2b:8c:7f:92:50,3,8,-43,0,0,4,4,1,7

+SCAN:Guest_E083,8e:25:93:7f:e0:83,4,9,-44,0,0,4,4,0,7

+SCAN:sky_dlink,80:26:89:73:bd:14,4,6,-45,0,0,5,3,1,7

+SCAN:esp-r-057,88:25:93:7f:e0:83,0,9,-45,0,0,0,0,0,7

+SCAN:testDrive,9c:3d:cf:35:4f:60,0,1,-47,0,0,0,0,0,3

+SCAN:test123,60:38:e0:d1:6d:dd,0,6,-47,0,0,0,0,1,7

+SCAN:test123-guest,62:38:e0:d1:6d:dd,0,6,-48,0,0,0,0,0,7

+SCAN:net,10:da:43:83:f3:8d,0,11,-50,0,0,0,0,1,7

+SCAN:test32,04:5f:a7:0b:17:ec,0,6,-51,0,0,0,0,0,3

+SCAN:bdtest,b0:6e:bf:79:36:08,0,6,-54,0,0,0,0,1,7

+SCAN:wpa2_test,88:9f:fa:f8:fa:77,5,1,-55,0,0,4,4,0,3

+SCAN:7890,30:b5:c2:0e:c1:50,0,1,-55,0,1,0,0,1,7

+SCAN:��测��试,4a:7d:2e:55:aa:a6,4,11,-55,0,2,4,4,0,7

+SCAN:tl5620,48:7d:2e:45:aa:a6,0,11,-56,0,2,0,0,0,7
I (21807) wifi: flush txq

+SCAN:chip-vf,50:fa:84:6d:03:7a,4,13,-56,0,2,4,4,0,7

+SCAN:myssid,80:89:17:1e:92:fc,2,11,-59,0,2,4,4,1,7

+SCAN:net666,30:ae:a4:15:2e:09,0,1,-60,0,1,0,0,0,7
I (21827) wifi: stop sw txq

+SCAN:esp-office-2.4G,70:3a:73:06:08:5f,4,1,-62,0,0,4,4,0,7
I (21837) wifi: lmac stop hw txq

+SCAN:esp-guest,70:3a:73:0e:08:5f,4,1,-62,0,0,4,4,0,7

+SCAN:test_b,a4:50:46:fe:84:3d,3,6,-66,0,0,4,4,0,7

+SCAN:esp-guest,70:3a:73:0e:09:03,4,11,-67,0,0,4,4,0,7

+SCAN:btwifitest,ec:56:23:e9:7e:f0,3,1,-68,0,0,4,4,1,7

+SCAN:esp-eye-9cd6,c4:4f:33:18:9c:d6,0,1,-68,0,1,0,0,0,7

+SCAN:test_auth,78:44:fd:81:8b:20,4,3,-68,0,0,4,4,0,3

+SCAN:VPN-102gate,60:38:e0:10:5d:26,3,5,-69,0,2,4,4,0,7

+SCAN:esp-office-2.4G,70:3a:73:06:09:03,4,11,-69,0,0,4,4,0,7

+SCAN:APTEST3,ac:22:0b:87:26:90,3,6,-70,0,0,4,4,1,7

+SCAN:test123,74:05:a5:3f:ec:41,0,11,-71,0,2,0,0,0,7

+SCAN:GUNQIU_FYX,ec:6c:9f:31:22:2c,3,4,-72,0,0,4,4,0,7

+SCAN:buyaolianwo,a0:63:91:d2:66:76,3,1,-74,0,0,4,4,1,7

+SCAN:sky_asus,1c:87:2c:ce:f2:80,0,6,-74,0,0,0,0,1,7

+SCAN:ESP32-CAMERA,30:ae:a4:df:1c:d9,4,1,-78,0,1,5,3,0,7

+SCAN:esp-guest,70:3a:73:0e:09:0e,4,1,-79,0,0,4,4,0,7

+SCAN:HUAWEI-PYDY4W,34:79:16:24:27:c0,3,11,-82,0,2,4,4,1,7

+SCAN_AP_COUNT=37

+SCANDONE

+WIFI:STA_STOP

closed 2020-06-11 by Alvin1Zhang