espressif/esp-idf #568: BLE indicate/notification issue

Nicholas3388, 2017-05-04:
Hi, is there a sample to show how to send and receive indicate/noitification between two ESP32 devices? I can send notification between ESP32 (sender) and iPhone (receiver). But I cannot receive notification between two ESP32.

xiewenxiang, 2017-05-04:
@Nicholas3388 
If you want receive notify or indicate from server:

1. register for notification.
2. write char description.

In step 1, call esp_ble_gattc_register_for_notify(gattc_if, server_bda, srvc_id, char_id)
```
/**
 * @brief           This function is called to register for notification of a service.
 *
 * @param[in]       gattc_if: Gatt client access interface.
 * @param[in]       server_bda : target GATT server.
 * @param[in]       srvc_id : pointer to GATT service ID.
 * @param[in]       char_id : pointer to GATT characteristic ID.
 *
 * @return
 *                  - ESP_OK: registration succeeds
 *                  - other: failed
 *
 */
esp_gatt_status_t esp_ble_gattc_register_for_notify (esp_gatt_if_t gattc_if,
                                                    esp_bd_addr_t server_bda,
                                                    esp_gatt_srvc_id_t *srvc_id,
                                                    esp_gatt_id_t *char_id);
```

after invoke this API, the event  ESP_GATTC_REG_FOR_NOTIFY_EVT will come.

In step 2, write char descr, if the char prem is notify, write value 0x1, if the char prem is indicate, write value 0x2.
```
    case ESP_GATTC_REG_FOR_NOTIFY_EVT:
        esp_ble_gattc_write_char_descr(......)
        break;

/**
 * @brief           This function is called to write characteristic descriptor value.
 *
 * @param[in]       gattc_if: Gatt client access interface.
 * @param[in]       conn_id : connection ID
 * @param[in]       srvc_id : service ID.
 * @param[in]       char_id : characteristic ID.
 * @param[in]       descr_id : characteristic descriptor ID to write.
 * @param[in]       value_len: length of the value to be written.
 * @param[in]       value : the value to be written.
 * @param[in]       write_type : the type of attribute write operation.
 * @param[in]       auth_req : authentication request.
 *
 * @return
 *                  - ESP_OK: success
 *                  - other: failed
 *
 */
esp_err_t esp_ble_gattc_write_char_descr (esp_gatt_if_t gattc_if, 
                                        uint16_t conn_id,
                                        esp_gatt_srvc_id_t *srvc_id,
                                        esp_gatt_id_t *char_id,
                                        esp_gatt_id_t *descr_id,
                                        uint16_t value_len,
                                        uint8_t *value,
                                		esp_gatt_write_type_t write_type,
                                        esp_gatt_auth_req_t auth_req);

```

OK, you can receive notify or indicate now!

Nicholas3388, 2017-05-04:
@xiewenxiang Hi, thanks for you code. I follow your steps, on the client I register notify for char, say 0xFF01, and then write char descriptor (0x2902). For server, I use the esp-idf gatts_demo. 

The server calls `esp_ble_gatts_add_char` to add char  0xFF01 (read|write|notify), and then calls `esp_ble_gatts_add_char_descr` to add the descriptor 0x2902. When char and desc is added, I create a task to send notification as following:

`
static void led_task(void *arg)
{
  while (1) {
    vTaskDelay(3000 / portTICK_RATE_MS);
    esp_ble_gatts_send_indicate(client_if, client_conn, char_handle, notif_len, value, false);
  }
}
`

The `char_handle` is the handle for char 0xFF01. Is there anything wrong with my steps?  The ESP32 client  cannot receive notification from the ESP32 server.

xiewenxiang, 2017-05-04:
@Nicholas3388 
Hi, Nicholas, I'd like to confirm the procedure for your operation:

1. server: create service 、start service and start adv
2. client connected to server
3. client discover service  `esp_ble_gattc_search_service(gattc_if, conn_id, NULL)` , in this step, If there is only one service on the server, Event `case ESP_GATTC_SEARCH_RES_EVT:` will come three times, the last one is the real service.
4. client discover char `esp_ble_gattc_get_characteristic(gatt_if, conn_id, &srvc_id, NULL)` , in this step, event `case ESP_GATTC_GET_CHAR_EVT:` will come, you can find the descripton of this char, and find next char
```
    case ESP_GATTC_GET_CHAR_EVT:
        if (p_data->get_char.status != ESP_GATT_OK) {
            break;
        }
        char_id.inst_id = p_data->get_char.char_id.inst_id;
        char_id.uuid.len = p_data->get_char.char_id.uuid.len;
        if (char_id.uuid.len == ESP_UUID_LEN_16) {
            char_id.uuid.uuid.uuid16 = p_data->get_char.char_id.uuid.uuid.uuid16;
        } else if (char_id.uuid.len == ESP_UUID_LEN_32) {
            char_id.uuid.uuid.uuid32 = p_data->get_char.char_id.uuid.uuid.uuid32;
        } else if (char_id.uuid.len == ESP_UUID_LEN_128) {
            for (int i = 0; i < 16; i++) {
                char_id.uuid.uuid.uuid128[i] = p_data->get_char.char_id.uuid.uuid.uuid128[i];
            }
        }
        //TODO:
        esp_ble_gattc_get_descriptor(gatt_if, conn_id, &srvc_id, &char_id, NULL);
        break;
    case ESP_GATTC_GET_DESCR_EVT:
        if (p_data->get_descr.status != ESP_GATT_OK) {
            esp_ble_gattc_get_characteristic(gatt_if, conn_id, &srvc_id, &char_id);
            break;
        }
        descr_id.inst_id = p_data->get_descr.descr_id.inst_id;
        descr_id.uuid.len = p_data->get_descr.descr_id.uuid.len;

        if (descr_id.uuid.len == ESP_UUID_LEN_16) {
            descr_id.uuid.uuid.uuid16 = p_data->get_descr.descr_id.uuid.uuid.uuid16;
        } else if (descr_id.uuid.len == ESP_UUID_LEN_32) {
            descr_id.uuid.uuid.uuid32 = p_data->get_descr.descr_id.uuid.uuid.uuid32;
        } else if (descr_id.uuid.len == ESP_UUID_LEN_128) {
            for (int i = 0; i < 16; i++) {
                descr_id.uuid.uuid.uuid128[i] = p_data->get_descr.descr_id.uuid.uuid.uuid128[i];
            }
        }
        //TODO:
        esp_ble_gattc_get_descriptor(gatt_if, conn_id, &srvc_id, &char_id, &descr_id);
        break;
```
5. Make sure that the parameters are correct 
`esp_ble_gattc_register_for_notify(gattc_if, server_bda, srvc_id, char_id)`  `esp_ble_gattc_write_char_descr(...)`

If possible, please provide your code. Thanks

Nicholas3388, 2017-05-05:
@xiewenxiang  Hi, the following code is client:

`#include <stdint.h>
#include <string.h>
#include <stdbool.h>
#include <stdio.h>
#include "controller.h"

#include "bt.h"
#include "bt_trace.h"
#include "bt_types.h"
#include "btm_api.h"
#include "bta_api.h"
#include "bta_gatt_api.h"
#include "esp_gap_ble_api.h"
#include "esp_gattc_api.h"
#include "esp_gatt_defs.h"
#include "esp_bt_main.h"
#include "driver/gpio.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#define BT_BD_ADDR_STR         "%02x:%02x:%02x:%02x:%02x:%02x"
#define BT_BD_ADDR_HEX(addr)   addr[0], addr[1], addr[2], addr[3], addr[4], addr[5]

#define TEST_SRV_UUID			0x00FF
#define TEST_CHAR_UUID			0xFF01
#define LED_GPIO				2

static esp_gatt_if_t client_if;
static uint16_t client_conn = 0;
esp_gatt_status_t status = ESP_GATT_ERROR;
bool connet = false;
uint16_t simpleClient_id = 0xEE;

const char device_name[] = "ESP32";
static char tar_dev_mac[6] = {0xff, 0xb5, 0x30, 0x4e, 0x0a, 0xcb};

static esp_bd_addr_t server_dba;
static bool is_led_on = false;

static esp_ble_scan_params_t ble_scan_params = {
    .scan_type              = BLE_SCAN_TYPE_ACTIVE,
    .own_addr_type          = ESP_PUBLIC_ADDR,
    .scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
    .scan_interval          = 0x50,
    .scan_window            = 0x30
};

static esp_gatt_srvc_id_t test_service_id = {
    .id = {
        .uuid = {
            .len = ESP_UUID_LEN_16,
            .uuid = {.uuid16 = TEST_SRV_UUID,},
        },
        .inst_id = 0,
    },
    .is_primary = true,
};

static esp_gatt_id_t test_char_id = {
	.uuid = {
        .len = ESP_UUID_LEN_16,
        .uuid = {.uuid16 = TEST_CHAR_UUID,},
    },
    .inst_id = 0,
};

static esp_gatt_id_t notify_descr_id = {
[... 549 lines cut ...]
{
    esp_bluedroid_init();
    esp_bluedroid_enable();
    ble_client_appRegister();
}

void app_main()
{
	gpio_pad_select_gpio(LED_GPIO);
    /* Set the GPIO as a push/pull output */
    gpio_set_direction(LED_GPIO, GPIO_MODE_OUTPUT);
    gpio_set_level(LED_GPIO, 0);

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    esp_err_t ret = esp_bt_controller_init(&bt_cfg);

	if (ret) {
        ESP_LOGE(TAG, "%s initialize controller failed\n", __func__);
        return;
    }

    esp_bt_controller_enable(ESP_BT_MODE_BTDM);
    gattc_client_test();

}`

Nicholas3388, 2017-05-05:
The following is the server code:

`#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "bt.h"
#include "bta_api.h"

#include "esp_gap_ble_api.h"
#include "esp_gatts_api.h"
#include "esp_bt_defs.h"
#include "esp_bt_main.h"
#include "esp_bt_main.h"

#include "driver/gpio.h"
#include "sdkconfig.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#define GATTS_TAG "GATTS_DEMO"

///Declare the static function 
static void gatts_profile_a_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if, esp_ble_gatts_cb_param_t *param);
static void gatts_profile_b_event_handler(esp_gatts_cb_event_t event, esp_gatt_if_t gatts_if, esp_ble_gatts_cb_param_t *param);

static bool task_run = false;
static esp_gatt_if_t client_if;
static uint16_t client_conn = 0;
static uint16_t ccc_handle = 0;
static uint16_t notif_test_v = 1;

#define GATTS_SERVICE_UUID_TEST_A   0x00FF
#define GATTS_CHAR_UUID_TEST_A      0xFF01
#define GATTS_DESCR_UUID_TEST_A     0x3333
#define GATTS_NUM_HANDLE_TEST_A     4

#define GATTS_SERVICE_UUID_TEST_B   0x00EE
#define GATTS_CHAR_UUID_TEST_B      0xEE01
#define GATTS_DESCR_UUID_TEST_B     0x2222
#define GATTS_NUM_HANDLE_TEST_B     4

#define TEST_DEVICE_NAME            "ESP32"
#define TEST_MANUFACTURER_DATA_LEN  17

#define GATTS_DEMO_CHAR_VAL_LEN_MAX 0x40

#define LED_GPIO					2

uint8_t char1_str[] = {0x11,0x22,0x33};
esp_attr_value_t gatts_demo_char1_val = 
{
    .attr_max_len = GATTS_DEMO_CHAR_VAL_LEN_MAX,
    .attr_len     = sizeof(char1_str),
    .attr_value   = char1_str,
};

static uint8_t test_service_uuid128[32] = {
    /* LSB <--------------------------------------------------------------------------------> MSB */
    //first uuid, 16bit, [12],[13] is the value
    0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0xAB, 0xCD, 0x00, 0x00,
    //second uuid, 32bit, [12], [13], [14], [15] is the value
    0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0xAB, 0xCD, 0xAB, 0xCD,
};
[... 416 lines cut ...]
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s init bluetooth failed\n", __func__);
        return;
    }
    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(GATTS_TAG, "%s enable bluetooth failed\n", __func__);
        return;
    }

	gpio_pad_select_gpio(LED_GPIO);
    /* Set the GPIO as a push/pull output */
    gpio_set_direction(LED_GPIO, GPIO_MODE_OUTPUT);
    gpio_set_level(LED_GPIO, 0);

    esp_ble_gatts_register_callback(gatts_event_handler);
    esp_ble_gap_register_callback(gap_event_handler);
    esp_ble_gatts_app_register(PROFILE_A_APP_ID);
    esp_ble_gatts_app_register(PROFILE_B_APP_ID);

	xTaskCreate(&led_task, "led_task", 2048, NULL, 6, NULL);

    return;
}
`

xiewenxiang, 2017-05-05:
@Nicholas3388 
Hi, Nicholas, The following code is untested：

1. connected
2. service search
3. get char by test_service_id
4. if (char_prop & 0x30) == 0x0, get next char
5. if (char_prop & 0x30) != 0x0, get the char descr
    register notification for this char
    write this char descr

```
static uint16_t write_desc_value = 0;

static void esp_gattc_cb(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if,esp_ble_gattc_cb_param_t *param)
{
    uint16_t conn_id = 0;
    esp_ble_gattc_cb_param_t *p_data = (esp_ble_gattc_cb_param_t *) param;

    switch (event) {
    case ESP_GATTC_OPEN_EVT:
        client_conn = p_data->open.conn_id;
        at_memcpy(server_dba, p_data->open.remote_bda, sizeof(esp_bd_addr_t));
        esp_ble_gattc_search_service(gattc_if, client_conn, NULL);
        break;
    case ESP_GATTC_SEARCH_RES_EVT:
        test_service_id.is_primary = p_data->search_res.srvc_id.is_primary;
        test_service_id.id.inst_id = p_data->search_res.srvc_id.id.inst_id;
        test_service_id.id.uuid.len = p_data->search_res.srvc_id.id.uuid.len;
        if (test_service_id.id.uuid.len == ESP_UUID_LEN_16) {
            test_service_id.id.uuid.uuid.uuid16 = p_data->search_res.srvc_id.id.uuid.uuid.uuid16;
        } else if (test_service_id.id.uuid.len == ESP_UUID_LEN_32) {
            test_service_id.id.uuid.uuid.uuid32 = p_data->search_res.srvc_id.id.uuid.uuid.uuid32;
        } else if (test_service_id.id.uuid.len == ESP_UUID_LEN_128) {
            for (int i = 0; i < 16; i++) {
                test_service_id.id.uuid.uuid.uuid128[i] = p_data->search_res.srvc_id.id.uuid.uuid.uuid128[i];
            }
        }
        break;
    case ESP_GATTC_SEARCH_CMPL_EVT:
        esp_ble_gattc_get_characteristic(gattc_if, p_data->search_res.conn_id,&test_service_id, NULL);
        break;
    case ESP_GATTC_GET_CHAR_EVT:
        if (p_data->get_char.status != ESP_GATT_OK) {
            break;
        }
        test_char_id.inst_id = p_data->get_char.char_id.inst_id;
        test_char_id.uuid.len = p_data->get_char.char_id.uuid.len;
        pGattc_char_node->char_prop = p_data->get_char.char_prop;
        if (test_char_id.uuid.len == ESP_UUID_LEN_16) {
            test_char_id.uuid.uuid.uuid16 = p_data->get_char.char_id.uuid.uuid.uuid16;
        } else if (test_char_id.uuid.len == ESP_UUID_LEN_32) {
            test_char_id.uuid.uuid.uuid32 = p_data->get_char.char_id.uuid.uuid.uuid32;
        } else if (test_char_id.uuid.len == ESP_UUID_LEN_128) {
            for (int i = 0; i < 16; i++) {
                test_char_id.uuid.uuid.uuid128[i] = p_data->get_char.char_id.uuid.uuid.uuid128[i];
            }
        }
        if(((p_data->get_char.char_prop)&0x10)||((p_data->get_char.char_prop)&0x20)){
            if(((p_data->get_char.char_prop)&0x10)){
                write_desc_value = 0x1;
            }else if(((p_data->get_char.char_prop)&0x20)){
                write_desc_value = 0x2;
            }
            esp_ble_gattc_get_descriptor(gattc_if, client_conn, &test_service_id, &test_char_id, NULL);
        }else{
            esp_ble_gattc_get_characteristic(gattc_if, client_conn, &test_service_id, &test_char_id);
        }
        break;
    case ESP_GATTC_GET_DESCR_EVT:
        if (p_data->get_descr.status != ESP_GATT_OK) {
            break;
        }
        notify_descr_id.inst_id = p_data->get_descr.descr_id.inst_id;
        notify_descr_id.uuid.len = p_data->get_descr.descr_id.uuid.len;

        if (notify_descr_id.uuid.len == ESP_UUID_LEN_16) {
            notify_descr_id.uuid.uuid.uuid16 = p_data->get_descr.descr_id.uuid.uuid.uuid16;
        } else if (notify_descr_id.uuid.len == ESP_UUID_LEN_32) {
            notify_descr_id.uuid.uuid.uuid32 = p_data->get_descr.descr_id.uuid.uuid.uuid32;
        } else if (notify_descr_id.uuid.len == ESP_UUID_LEN_128) {
            for (int i = 0; i < 16; i++) {
                notify_descr_id.uuid.uuid.uuid128[i] = p_data->get_descr.descr_id.uuid.uuid.uuid128[i];
            }
        }
        esp_ble_gattc_register_for_notify(gattc_if, server_dba,&test_service_id, &test_char_id);
        
        break;
    case ESP_GATTC_REG_FOR_NOTIFY_EVT:
        esp_ble_gattc_write_char_descr(
                gattc_if,
                client_conn,
                &test_service_id,
                &test_char_id,
                &notify_descr_id,
                sizeof(write_desc_value),
                (uint8_t *) &write_desc_value,
                ESP_GATT_WRITE_TYPE_NO_RSP,
                ESP_GATT_AUTH_REQ_NONE);
        
        esp_ble_gattc_get_characteristic(gattc_if, client_conn, &test_service_id, &test_char_id);
        break;
    case ESP_GATTC_WRITE_DESCR_EVT:
        break;
    case ESP_GATTC_NOTIFY_EVT:
        break;
    case ESP_BLE_EVT_CONN_DIR_ADV:
        break;
    default:
        break;
}
```

xiewenxiang, 2017-05-05:
@Nicholas3388 
Hi, Nicholas, can you receive indicate/noitification now?

Nicholas3388, 2017-05-06:
@xiewenxiang  Hi, thanks for you code. Unfortunately, I still cannot receive notification now. I think my code is almost the same with your untested code. The character property is set as ESP_GATT_CHAR_PROP_BIT_NOTIFY on server, and on client  I can search the character whose property is accurately notify. But the client cannot receive notification, when the server called  `esp_ble_gatts_send_indicate`

By the way, in your code is it necessary to call  `esp_ble_gattc_get_characteristic` after `esp_ble_gattc_write_char_descr`?

xiewenxiang, 2017-05-12:
@Nicholas3388
I don't find register for notification in your client code；
```
/**
 * @brief           This function is called to register for notification of a service.
 *
 * @param[in]       gattc_if: Gatt client access interface.
 * @param[in]       server_bda : target GATT server.
 * @param[in]       srvc_id : pointer to GATT service ID.
 * @param[in]       char_id : pointer to GATT characteristic ID.
 *
 * @return
 *                  - ESP_OK: registration succeeds
 *                  - other: failed
 *
 */
esp_gatt_status_t esp_ble_gattc_register_for_notify (esp_gatt_if_t gattc_if,
                                                    esp_bd_addr_t server_bda,
                                                    esp_gatt_srvc_id_t *srvc_id,
                                                    esp_gatt_id_t *char_id);
```

wolfking0815, 2017-05-30:
Hi,

I also met up with this issue. It seems that the esp_gattc_write_char_descriptor can not write the descriptor correctly. Because when I tried to write the descriptor, after I registered the notification, the BT will dump out "NO pending cmd" for this function. However, I can write the characteristic successfully. Any suggestions?

Thanks.
kaikai

Nicholas3388, 2017-05-31:
@wolfking0815  There is no problem with writing characteristic. But I still cannot receive notifications between two ESP32 devices

danicampora, 2017-07-20:
I am having this issue as welll... Has anyone had luck with this? I can register for Notifications from an smartphone, but not from a ESP32 acting as a GATT client.

TianHao-Espressif, 2017-07-26:
@Nicholas3388 ， @Weijian-Espressif will reply you and provide a demo.

Weijian-Espressif, 2017-07-26:
@Nicholas3388 I  provide you a demo. you can test it. 
[gatts_demo.c.zip](https://github.com/espressif/esp-idf/files/1175389/gatts_demo.c.zip)
[gattc_demo.c.zip](https://github.com/espressif/esp-idf/files/1175390/gattc_demo.c.zip)

The client demo automatically connect the server demo, the clinet would enable the notify of server demo after connection is established. then the server demo would send some  notify value to the client . The client will send back the value to the server after receiving the notify value. 

 if any question, let me know.

Nicholas3388, 2017-07-27:
@Weijian-Espressif @TianHao-Espressif  Thanks. It works for me

closed 2017-07-27 by Nicholas3388