espressif/esp-idf #569: Can't access SD card from task other than main

jsarrett, 2017-05-04:
Accessing a file on a VFS mounted SD card from a task other than app_main doesn't work.  This is likely some sort of re-entrancy problem.  Attached is a very simple test case.
[test_sd.zip](https://github.com/espressif/esp-idf/files/975278/test_sd.zip)

jsarrett, 2017-05-04:
stdout here:

https://gist.github.com/jsarrett/a801b6967a0a089d41a2e19fe85af9bf

igrr, 2017-05-04:
There are two issues observed with the code you have posted:

1. Illegalinstruction exception. In FreeRTOS, a task function is not allowed to return. Behaviour in this case is undefined, and is implementation-specific. In the ESP32 port, this causes an illegal instruction exception. When a task terminates, it needs to call `vTaskDelete(NULL)`.

2. SDMMC driver error when communicating with the card from a task. This issue is caused by a race condition: the task code starts initializing the card, which main task code initializes GPIO4. GPIO4 is one of IOs used to communicate with the SD card. `gpio_set_direction` function changes the pin direction while the SDMMC peripheral is communicating with the card. This causes an error to be returned. To avoid this error, use a synchronization primitive to wait until SD card operation is complete, and then continue with the main task.

The example with both issues fixed can be found here: https://gist.github.com/igrr/57d9a0c4c4605f711634ee5c3e825d10

jsarrett, 2017-05-06:
Indeed that fixed it.

Thanks for the second set of eyes, that was a pretty silly bug!

-James

closed 2017-05-06 by jsarrett