espressif/esp-idf #570: Stack overflow in task btuT has been detected

johnnorbury, 2017-05-04:
Updated to latest master.

When using Lightblue to send data to the ESP32:

```
.[0;32mI (71891) ESP32-DEV: Profile A Handler.[0m
.[0;32mI (71891) ESP32-DEV: EVENT OF TYPE: 15.[0m
ESP32 SDK version:master, RAM left 136812
.[0;31mE (71901) ESP32-DEV: Disconnected event, advertising....[0m
.[0;31mE (71901) BT: ####################adv_type = 0.[0m
D (74901) nvs: nvs_open bt_config.conf 1.[0m
D (74901) nvs: nvs_set_blob bt_cfg_key 216.[0m
D (74901) nvs: nvs_close 7.[0m
.[0;31mE (77981) BT: BLE HCI(id=62) event = 0x0a).[0m
.[0;31mE (77981) BT: status = 0, handle = 0, role = 1, bda_type = 0.[0m
.[0;31mE (77981) BT: btm_ble_resolve_random_addr_on_conn_cmpl unable to match and resolve random address.[0m
.[0;32mI (77991) ESP32-DEV: Profile A Handler.[0m
.[0;32mI (77991) ESP32-DEV: EVENT OF TYPE: 14.[0m
ESP32 SDK version:master, RAM left 136796
.[0;31mE (78011) BT: MTU request PDU with MTU size 104
.[0m
.[0;31mE (78011) BT: BTM_SetBleDataLength failed, peer does not support request.[0m
.[0;32mI (78011) ESP32-DEV: Profile A Handler.[0m
.[0;32mI (78011) ESP32-DEV: EVENT OF TYPE: 4.[0m
ESP32 SDK version:master, RAM left 136752
MTU EVENT
***ERROR*** A stack overflow in task btuT has been detected.
abort() was called at PC 0x40087f4c on core 0
Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x4000bff0:0x3ffdcf20 0x40085dc4:0x3ffdcf40 0x400858d6:0x3ffdcf60 0x40087033:0x3ffdcf80 0x40086fed:0x3ffcfa3c
```

When using the nRF android app the MTU event isn't called and so this error is unseen. 

Communication between Lightblue and the ESP was fine with master from a month ago.

Any ideas?

YelongHuang, 2017-05-23:
@johnnorbury Can you update the IDF sdk, I think this bug is fixed in the new sdk.

FayeY, 2017-06-15:
Hi @johnnorbury , is this problem solved? Can we close this issue?

closed 2017-07-13 by FayeY