espressif/esp-idf #571: Incorrect bootstrap interpretation after BOR reset

alarruskain, 2017-05-04:
I've been testing new silicon V1 with VCC dips and I've been facing system hangouts waiting for UART or SDIO download when the GPIO_STRAP_REG should remain unmodified as GPIO is pulled up and the program has no modified the register.
There is some more information posted in 
[https://esp32.com/viewtopic.php?f=2&t=1801](url)

I've read documentation and it doesn't mention the First stage bootloader start up code behavior when the reset reason is BOR. The documentation says that GPIO_STRAP_REG should be checked after POR, Software SOC Reset or WDR. Thus, I understand that something is missing...

alarruskain, 2017-05-23:
I've noticed that the link wasn't correctly set. Now it is correct

FayeY, 2017-06-21:
Hi @alarruskain , is this problem solved? Can we close this issue?

alarruskain, 2017-06-21:
HI, 
the truth is that is not solved, I will try to post the info here:

We are developing products for industrial applications, so we are really concerned about their reliability and thus we usually make a bunch of tests during their validation.
In our first tests we had ESP-WROOM-32 with ESP32 V0 in them and we immediately realized that the BOR was not running.
After you released silicon V1 we managed to get some samples of ESP32 V1 from our distributor ASTONE in Europe and we switched the ESP32 from those ESP-WROOM-32 we had. After validating that the re-soldering was OK with X-rays, we modified the esp-idf, where Brownout detection is disabled by default by removing the dependency on the non existing NEEDS_ESP32_NEW_SILICON_REV. and configured the esp-idf for example like follows:
[image]

And run the tests to te module. With this silicon version we saw better performance but we continued being able to lock in the module, having this log output:
[image]
As you may see the output indicates a BROWN_OUT_RESET and even if the strapping of the module is not changed, the module stars in Download Boot.
More precisely, this happened and trying to find out a solution, we got this behavior doing gaps longer than 10ms to our mains,.
Our circuit has an input DC DC and a LDO feeding the module like in this schema:

[image]

So, if we make gaps smaller than 10 ms, the capacitance from mains to the module hold the ESP32 vDD over 2.1 Volts during the gap period. If we make gaps longer than that, but no so long to generate a POWER ON RESET  when the main comes back, in the margin from 10 ms to 25-30 ms is were we find the phenomenon. 
In these captures you may see the capture of the following schematic points:
Yelllow: The signal named 4V6 to 24V
Blue: The signal named 3V6
Green: The signal named 3V3 Core
[image]
[image]
[image]
[image]

From my point of view, there shouldn't be a way to run into Download boot after a BROWN_OUT_RESET, but anyway it seems that when the boot is done the VDD remains under the BOR reset threshold. 

I hope this helps.
Thank you and kind regards

projectgus, 2017-07-10:
@igrr has made some changes to the BOD behaviour which should help in this case (strapping registers will no longer be read after a Brown Out Reset, so the ESP32 shouldn't BOR into the bootloader mode.)

He was unable to reproduce this exact failure mode, though. We think maybe the ETH_RMII_CLK_TX connection may be related, if the PHY is also browning out then it may be pulling this pin low while the BOR occurs on the ESP32.

As a quick check you could try disconnecting/cutting the ETH_RMII_CLK_TX trace and see if this makes a difference. For maximum reliability, perhaps consider adding a series resistor, stronger pullup, and/or buffer chip to that clock line so that it can't influence the strapping pin.

alarruskain, 2017-07-11:
Hi projectgus, 
first of all thank you for your answer.
I have just updated esp-idf, and enabled brownout for testing. 
We have no better performance with the changes. Additionally I have monitored IO0/ETH_RMII_CLK_TX to ensure that we are not having the PHY pulling down the pin.  I understand that you mean that the PHY should not pull down the pin when the BOR is released and not when it is asserted because that can happen during normal operation and the PHY must be able to drive the pin freely...

here you are some captures of how the system behaves currently. You will notice the effect of the strong pull up on IO0 that enables that noise from the RMII clock litters the 3V3. Probably should modify the schematics to pull the line up from one point before the Linear Regulator 3V6  but I think that for this case is illustrative of how the line remains high during the boot.

In yellow, input to DC/DC, pink Regulator input 3V6, green 3V3 Core, and in blue IO0/ETH_RMMI_CLK_TX

[image]
[image]
As you may see the IO0 is not pulled down when the BOR condition is released.
This exact capture derives the following log:
```

ets Jun  8 2016 00:22:57

rst:0xf (RTCWDT_BROWN_OUT_RESET),boot:0x3 (DOUUIQ▒/UART1/SDIO_REI_REO_V2))
waiting for download
```
but with same BOR patterns I sometimes get correct reboots with this following boot log:
```

rst:0xf (RTCWDT_BROWN_OUT_RESET),boot:0x13LSH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock dets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2308
load:0x40078000,len:9984
ho 0 tail 12 room 4
load:0x40080000,len:252

```
I have done some more test with longer BOR conditions, 30ms and in this case with these captures:

[image]
[image]

I get the same locked status but with the following boot log:
```
rst:0xf (RTCWDT_BROWN_OUT_RESET),boot:0xets Jun  8 2016 00:22:57

```
I understand that the BOR is asserted twice. ¿ Does it have any sense if the main has gone down and up once ?
Thank you and Best regards
Iker

alarruskain, 2017-07-20:
Hi, 
tested again with new HW (this time was a ESP-WROOM-32 V1 not a refurbished V0 one with ESP32 replaced) and the new updates regarding BOR on 13th. 
I get different performance and better behavior with 0V gaps but with other patterns I get same result,.
Now I see the message of the BOR interrupt handler but then I get the incorret bootstrap coming from a RTCWDT_RTC_RESET.
```
...
Brownout detector wa.▒▒erets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```
In the test I switch from a 10Volt input to 4.2V, 0.3V  under the limit where the DC/DC is able to provide the 3.48V necessary to get the 3.3V regulator output.
Thus, I can control how long the Brownout "status" lasts.

In yellow DCDC input, in green DCDC output, in pink ESP-WROOM-32 3V3 input in blue IO0

[image]

[image]

Additionally as the voltage doesn't keep going down, the peripherals, like the PHY ar not reseting and thus I get the RMII clk clock in IO0 as you suggested and there might be something to be done to reduce the impact. 
Anyway there are two things that they are not running as spected. In one side the boot message does not belong to a BOR reset.
In the other side, our HW has power control of the PHY and it is disabled in the default status of the corresponding IO, so in a power on boot the PHY remains unfed until the FW requires it. So, I expect that if the reset is asserted by the BOR handler, the IOs go to their default uninitialized status.
[image]
[image]

Thank you and best regards

igrr, 2017-07-23:
>In one side the boot message does not belong to a BOR reset.

This is because the the reset is not performed by the brownout detector. Instead, BOD triggers an interrupt, which is handled in software. Interrupt handler prints "Brownout detector was triggered" message and performs software reset of CPUs. If somewhere along the way the CPU faults (because of brownout, or for any other reason), then the SoC will be reset using RTC watchdog, which is what happens here.

> So, I expect that if the reset is asserted by the BOR handler, the IOs go to their default uninitialized status.

We had different requirements from other customers; their applications required that if panic/BOR/WDT reset happen, IOs should not change their states, if possible. We will consider making this behavior configurable somehow, to fit your use case. In the meantime, you may modify components/esp32/brownout.c and add any behaviour your need (such as changing GPIO states) into the BOD interrupt handler.

alarruskain, 2017-07-25:
hello @igrr 
> you may modify components/esp32/brownout.c and add any behaviour your need (such as changing GPIO states) into the BOD interrupt handler. 

Good, I can manage the IO states of the ESP32 into the BOD interrupt handler before the panic/BOR/WDT reset happen.
But from your clarification, 
>  If somewhere along the way the CPU faults (because of brownout, or for any other reason), then the SoC will be reset using RTC watchdog, which is what happens here.

I understand that if any other cause, not Brownout where actions can be done, causes a RTC watchdog reset, then the IOs will remain in their current status. Thus, in any design with an Ethernet PHY driving the clock to the ESP32 may happen that after a RTC watchdog, if the PHY is driving low, the ESP32 may interpret as a Download Boot. Am I right?

marcogiuliato, 2022-01-17:
Hi @igrr ,

Sorry if I re-open this issue, but we are encountering the same problem of @itc-jalonso in our devices.
We are using esp-idf v4.1 and ESP32 WORVER-E.

From the logs we see the same problem :
"
Brownout detector ets ...

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
"

So, we are planning to set to 0 the state of the GPIO 0 (boot pin) into the BOD interrupt handler in order to fix the problem.
As you've suggested, we're going to change its state inside components/esp32/brownout.c, using the function gpio_set_level inside the rtc_brownout_isr_handler. 
Is this the correct way to implement your suggested solution?

Thanks in advance for your help

igrr, 2022-01-17:
Hi @marcogiuliato, what you propose could work, it depends mainly on the circuit attached to GPIO0. If there is some capacitance on that pin which keeps GPIO0 low and prevents it from going high before the strapping pins are sampled, then forcing the output high could fix the issue.

marcogiuliato, 2022-01-17:
Hi @igrr 

We have no capacitances on this pin.

Can you check if the fix is correctly implemented in the following code of modified brownout.c?

`// Copyright 2015-2017 Espressif Systems (Shanghai) PTE LTD
//
// Licensed under the Apache License, Version 2.0 (the "License");
// you may not use this file except in compliance with the License.
// You may obtain a copy of the License at
//
//     http://www.apache.org/licenses/LICENSE-2.0
//
// Unless required by applicable law or agreed to in writing, software
// distributed under the License is distributed on an "AS IS" BASIS,
// WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
// See the License for the specific language governing permissions and
// limitations under the License.

#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <stdbool.h>
#include "sdkconfig.h"
#include "soc/soc.h"
#include "soc/cpu.h"
#include "soc/rtc_periph.h"
#include "esp32/rom/ets_sys.h"
#include "esp_private/system_internal.h"
#include "driver/rtc_cntl.h"
#include "freertos/FreeRTOS.h"
#include "driver/gpio.h"

#ifdef CONFIG_ESP32_BROWNOUT_DET_LVL
#define BROWNOUT_DET_LVL CONFIG_ESP32_BROWNOUT_DET_LVL
#else
#define BROWNOUT_DET_LVL 0
#endif //CONFIG_ESP32_BROWNOUT_DET_LVL

static void rtc_brownout_isr_handler(void *arg)
{
    /* Normally RTC ISR clears the interrupt flag after the application-supplied
     * handler returns. Since restart is called here, the flag needs to be
     * cleared manually.
     */
    gpio_set_level(GPIO_NUM_0, 1);
    REG_WRITE(RTC_CNTL_INT_CLR_REG, RTC_CNTL_BROWN_OUT_INT_CLR);
    /* Stall the other CPU to make sure the code running there doesn't use UART
     * at the same time as the following ets_printf.
     */
    esp_cpu_stall(!xPortGetCoreID());
    esp_reset_reason_set_hint(ESP_RST_BROWNOUT);
    ets_printf("\r\nBrownout detector was triggered\r\n\r\n");
    esp_restart_noos();
}

void esp_brownout_init(void)
{
    REG_WRITE(RTC_CNTL_BROWN_OUT_REG,
            RTC_CNTL_BROWN_OUT_ENA /* Enable BOD */
            | RTC_CNTL_BROWN_OUT_PD_RF_ENA /* Automatically power down RF */
            /* Reset timeout must be set to >1 even if BOR feature is not used */
            | (2 << RTC_CNTL_BROWN_OUT_RST_WAIT_S)
            | (BROWNOUT_DET_LVL << RTC_CNTL_DBROWN_OUT_THRES_S));

    ESP_ERROR_CHECK( rtc_isr_register(rtc_brownout_isr_handler, NULL, RTC_CNTL_BROWN_OUT_INT_ENA_M) );

    REG_SET_BIT(RTC_CNTL_INT_ENA_REG, RTC_CNTL_BROWN_OUT_INT_ENA_M);
	
	//zero-initialize the config structure.
    gpio_config_t io_conf = {};
    //disable interrupt
    io_conf.intr_type = GPIO_INTR_DISABLE;
    //set as output mode
    io_conf.mode = GPIO_MODE_OUTPUT;
    //bit mask of the pins that you want to set,i.e. boot pin
    io_conf.pin_bit_mask = GPIO_NUM_0;
    //disable pull-down mode
    io_conf.pull_down_en = 0;
    //disable pull-up mode
    io_conf.pull_up_en = 0;
    //configure GPIO with the given settings
    gpio_config(&io_conf);
	
}
`

igrr, 2022-01-17:
Is GPIO0 being used in the application in any way? If yes, then initializing it in esp_brownout_init could conflict with the usage of the pin later. If not, perhaps you don't need to set the pin to 1 in the brownout handler, just set it to 1 from the very beginning?

marcogiuliato, 2022-01-17:
No, it is not used in our code. 
I'm thinking of setting it to 1 from the very beginning and also in the rtc_brownout_isr_handler to make sure it's level hasn't been changed by voltage disturbances. Do you think it could be a valid solution?

igrr, 2022-01-17:
I think you can simply set it to 1 in the beginning then. GPIO pad has a latch for the output value, so setting the output again to the same value doesn't do anything, and it doesn't "reinforce" the value.

marcogiuliato, 2022-01-17:
Ok, thank you a lot for your help. I'll let you kwnow if this fix solves our issue.

marcogiuliato, 2022-01-17:
Sorry @igrr, is esp_brownout_init automatically called at boot? 
If so, in order to set GPIO 0 to 1 at the very beginning I can set it in this way:

`
void esp_brownout_init(void)
{
    REG_WRITE(RTC_CNTL_BROWN_OUT_REG,
            RTC_CNTL_BROWN_OUT_ENA /* Enable BOD */
            | RTC_CNTL_BROWN_OUT_PD_RF_ENA /* Automatically power down RF */
            /* Reset timeout must be set to >1 even if BOR feature is not used */
            | (2 << RTC_CNTL_BROWN_OUT_RST_WAIT_S)
            | (BROWNOUT_DET_LVL << RTC_CNTL_DBROWN_OUT_THRES_S));

    ESP_ERROR_CHECK( rtc_isr_register(rtc_brownout_isr_handler, NULL, RTC_CNTL_BROWN_OUT_INT_ENA_M) );

    REG_SET_BIT(RTC_CNTL_INT_ENA_REG, RTC_CNTL_BROWN_OUT_INT_ENA_M);
	
	//zero-initialize the config structure.
    gpio_config_t io_conf = {};
    //disable interrupt
    io_conf.intr_type = GPIO_INTR_DISABLE;
    //set as output mode
    io_conf.mode = GPIO_MODE_OUTPUT;
    //bit mask of the pins that you want to set,i.e. boot pin
    io_conf.pin_bit_mask = GPIO_NUM_0;
    //disable pull-down mode
    io_conf.pull_down_en = 0;
    //disable pull-up mode
    io_conf.pull_up_en = 0;
    //configure GPIO with the given settings
    gpio_config(&io_conf);
    gpio_set_level(GPIO_NUM_0, 1);
	`

Am i right?

marcogiuliato, 2022-01-17:
Or do you suggest to set pin 0 to 1 directly from our application source code?