espressif/esp-idf #574: components/driver/spi_master: Usage of command and address fields for TX MSB first

devsaurus, 2017-05-05:
I've stumbled over the question how the application should populate the `command` and `address` fields in `spi_transaction_t` for the TX MSB first mode. I.e. when `SPI_DEVICE_TXBIT_LSBFIRST` is not set in `spi_device_interface_config_t.flags`.

I observed from LA traces that the hardware appears to follow distinctive patterns for the command and address fields. Specifically, command is shifted out starting with the lower byte first, followed by the upper byte. Each of them most-significant bit first. Address in contrast is shifted out starting with bit 63.

As an application developer, I can't determine how to provide command and address to achieve the desired waveform on MOSI. The verbatim copy of the struct elements to the regs exposes HW characteristics that are not specified in the technical reference manual.
My preference would be that the driver abstracts this so that command and address fields are always shifted out from bit 0 to n-1 for SPI_DEVICE_TXBIT_LSBFIRST or bit n-1 downto 0 for the inverse case (n is defined by `command_bits` or `address_bits`).

ginkgm, 2017-08-28:
hi @devsaurus ,  
This issue has been resolved in commit b636f8161930ceb3f07c00f415001a9ac4314692 and ed1d084aeaef1c7216004354bd5b1a08f4a584cf.  
please have a try. if you have any problem using this, please let us know.

ginkgm, 2017-08-28:
we've rewrite the command and address field to make you use it easier in commit b636f8161930ceb3f07c00f415001a9ac4314692 and 4c9754726f6d96f6b6f02039f52d18c671dff660

please update to the latest idf and have a try.

devsaurus, 2017-09-01:
Thanks @ginkgm, I'll test soon.

closed 2017-09-02 by devsaurus