espressif/esp-idf #575: How to test 'tcp_perf'? Does not seem to work for large files

mindThomas, 2017-05-05:
I have been trying to test the TCP performance of the ESP32 by wanting to send a large file to it.
Seeing your "tcp_perf" example I guessed that this project was exactly intended for such a performance test.
For testing I am sending a large file (>2 MB) to the device using the following TCP sending script: https://github.com/mindThomas/Merged-TCP-UDP-sender-receiver/blob/master/Sender.c

I have built the "tcp_perf" application from ESP-IDF without any changes to the actual code. Initially I run "make menuconfig" only to change the COM port and to set the example to "Server" and "Receive" mode.

Without touching the Watchdog settings the device seem to stop working when I start transmitting the file:
`Task watchdog got triggered. The following tasks did not feed the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
`

If I disable the watchdog I get no response and the socket seem closed.
Within the transmitting application it also times out and returns '-1'.

liuzfesp, 2017-05-10:
Hi @mindThomas we will merge iperf into idf soon, it's for performance test, I think maybe you can use iperf to do your test after the iperf is ready.

liuzfesp, 2017-05-10:
Hi @mindThomas I guess the watchdog happens because following code
`        while ((bytes_left = fread (buffer_data_ptr, 1, MAXDATASIZE-4, file_to_read)) > 0) {
            printf("Number of bytes read: %d\n", bytes_left);
            uint32_t HEADER_netorder = htonl(HEADER);
            memcpy(&buffer[0], (void *)&HEADER_netorder, (size_t)sizeof HEADER);
        
            bytes_sent = sendto(sockfd, buffer_beg_ptr, bytes_left+4, 0, receiver_info->ai_addr, receiver_info->ai_addrlen);
            bytes_left -=(int)bytes_sent;
            total_bytes_sent += (int)bytes_sent;
            printf("Total bytes sent: %d\n", (int)total_bytes_sent);
            packet_counter++;
}`

For UDP, the default socket is always unblocking, in other words, the sendto will return immediately no matter what happens (such as out of memory etc), in other words, the while loop will continuously using CPU, of course, the watchdog will surely happen here.

mindThomas, 2017-06-09:
Oh, you are referring to the code I am running on a PC. The Sender.c file is used for the sending application running on my host PC connected to the same network as the ESP32.

For receiving on the ESP32 I am using the "tcp_perf" example. The receiving function of this example is found here: https://github.com/espressif/esp-idf/blob/master/examples/performance/tcp_perf/main/tcp_perf.c#L141-L156

Now that this has been clarified, do you have any clue of why the watchdog is triggered?
Please try the "tcp_perf" yourself with my mentioned Sender.c on a host PC. Just compile with a regular GCC installation.

liuzfesp, 2017-10-10:
Hi @mindThomas , the iperf demo is already added into latest IDF, please use iperf demo example to test the performance. Iperf demo is placed in: example/wifi/iperf

closed 2017-11-21 by FayeY