espressif/esp-idf #578: esp_ota_begin() runs into default 5s watchdog on large firmware binaries

flyinggorilla, 2017-05-07:
hi, i am working on implementing OTA support and ran into this watchdog issue. I think the problem is that erasing the flash of large OTA partitions >1MB (i.e. 1280K) takes longer than 5s and then runs into the watchdog. 

This issue can be worked around by increasing the "Task watchdog timeout (seconds)" from default 5s to 10s. So maybe thats good enough of a solution. However, since I dont like to deviate from defaults, I felt I report this back to you.

 
calling....
` err = esp_ota_begin(mpUpdatePartition, OTA_SIZE_UNKNOWN, &mOtaHandle);
`

leads to... 
```
I (32644) ota: Starting OTA example...
I (32644) ota: Running partition type 0 subtype 17 (offset 0x00290000)
I (32654) ota: Writing to partition subtype 16 at offset 0x150000
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: ipc0
CPU 1: IDLE
I (32674) ota: esp_ota_begin succeeded
```

with almost maximum OTA partition sizes...
```
# Name,   Type, SubType, Offset,   Size
nvs,      data, nvs,     0x9000,  0x4000
otadata,  data, ota,     0xd000,  0x2000
phy_init, data, phy,     0xf000,  0x1000
factory,  app,  factory, 0x10000,  1280K
ota_0,    app,  ota_0,   ,         1280K
ota_1,    app,  ota_1,   ,         1280K
data, 	  data, fat,  ,            192k  
```

Hope that helps,
Best,
FG

tzxespressio, 2017-05-08:
Hi @flyinggorilla ,thank for your report,yes, it should let IDLE task feed task WDT,this is a loop in functions "spi_flash_erase_range", add a "vTaskDelay" in the loop will fix your issue,I think @igrr can give a perfect solution about it.  

> if (rc == ESP_ROM_SPIFLASH_RESULT_OK) {
        for (size_t sector = start; sector != end && rc == ESP_ROM_SPIFLASH_RESULT_OK; ) {
            spi_flash_guard_start();
            if (sector % sectors_per_block == 0 && end - sector > sectors_per_block) {
                rc = esp_rom_spiflash_erase_block(sector / sectors_per_block);
                sector += sectors_per_block;
                COUNTER_ADD_BYTES(erase, sectors_per_block * SPI_FLASH_SEC_SIZE);
            } else {
                rc = esp_rom_spiflash_erase_sector(sector);
                ++sector;
                COUNTER_ADD_BYTES(erase, SPI_FLASH_SEC_SIZE);
            }
            spi_flash_guard_end();
        }
    }

blmhemu, 2017-07-05:
It gives this error even when i try upload blink code via OTA

projectgus, 2017-07-05:
@blmhemu  as @tzxespressio mentioned, adding a vTaskDelay(1) inside the loop where you call esp_ota_write() should fix this problem.

It's worth noting that the Task Watchdog is (in default config) not fatal, so even though these warnings are printed the OTA should complete OK.

You can also adjust the Task Watchdog timeout in menuconfig (under Component Config -> ESP32-Specific Config) up from 5 seconds to a larger number. It's firing because the OTA process is monopolising the system while writing to flash. It probably won't happen if you're not on a high speed connection like a LAN, where packets are coming faster than they can be written to flash - meaning the task is constantly starting a new write, and never pausing.

All this said, it's still a bug we plan to fix so the default configuration doesn't print this error.

That said, it's still a bug

igrr, 2017-07-07:
@liuzfesp please take a look (regarding `wifi: pp.c 3125`)

chegewara, 2017-11-21:
Im working with example ota right now and im not having issues with flashing 1,5MB app over internet.

FayeY, 2017-11-21:
Since this is a very old issue, I'm going to close it. Please feel free to reopen it if your problem is still unsolved. Thanks.

closed 2017-11-21 by FayeY