espressif/esp-idf #579: Crashes in libnet80211.a

tedwoodt25, 2017-05-08:
In AP Mode I'm experiencing crashes in  libnet80211.a.

Guru Meditation Error of type LoadProhibited occurred on core 0. Exception was unhandled.
Register dump:
PC : 0x400e0548 PS : 0x00060f30 A0 : 0x800e0f4a A1 : 0x3ffc22c0
0x400e0548: hostap_input at ??:?

A2 : 0x3ffcd2bc A3 : 0x3ffcdf7c A4 : 0x00000000 A5 : 0x3ffcd2bc
A6 : 0x3ffce218 A7 : 0x3ffb7ce4 A8 : 0x800e0544 A9 : 0x3ffc22a0
A10 : 0x00000000 A11 : 0x000000d9 A12 : 0x3ffcdf7c A13 : 0x3ffcd2bc
A14 : 0x00000000 A15 : 0x00000000 SAR : 0x00000019 EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000 LBEG : 0x4000c2e0 LEND : 0x4000c2f6 LCOUNT : 0xffffffff

Backtrace: 0x400e0548:0x3ffc22c0 0x400e0f4a:0x3ffc2390 0x400f3388:0x3ffc23b0 0x400f4a96:0x3ffc2400
0x400e0548: hostap_input at ??:?
0x400e0f4a: ap_rx_cb at ??:?
0x400f3388: ppRxPkt at ??:?
0x400f4a96: ppTask at ??:?

It was originally reported here:
https://github.com/warmcat/lws-esp32-factory/issues/3

But diagnosed as part of esp-idf

liuzfesp, 2017-05-10:
Hi @tedwoodt25 this issue already fixed in latest esp32 idf, but it wan't merged into official release2.0, so you can update your idf to latest one or wait for idf2.1 because it will come soon, maybe this week or next week.

liuzfesp, 2017-05-10:
Hi @tedwoodt25 can I reproduce this issue with esp-idf example code? If yes, I can verify it myself first and double check this issue is really fixed in latest idf, thanks!

tedwoodt25, 2017-05-10:
Building git@github.com:warmcat/lws-esp32-factory.git with the latest esp-idf, this appears to be fixed.

closed 2017-05-10 by tedwoodt25

liuzfesp, 2017-05-11:
HI @tedwoodt25 we fixed one issue about hostap_input in release 2.1, I just checked the latest code and the fixed is not merged into release2.0 yet, anyway, just reopen this issue if you see it again.