espressif/esp-idf #580: A2DP Audio Streaming

narg95, 2017-05-08:
Hi everyone!

Two questions: 
- How can I achieve Audio Streaming (A2DP SRC) from ESP32 to an A2DP SINK compatible device? I could not find any functionality on the ESP Bluetooth API nor in the examples.
- Would it be technically possible to stream at the same time to multiple A2DP SINK devices even with low sound quality? 

Thanks,
Nestor Rodriguez

ppisljar, 2017-05-29:
would be great to have an example of this

mywang-espressif, 2017-06-07:
@narg95 @ppisljar , here are my answers to your questions:
1. For now A2DP SRC is not implemented for ESP32 yet, but it is in our plan. We are going to port and adapt the code from bluedroid and provide ESP bluetooth APIs as well as an example.
2. I think it is possible. Taking a glimpse at the bluedroid source code, I found some clues that multiple A2DP link can be supported in the bluetooth stack, e.g. the macro "AVDT_NUM_LINKS". But I also find that the upper layer "btif" stuff in bluedroid are not designed to support the use of multiple AVDTP links. Therefore to satisfy your requirement, there will be some modifications to the original bluedroid logic. I cannot provide more information before the A2DP SRC is ported and verified. For the first step, A2DP SRC that allows one single SINK device is likely to be provided.

ppisljar, 2017-06-07:
any idea about the timeline (when could we expect to see first versions of A2DP SRC implementation) ? thanks a lot for your feedback!

mywang-espressif, 2017-06-08:
@ppisljar , I think you can see the first version of A2DP SRC in about one and a half month.

FayeY, 2017-06-19:
Hi @narg95 , is this problem solved? Can we close this issue?

ppisljar, 2017-06-19:
@FayeY are you saying this is now implemented ?

FayeY, 2017-06-19:
@ppisljar , Sorry, my mistake. It is not implemented now.

Pmarva, 2017-08-25:
i really like to see A2DP SRC implemented for ESP32, do you have any dates or plans when it can happen?

ppisljar, 2017-09-08:
i was looking at the code .. in `btc_avk.c` i added this function:
```
bt_status_t btc_a2d_source_connect(bt_bdaddr_t *remote_bda)
{
    LOG_DEBUG("%s\n", __FUNCTION__);
    CHECK_BTAV_INIT();

    return btc_queue_connect(UUID_SERVCLASS_AUDIO_SOURCE, remote_bda, connect_int);
}
```

the init seems to be the same (so we can call sink init as it only inits a2dp), so whole init looks like this then:
```
       /* initialize A2DP source */
        esp_a2d_register_callback(&bt_app_a2d_cb);
        esp_a2d_register_data_callback(bt_app_a2d_data_cb);
        esp_a2d_source_init();

        /* initialize AVRCP controller */
        esp_avrc_ct_init();
        esp_avrc_ct_register_callback(bt_app_rc_ct_cb);
```

now i have no idea how to actually send audio data to the sink, nor if the above init is OK.

mywang-espressif, 2017-09-12:
Hi @Pmarva , the implementation of A2DP SRC for esp32 is almost done and it functionally work for now, some review and tests are needed. But I am afraid this feature will not occur until some time later when Classic BT has APIs of GAP, i.e, device discovery, service discovery, authentication, and pairing are implemented. The reason is that the APIs mentioned are necessary for an "initiative" device which can take the role of A2DP SRC, for example.
These features can hopefully occur in a minor release after version 3.0 of ESP-IDF, which may take about 2 months. I am sorry for the change of the plan.

mywang-espressif, 2017-09-12:
Hi @ppisljar . The code ported bluedroid(Android 6.0) does not seperate A2DP SRC and A2DP SNK that clearly. Even though the APIs for A2DP SRC and SNK are provided together, the lower stack does not support the two roles simultaneously. In fact, the A2DP role is decided at compile time, and modifying only "btc" layer(which corresponds to "btif" layer in original Bluedroid) can not let it work.

The logic for A2DP SRC in btc_media.c is removed from current ESP-IDF. And you need to add the code back according to bluedroid to use A2DP SRC. Moreover, the "UIPC" part is also modifed when we were porting the code to ESP-IDF. You may also need to disable macros for A2DP SNK such as "BTA_AV_SNK_INCLUDED", and enable the SBC encoders. Please refer to original bluedroid for more information.

NikiSchlifke, 2017-09-29:
@mywang-espressif could you add the current functionality to the repository in a separate feature branch please?
I still want to risk playing around with it.

adherent88, 2017-12-07:
any update on A2DP SRC after 3 months already :) ?

Jongraknie, 2017-12-15:
please update A2DP SRC!!!

Spritetm, 2017-12-15:
At the moment, we do have somewhat of an A2DP SRC solution, but it changes some things in other components as well, and is not deemed stable at the moment. I'll see if we can maybe release something as a no-guarantees-given branch, but I can't say if we'll actually do that.

adherent88, 2017-12-22:
Thx for the update @Spritetm 

Do you have a timeline when A2DP SRC will be officially released(or if it will be released at all)?

Because the missing A2DP SRC example is currently a bottleneck for our project. If we have to wait long, then we need to look for other solutions.

negativekelvin, 2018-01-20:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/a2dp_source

ESP32DE, 2018-02-08:
@negativekelvin 
do you can say [which format and how we set the format as parameters](https://esp32.com/viewtopic.php?f=2&t=4636) for the audio src we must do?

txs

blueMoodBHD, 2018-03-21:
According to the profile, a2dp source should supported at least one of the values between 44.1 kHz and 48 kHz. 16 kHz and 32 kHz is optional. And, we set SBC codec to 44.1 kHz, now.

blueMoodBHD, 2018-03-21:
If you want to set SBC codec to 48 kHz. The changes are as follows:

In bta_av_co.c line 73
```
// (A2D_SBC_IE_SAMP_FREQ_44), /* samp_freq */
(A2D_SBC_IE_SAMP_FREQ_48 | A2D_SBC_IE_SAMP_FREQ_44), /* samp_freq */
```
2 In bta_av_co.c line 94
```
#if !defined(BTC_AV_SBC_DEFAULT_SAMP_FREQ)
// #define BTC_AV_SBC_DEFAULT_SAMP_FREQ A2D_SBC_IE_SAMP_FREQ_44
#define BTC_AV_SBC_DEFAULT_SAMP_FREQ A2D_SBC_IE_SAMP_FREQ_48
#endif
```

3 In btc_a2dp_source.c line 498
``` 
/* for now hardcode 44.1 khz 16 bit stereo PCM format */
media_feeding.cfg.pcm.sampling_freq = 48000;//44100;
```

closed 2018-03-26 by FayeY

rob-bits, 2018-06-13:
Hello all,

I am interesting in the functionality of stream at the same time to multiple A2DP SINK devices. I want to stream audio to two device from the ESP32. Is it possible with the current esp-idf?

Can be the A2DP_source example code modified to work with two sink devices?

I went through the current source example and I think it can be modificated to discover multiple devices and connect to them, but will it work in the end?

Can be the esp_a2d_source_connect() function called if we already connected to one sink device?

If we can connect multiple sink devices already then what about the following code (in /a2dp_source/main/main.c line 277 - 280):
```
/* initialize A2DP source */
esp_a2d_register_callback(&bt_app_a2d_cb);
esp_a2d_source_register_data_callback(bt_app_a2d_data_cb); 
```

What will be the role of the data callback function? Will it stream the data to both of the connceted device? Or how can we handle which data goes where?

Rob

blueMoodBHD, 2018-06-14:
Hello @rob-bits ,
I am sorry to tell you that, streaming audio to two device from the ESP32 is impossible with the current esp-idf.

rob-bits, 2018-06-14:
Hello @blueMoodBHD ,

Thank you very much for your reply.
And would you please point out which function blocks this functionality?

In theory, if I would have enought resources to rework the stack to get this functionality where should I start?

Any input welcomes with joy. 

Rob

pad52, 2018-10-25:
> If you want to set SBC codec to 48 kHz. The changes are as follows:
>
> In bta_av_co.c line 73
>
> ```
> // (A2D_SBC_IE_SAMP_FREQ_44), /* samp_freq */
> (A2D_SBC_IE_SAMP_FREQ_48 | A2D_SBC_IE_SAMP_FREQ_44), /* [...]

Hello @blueMoodBHD 
Is it possible to enable 48Khz also for A2DP Sink?
or at least use the resampling library provided by bluedrod?

Thanks.

shalnoff, 2018-11-08:
> Hello @rob-bits ,
> I am sorry to tell you that, streaming audio to two device from the ESP32 is impossible with the current esp-idf.

But would it be possible theoretically? What is it dependent? 

What about BT multicasting support? (I'd read abiut this new feature standard implementation some time ago, unfortunatley can't find the link)

rob-bits, 2018-11-09:
Hi @shalnoff,
Yes, it is possible. I have already a working proof-of-concepct example with the bluekitchen btstack.
Please see my issue:
[TW#23410] A2DP Source stream audio to two Sink device #2060

I shared a bluetooth specification for multiple streams.