espressif/esp-idf #582: Problem compiling example project

Edzelf, 2017-05-09:
I have a problem compiling any example.  In this case "blink".  It seems impossible to create an archive with "xtensa-esp32-elf-ar.exe".  I'm using mingw32.exe on Windows 10.

$ make --trace
.
.
.
AR libbootloader_support.a
rm -f libbootloader_support.a
xtensa-esp32-elf-ar cru libbootloader_support.a src/bootloader_random.o src/flash_encrypt.o src/esp_image_format.o src/flash_partitions.o src/secure_boot_signatures.o src/secure_boot.o src/efuse.o src/bootloader_flash.o
C:\Users\Ed\msys32\opt\xtensa-esp32-elf\bin\xtensa-esp32-elf-ar.exe: **unable to rename 'libbootloader_support.a'; reason: Permission denied**
make[2]: *** [/home/Ed/esp/esp-idf/make/component_wrapper.mk:140: libbootloader_support.a] Fout 1
make[1]: *** [/home/Ed/esp/esp-idf/make/project.mk:386: bootloader_support-build] Fout 2
make: *** [/home/Ed/esp/esp-idf/components/bootloader/Makefile.projbuild:31: /home/Ed/esp/blink/build/bootloader/bootloader.bin] Fout 2

Edzelf, 2017-05-09:
Solved.  Problem disappears if mingw32 is executes as "Administrator".

closed 2017-05-09 by Edzelf

projectgus, 2017-05-10:
Thanks for updating us @Edzelf .

If you think there's a problem with the way MSYS2 is configured here, please feel free to reopen. On a "standard" Windows install I don't think it should be necessary to Run As Administrator, especially if you're working from your Users directory.