espressif/esp-idf #584: time_t is 32-bit wide, please change to 64-bit to make applications Y2K38 safe

Endres, 2017-05-09:
The `time_t` base type on which all unix-timestamp (seconds after 1970-01-01) related functions (such as `gettimeofday`) do rely on, is only 4 bytes (32-bit) wide. It is suggested to use an 8 byte (64-bit) variable here, as otherwise all unix-timestamps after 2038-01-18 will be wrong. To sketch this, I coded an example ([main.c](https://github.com/espressif/esp-idf/files/987654/main.c.txt)), which shows the exact problem. Notably is, that not only the current wall-clock time is affected by this, also calculations on the time (for example, calendar events and the like) which result in a value higher than the noted limit in 2038, are affected. This could become at some time a serious issue, so it would be best to act soon. See also the [wikipedia page](https://en.wikipedia.org/wiki/Year_2038_problem) on this problem.

So it would be very appreciating to change the `time_t` base type to an "signed long long" type.

The current output of my sample code can be seen here:
```
I (508) example: sizeof(time_t): 4 (32-bit)
I (1508) example: Time now: 000000007FFFFFF8 (2038-01-19 03:14:00)
I (2508) example: Time now: 000000007FFFFFF9 (2038-01-19 03:14:01)
I (3508) example: Time now: 000000007FFFFFFA (2038-01-19 03:14:02)
I (4508) example: Time now: 000000007FFFFFFB (2038-01-19 03:14:03)
I (5508) example: Time now: 000000007FFFFFFC (2038-01-19 03:14:04)
I (6508) example: Time now: 000000007FFFFFFD (2038-01-19 03:14:05)
I (7508) example: Time now: 000000007FFFFFFE (2038-01-19 03:14:06)
I (8508) example: Time now: 000000007FFFFFFF (2038-01-19 03:14:07)
I (9508) example: Time now: FFFFFFFF80000000 (1901-12-13 20:45:52)
I (10508) example: Time now: FFFFFFFF80000001 (1901-12-13 20:45:53)
I (11508) example: Time now: FFFFFFFF80000002 (1901-12-13 20:45:54)
I (12508) example: Time now: FFFFFFFF80000003 (1901-12-13 20:45:55)
I (13508) example: Time now: FFFFFFFF80000004 (1901-12-13 20:45:56)
I (14508) example: Time now: FFFFFFFF80000005 (1901-12-13 20:45:57)
I (15508) example: Time now: FFFFFFFF80000006 (1901-12-13 20:45:58)
I (16508) example: Time now: FFFFFFFF80000007 (1901-12-13 20:45:59)
I (17508) example: Time now: FFFFFFFF80000008 (1901-12-13 20:46:00)
```

hwmaier, 2017-06-02:
Or alternatively make it an unsigned 32-bit integer postponing the 32-bit overflow by further 68 years.

jack0c, 2018-03-21:
@igrr Can we close this issue?

igrr, 2018-03-21:
@jack0c no, we will be working on it.

igrr, 2018-06-19:
Currently we supply newlib as a set of pre-built libraries. `time_t` size is fixed at the time these libraries are built. We plan to also allow for doing a custom builds of newlib in IDF, which would allow changing sizeof(time_t) and other things.

For the change of default time_t size, we will have to wait until one of the major releases, since that would be a breaking change for some existing applications.

vielmetti, 2018-06-22:
I see a v2.0 milestone that looks complete at https://github.com/espressif/esp-idf/milestone/1 being tracked (and complete), but no other issue-based tracking of release milestones in Github.

https://github.com/espressif/esp-idf/releases/tag/v3.1-beta1 is the v3.1 beta.

Do you anticipate this waiting for a v4.0 major release? I know that might be some time out (maybe 2019 at your current release cadence?)

projectgus, 2018-06-22:
Hi @vielmetti ,

Thanks for the reminder about the GitHub Milestones feature. As you can see, we haven't been using it. We plan to start publishing a version roadmap soon, but at the moment there isn't one publically accessible.

Currently the plan is for V4.0 to be in beta release by the end of the year, for a final release in early 2019.

vielmetti, 2018-08-31:
Thanks @projectgus - hope that v4.0 is perking right along, and that you can confirm that the time_t changes will be in it.

BarryAndrews, 2018-12-13:
Is there any further update on when this year 2038 issue will be resolved?

projectgus, 2019-01-31:
No ETA for the newlib update & resolution yet, I'm afraid. Although we are tracking this.

Our timeline slipped a bit, currently v3.3 is in beta and we're about to start work on v4.0 breaking features, of which the newlib update will be one.

remcoNL, 2019-02-04:
@projectgus Can you maybe consider a major update with just this change, (since it has a deadline..) In the spirit of continuous delivery (-;

remcoNL, 2019-02-05:
That's a relief...

I will now subtract 883612800 seconds and dates and weekdays are the same again! Also this gives you an extra 28 years to make the update! 

https://math.stackexchange.com/questions/536847/how-often-in-years-do-calendars-repeat-with-the-same-day-date-combinations-juli

projectgus, 2019-05-23:
Hi @remcoNL,

Glad you found a workaround!

We've decided to push this back from the v4.0 release, as we're already making a lot of other newlib changes. Sorry for the inconvenience. The current plan is as follows:

* 64-bit time_t will be supported in IDF v4.1 as an optional configuration. It may require building a custom toolchain where newlib is configured with 64-bit time_t (this process will be documented).
* 64-bit time_t is planned to become default in IDF v5.0

We're currently averaging a major version per year, so there should still be plenty of time between the v5.0 release and the 2038 deadline.

OTA updating a firmware with 32-bit time_t to 64-bit time_t should work fine, also (assuming the OTA update is complete before 2038.)

remcoNL, 2019-05-23:
Really appreciate the work, and i'll schedule my updates for next year (or one year later).

closed 2020-01-10 by espressif-bot

themadsens, 2021-01-12:
Please remember ```struct timeval``` from ```gettimeofday()``` As of IDF 4.0 the ```tv_sec``` member is a ```long``` (defined in ```lwip/sockets.h```(!?)

themadsens, 2021-04-21:
Knock knock ..
Now that you have closed this issue by means of SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS, is there another ticket that tracks progress on an official 64bit time_t toolchain?

projectgus, 2021-04-22:
Hi @themadsens,

We have this on our internal list of breaking changes that we plan to make in ESP-IDF V5.0 (the toolchain support was unfortunately too late to be included in V4.0, so V5.0 is our next opportunity to introduce this). I'm afraid there's no public ticket apart from this one, though.

> Please remember struct timeval from gettimeofday() As of IDF 4.0 the tv_sec member is a long (defined in lwip/sockets.h(!?)

Thanks for pointing this out and sorry noone replied when you first commented about it. This is something that LWIP supports (for platforms without compliant sys/time.h). However, we don't use this in ESP-IDF:
* https://github.com/espressif/esp-idf/blob/v4.0/components/lwip/port/esp32/include/lwipopts.h#L814
* https://github.com/espressif/esp-lwip/blob/c9e3b53c6f04052943f97210b858cec805cc98d9/src/include/lwip/sockets.h#L523

Building a custom toolchain with `SDK_TOOLCHAIN_SUPPORTS_TIME_WIDE_64_BITS` and using it with ESP-IDF is supported since V4.1, so if you find anything else that doesn't work in this configuration (on V4.1 or newer) then please open a new GitHub issue for it and we'll fix it.

AxelLin, 2022-05-18:
> Hi @remcoNL,
>
> Glad you found a workaround!
>
> We've decided to push this back from the v4.0 release, as we're already making a lot of other newlib changes. Sorry for the inconvenience. The current [...]

So next toolchain release will support 64-bit time_t, right?

igrr, 2022-05-18:
Yes, we are trying to include this time_t change into the upcoming toolchain release based on GCC 11.

igrr, 2022-06-16:
time_t is now 64-bit in the GCC 11 based toolchain available on the master branch since https://github.com/espressif/esp-idf/commit/bf43076c044a58429ebd90ac508e32dd82321af1.

chipweinberger, 2022-11-29:
Is this change in v4.4.3? or only v5.0?

andrew-elder, 2022-12-13:
@igrr , what is the projected release date of v5.0?

chipweinberger, 2022-12-14:
It's already out!

https://github.com/espressif/esp-idf/releases/tag/v5.0

andrew-elder, 2022-12-14:
@chipweinberger - thank you!