espressif/esp-idf #588: Calling esp_ble_gattc_open results in a GATTS connection event

MrSurly, 2017-05-10:
I'm registering two apps, a GATTS and GATTC, which results in two different interfaces:

```network_bluetooth_gap_event_handler(event = 0E / LOCAL_IR, param = ()
network_bluetooth_gatts_event_handler(event = 00 / REG, if = 04, param = (status = 00 / OK, app_id = 0000)
network_bluetooth_gattc_event_handler(event = 00 / REG, if = 05, param = (status = 00 / OK, app_id = 0001)
network_bluetooth_gap_event_handler(event = 0F / LOCAL_ER, param = ()
```

As you can see, two different interfaces for GATTS and GATTC.

Code in question is here: https://github.com/MrSurly/micropython-esp32/commit/7e3214a8fe77b2aa1c0c37d3b790794865b4e506, part of adding BLE to MicroPython.  Specifically `esp32/network_bluetooth.c`.  

When I call `esp_ble_gattc_open(<if>, <adx>, true)`, I get an `ESP_GATTC_OPEN_EVT` for interface 05, this is expected, but I _also_ get an `ESP_GATTS_CONNECT_EVT` **for interface 04**, which is entirely unexpected.

The same thing happens with `ESP_GATTC_CLOSE_EVT` and `ESP_GATTS_DISCONNECT_EVT`.

This only happens for GATTC.

I'm using IDF f73c6f8.

The relevant commands entered, without the logging cruft :

```python
>>> import network as n
>>> b = n.Bluetooth()
>>> conn = b.connect(bytearray(b'\x00\x22\xd0\xcc\xe9\x97'))
```

The address for the connect is for a Polar H7 heart rate monitor.

Full text

```
MicroPython v1.8.6-1138-g71fae46-dirty on 2017-05-05; ESP32 module with ESP32
Type "help()" for more information.
>>> import network as n
I (35121) wifi: wifi firmware version: a8ef40a
I (35121) wifi: config NVS flash: enabled
I (35121) wifi: config nano formating: disabled
I (35131) wifi: Init dynamic tx buffer num: 32
I (35131) wifi: wifi driver task: 3ffb48bc, prio:23, stack:3584
I (35131) wifi: Init static rx buffer num: 10
I (35131) wifi: Init dynamic rx buffer num: 0
I (35141) wifi: Init rx ampdu len mblock:7
I (35141) wifi: Init lldesc rx ampdu entry mblock:4
I (35151) wifi: wifi power manager task: 0x3ffdfbf4 prio: 21 stack: 2560
I (35151) wifi: wifi timer task: 3ffdfd64, prio:22, stack:3584
I (35181) phy: phy_version: 350, Mar 22 2017, 15:02:06, 0, 0
I (35181) wifi: mode : null
>>> b = n.Bluetooth()
BT is deinit, initializing
BTDM CONTROLLER VERSION: 010101
BTDM ROM VERSION 0101
BD_ADDR: 24:0A:C4:05:7E:E6
NVDS MAGIC FAILED
RF Init OK with coex
Enable Classic BT
Enable Low Energy
E (35361) BT: Startup BTU

E (35361) BT: GATTS_CreateService:  handle of service handle1
E (35371) BT: bta_dm_co_ble_load_local_keys: func not ported

>>> network_bluetooth_gap_event_handler(event = 0E / LOCAL_IR, param = ()
network_bluetooth_gatts_event_handler(event = 00 / REG, if = 04, param = (status = 00 / OK, app_id = 0000)
network_bluetooth_gattc_event_handler(event = 00 / REG, if = 05, param = (status = 00 / OK, app_id = 0001)
network_bluetooth_gap_event_handler(event = 0F / LOCAL_ER, param = ()
conn = b.connect(bytearray(b'\x00"\xd0\xcc\xe9\x97'))
>>> E (40171) BT: smp_br_connect_callback is called on unexpected transport 2

network_bluetooth_gatts_event_handler(event = 0E / CONNECT, if = 04, param = (conn_id = 0000, remote_bda = 00:22:D0:CC:E9:97, is_connected = true)
network_bluetooth_gattc_event_handler(event = 02 / OPEN, if = 05, param = (status = 00 / OK, conn_id = 0000, addr = 00:22:D0:CC:E9:97, mtu = 0017)
network_bluetooth_gap_event_handler(event = 08 / AUTH_CMPL, param = ()
network_bluetooth_gap_event_handler(event = 08 / AUTH_CMPL, param = ()

```

YelongHuang, 2017-06-01:
@MrSurly , as you said, when you register the gatts & gattc module to the APP to the BLE stack, the BLE stack will have two profile in it(each register will return a gatt_if). So, when you sent gattc_open to connect to the remote BLE device, when the connected has been established, the BLE stack will sent the connect event to all of the profile(gattc_open event and gatts connect event), in your case it is normal that the gatts receive the connect event when you sent the gattc open.

MrSurly, 2017-06-01:
@YelongHuang 
> when the connected has been established, the BLE stack will sent the connect event to all of the profile(gattc_open event and gatts connect event),

That would only make sense if the local GATTC was connecting to the local GATTS.

How do you tell real GATTS CONNECT events from remote GATTC from the events caused by local GATTC connecting to remote GATTS?

wolfking0815, 2017-06-22:
@MrSurly: just change your GATTS event handler from GATTS_CONNECT to GATTS_OPEN can help. The GATTA_CONNECT is sent out when there is a physical connection, GATTS_OPEN deals with the virtual connection. Hope it can help.
Kaikai

FayeY, 2017-07-05:
Hi @MrSurly , is this problem solved now?

MrSurly, 2017-07-05:
> Hi @MrSurly , is this problem solved now?

I don't believe it is.  My position is that a GATTC connection should not result in a GATTS_OPEN event.  

However, it seems the official position on this is "it's not a problem, so there will be no fix."  I don't agree, but if that's the answer, then I guess there's no reason not to close this issue.

closed 2017-10-10 by FayeY

zaazbb, 2018-12-26:
i met the same problem.