espressif/esp-idf #589: Building toolchain on Raspbian fails

gregstewart90, 2017-05-10:
I tried using the precompiled 64 bit linux toolchain from the read the docs page. 

When I run make, I get the following errors:
`/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: ELF: not found
/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 2: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: ELF: not found
/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 2: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: ELF: not found
/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 2: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: ELF: not found
/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 2: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
CC src/bootloader_flash.o
/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: ELF: not found
/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 2: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
/home/pi/esp/esp-idf/make/component_wrapper.mk:176: recipe for target 'src/bootloader_flash.o' failed
make[2]: *** [src/bootloader_flash.o] Error 2
/home/pi/esp/esp-idf/make/project.mk:386: recipe for target 'bootloader_support-build' failed
make[1]: *** [bootloader_support-build] Error 2
/home/pi/esp/esp-idf/components/bootloader/Makefile.projbuild:31: recipe for target '/home/pi/esp/hello_world/build/bootloader/bootloader.bin' failed
make: *** [/home/pi/esp/hello_world/build/bootloader/bootloader.bin] Error 2`

I ran both 
export PATH=$PATH:$HOME/esp/xtensa-esp32-elf/bin
export IDF_PATH=/home/pi/esp/esp-idf

I also confirmed that /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc exists.

I thought the toolchain might now work on the raspberry pi, so I was going to compile it myself. I followed the steps and when I run ./ct-ng build, after a few minutes of it running, the raspberry pi reboots. I tried using the toolchain that Kolban compiled last year, but it is out of date with the current IDF.

waveform80, 2017-05-11:
Raspbian is 32-bit (to the best of my knowledge there aren't any supported pure 64-bit Linux distros for the Pi yet, although work in that area is ongoing). More importantly though, I strongly suspect the ESP-IDF Linux builds are for Intel processors, not ARM.

I'll have a go at compiling the toolchain myself on a Pi3 in a bit (I've managed it happily on an Ubuntu box, so the procedure is probably fairly similar given both are Debian based distros).

waveform80, 2017-05-12:
Left it building overnight on a Pi3 and it all seems to work nicely. Follow the instructions in [Setup Linux Toolchain from Scratch](http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup-scratch.html), specifically the Ubuntu 16.04 instructions (as you need `libtool` and `libtool-bin` under the latest Raspbian). The compile takes about 2.5 hours on a Pi 3B (I wouldn't recommend trying it on anything less than a Pi 2B), and you'll want an SD card (or some form of mounted storage) with at least 4Gb free.

gregstewart90, 2017-05-23:
Great. I got it working, but needed to do a couple of things. I followed [this ](https://esp32.com/viewtopic.php?f=17&t=256&hilit=raspberry+pi) guide from the ESP32 form.
`sudo apt-get update
sudo apt-get upgrade
sudo apt-get install gawk gperf grep gettext automake bison flex texinfo help2man libtool libtool-bin git wget make libncurses-dev python python-serial python-dev python-pip
sudo pip install pyserial
cd ~/esp
git clone -b xtensa-1.22.x https://github.com/espressif/crosstool-NG.git
cd crosstool-NG
./bootstrap && ./configure --enable-local && make install
./ct-ng xtensa-esp32-elf
nano ./.config`

-- > Find CT_PARALLEL_JOBS=0 and change 0 to 1

`./ct-ng build build.1
chmod -R u+w builds/xtensa-esp32-elf`

closed 2017-07-23 by igrr