espressif/esp-idf #591: Static allocation for software timers?

waveform80, 2017-05-11:
Are there any plans to add support for statically allocated software timers, i.e. the [`xTimerCreateStatic`](http://www.freertos.org/xTimerCreateStatic.html) API?

Currently I'm working on something involving the [micropython-esp32](https://github.com/micropython/micropython-esp32) project. This recently moved to static allocation when threading support was added. Static allocation works very nicely with tasks, semaphores, queues, etc. but having recently found a need for software timers I was a little surprised to find `xTimerCreateStatic` wasn't in ESP-IDF. I'm guessing this is something that was introduced in FreeRTOS 9, and ESP-IDF is based on 8.2?

(`vTimerSetTimerID` appears to be something else introduced in 9, which would also make things a little easier as one could use it to pass varying information per timer)

danicampora, 2017-05-23:
@waveform80 you can also use dynamic allocation with the micropython-esp32 port. If you really need static allocated timers, then feel free to back-port that from FreeRTOS 9 and submit a pull request ;-)

igrr, 2017-12-15:
Static allocation of software timers has been backported in https://github.com/espressif/esp-idf/commit/c1d101dd41af4c883f6c3222a9295323e2cc51ae.

closed 2017-12-15 by igrr