espressif/esp-idf #595: Peripheral write ordering issues

rojer, 2017-05-11:
Not sure if it applies elsewhere, but i have a repro with I2C.
Here goes:
```c
// #define BUGME
void bugme(void) {
  i2c_dev_t *dev = &I2C0;
  int ci = 0;
  dev->command[ci].val = 0;  // RSTART
  ci++;
#ifdef BUGME
  dev->command[ci].val = 0x123;
#endif
  dev->command[ci].val = 0x0901;  // WRITE 1 byte
  ci++;
#ifdef BUGME
  dev->command[ci].val = 0x456;
#endif
  dev->command[ci].val = 0x0902;  // WRITE 2 bytes
  for (int i = 0; i <= ci; i++) {
    printf("cmd %d = %08x\n", i, dev->command[i].val);
  }
}
```
this produces the following expected output:
```
cmd 0 = 00000000
cmd 1 = 00000901
cmd 2 = 00000902
```
however, with `BUGME` defined, the output becomes:
```
cmd 0 = 00000000
cmd 1 = 00000123
cmd 2 = 00000456
```
so, earlier write to the register stomps the preceding one.
the function is compiled to the following assembly code:
```
(gdb) disas bugme
Dump of assembler code for function bugme:
   0x40143a4c <+0>:     entry   a1, 32
   0x40143a4f <+3>:     l32r    a8, 0x40143a40
   0x40143a52 <+6>:     movi.n  a2, 0
   0x40143a54 <+8>:     memw
   0x40143a57 <+11>:    s32i    a2, a8, 88
   0x40143a5a <+14>:    movi    a3, 0x123
   0x40143a5d <+17>:    memw
   0x40143a60 <+20>:    s32i    a3, a8, 92
   0x40143a63 <+23>:    l32r    a3, 0x401262f8
   0x40143a66 <+26>:    memw
   0x40143a69 <+29>:    s32i    a3, a8, 92
   0x40143a6c <+32>:    movi    a3, 0x456
   0x40143a6f <+35>:    memw
   0x40143a72 <+38>:    s32i    a3, a8, 96
   0x40143a75 <+41>:    l32r    a3, 0x40143a44
   0x40143a78 <+44>:    memw
   0x40143a7b <+47>:    s32i    a3, a8, 96
   0x40143a7e <+50>:    mov.n   a3, a8
   0x40143a80 <+52>:    addi    a8, a2, 20
   0x40143a83 <+55>:    addx4   a8, a8, a3
   0x40143a86 <+58>:    memw
   0x40143a89 <+61>:    l32i    a12, a8, 8
   0x40143a8c <+64>:    l32r    a10, 0x40143a48
   0x40143a8f <+67>:    or      a11, a2, a2
   0x40143a92 <+70>:    addi    a2, a2, 1
   0x40143a95 <+73>:    call8   0x400fde58 <printf>
   0x40143a98 <+76>:    bnei    a2, 3, 0x40143a80 <bugme+52>
   0x40143a9b <+79>:    retw.n
End of assembler dump.
(gdb) x/1x 0x40143a40                                                                                                                                                                                                                         
0x40143a40:     0x3ff53000
(gdb) x/1x 0x401262f8
0x401262f8:     0x00000901
(gdb) x/1x 0x40143a44
0x40143a44:     0x00000902
(gdb) quit
```

to me, everything here seems kosher: struct is properly recognized as volatile and `memw`s are inserted as required. yet, it seems that bus ordering of the writes is screwed up.

rojer, 2017-05-11:
i now believe this is errata 3.3 applied to I2C, which is not mentioned there yet.

negativekelvin, 2017-05-11:
Sounds like a good theory.

Side question: why does it use both movi and l32r? Is it because you can only encode up to a 12-bit signed literal in the instruction word?

rojer, 2017-05-11:
yes, i think it's just because constant is too large.

there's more to come though - switching to AHB address for I2C did not solve all of the problems, - so stay tuned ;)

rojer, 2017-05-11:
#597 is the second act of the saga.
as for this... can the list be made exhaustive? i swear i read the document before, and even used AHB register for UART FIFO access in my driver, but I2C caught me off-guard. there' no guarantee of course that i'd remember it, but maybe it would register that I2C _also_ had this issue. but it wasn't no the list, so it didn't.

FayeY, 2017-11-22:
This issue should have been solved on the latest IDF, if it is not, please feel free to reopen it.

closed 2017-11-22 by FayeY