espressif/esp-idf #597: Byte writes to peripheral struct fields mess up register contents

rojer, 2017-05-11:
As  workaround for #595, I changed I2C base address to the AHB area - 0x60013000
However, this only exposed the following issue: byte writes to registers mess up the upper 24 bits of the register value.
And why would anyone use `s8i` to write to a peripheral register? I agree, but compiler thinks it can.
So, consider [i2c_dev_t.command[N].byte_num](https://github.com/espressif/esp-idf/blob/15a61459610d6ead47d0a940bba65215c37a541d/components/soc/esp32/include/soc/i2c_struct.h#L249) (my favorite register - register of the day, without a doubt). It's an 8-bit sized field occupying the lower 8 bits of a 32-bit struct. So compiler is like - oh, i know! i'm going to just `s8i` this thing here... i'm so smart, so optimizing!
Except CPU goes all like - BZZZT! wrong! so you want to stick your byte int my word-size register? how about "no"...
Well, an unaligned store would make it obvious. But it doesn't cause an exception. Instead, some pretty random shit happens - bits from the byte get replicated into upper bits, certainly bits 8 to 15 get most of the bits from 0 to 7, and bits 16 to 31 get mostly zeroes, but not necessarily. In any case, upper 24 bits are nothing like what you expect them to be and it's all downhill from there.

@igrr- this may be relevant to your interests. i don't know if compiler needs to be somehow told to just not use byte access to struct fields or if it's supposed to work and it's just a hardware issue.

here's my test code:

```c
void __attribute__ ((noinline)) bugmemore(i2c_dev_t *dev, int ci, uint16_t bn) {
  dev->command[ci].val = 0;
  dev->command[ci].op_code = 1;
  dev->command[ci].ack_en = true;
  dev->command[ci].done = 1;
  dev->command[ci].byte_num = bn;  // BOOM!
  dev->command[ci].ack_val = 0;
}

void bugme(void) {
  i2c_dev_t *dev = (i2c_dev_t *) 0x60013000;  /* AHB address of &I2C0 */;
  int ci = 0;
  dev->command[ci].val = 0;  // RSTART
  ci++;
  bugmemore(dev, ci, 0xaa);
  for (int i = 0; i <= ci; i++) {
    printf("cmd %d = %08x\n", i, dev->command[i].val);
  }
}
```

`bugmemore` it compiles to the following assembly code:
```
(gdb) disas bugmemore
Dump of assembler code for function bugmemore:
   0x40143a8c <+0>:     entry   a1, 32
   0x40143a8f <+3>:     addx4   a9, a3, a2
   0x40143a92 <+6>:     movi.n  a8, 0
   0x40143a94 <+8>:     addi    a3, a3, 20
   0x40143a97 <+11>:    memw
   0x40143a9a <+14>:    s32i    a8, a9, 88
   0x40143a9d <+17>:    addx4   a2, a3, a2
   0x40143aa0 <+20>:    memw
   0x40143aa3 <+23>:    l32i.n  a8, a2, 8
   0x40143aa5 <+25>:    l32r    a3, 0x40143a84
   0x40143aa8 <+28>:    extui   a4, a4, 0, 8
   0x40143aab <+31>:    and     a8, a8, a3
   0x40143aae <+34>:    l32r    a3, 0x40123bd4
   0x40143ab1 <+37>:    or      a3, a8, a3
   0x40143ab4 <+40>:    memw
   0x40143ab7 <+43>:    s32i.n  a3, a2, 8
   0x40143ab9 <+45>:    memw
   0x40143abc <+48>:    l32i.n  a8, a2, 8
   0x40143abe <+50>:    movi    a3, 0x100
   0x40143ac1 <+53>:    or      a3, a8, a3
   0x40143ac4 <+56>:    memw
   0x40143ac7 <+59>:    s32i.n  a3, a2, 8
   0x40143ac9 <+61>:    memw
   0x40143acc <+64>:    l32i.n  a8, a2, 8
   0x40143ace <+66>:    l32r    a3, 0x40143a88
   0x40143ad1 <+69>:    or      a3, a8, a3
   0x40143ad4 <+72>:    memw
   0x40143ad7 <+75>:    s32i.n  a3, a2, 8
   0x40143ad9 <+77>:    memw
   0x40143adc <+80>:    s8i     a4, a9, 88                              ; BOOM!
   0x40143adf <+83>:    memw
   0x40143ae2 <+86>:    l32i.n  a8, a2, 8
   0x40143ae4 <+88>:    movi    a3, 0xfffffbff
   0x40143ae7 <+91>:    and     a3, a8, a3
   0x40143aea <+94>:    memw
   0x40143aed <+97>:    s32i.n  a3, a2, 8
   0x40143aef <+99>:    retw.n
End of assembler dump.
```

rojer, 2017-05-11:
here's the result of applying some perl-fu to `*_struct.h` under components/soc - it shows cases where a byte-size field comes first in a struct:
```
[rojer@nba ~/cesanta/esp/esp-idf/components/soc/esp32/include/soc master]$ perl -ne 'print "$ARGV:$. $_" if (/\s8\s*;/ && $ps); $ps = /struct/; close ARGV if eof;' *struct.h
apb_ctrl_struct.h:30             volatile uint32_t xtal_tick:     8;
apb_ctrl_struct.h:37             volatile uint32_t pll_tick:     8;
apb_ctrl_struct.h:44             volatile uint32_t ck8m_tick:     8;
apb_ctrl_struct.h:80             volatile uint32_t rstb_wait:    8;
apb_ctrl_struct.h:97             volatile uint32_t apll_tick:     8;
gpio_sd_struct.h:19             uint32_t duty:        8;
gpio_struct.h:23             uint32_t data:       8;                 /*GPIO32~39 output value*/
gpio_struct.h:30             uint32_t data:       8;                 /*GPIO32~39 output value write 1 to set*/
gpio_struct.h:37             uint32_t data:       8;                 /*GPIO32~39 output value write 1 to clear*/
gpio_struct.h:44             uint32_t sel:        8;                 /*SDIO PADS on/off control from outside*/
gpio_struct.h:54             uint32_t data:        8;                /*GPIO32~39 output enable*/
gpio_struct.h:61             uint32_t data:        8;                /*GPIO32~39 output enable write 1 to set*/
gpio_struct.h:68             uint32_t data:        8;                /*GPIO32~39 output enable write 1 to clear*/
gpio_struct.h:83             uint32_t data:       8;                 /*GPIO32~39 input value*/
gpio_struct.h:93             uint32_t intr_st:    8;                 /*GPIO32~39 interrupt status*/
gpio_struct.h:100             uint32_t intr_st:    8;                 /*GPIO32~39 interrupt status write 1 to set*/
gpio_struct.h:107             uint32_t intr_st:    8;                 /*GPIO32~39 interrupt status write 1 to clear*/
gpio_struct.h:120             uint32_t intr:       8;                 /*GPIO32~39 APP CPU interrupt status*/
gpio_struct.h:127             uint32_t intr:       8;                 /*GPIO32~39 APP CPU non-maskable interrupt status*/
gpio_struct.h:134             uint32_t intr:       8;                 /*GPIO32~39 PRO CPU interrupt status*/
gpio_struct.h:141             uint32_t intr:       8;                 /*GPIO32~39 PRO CPU non-maskable interrupt status*/
gpio_struct.h:148             uint32_t intr:       8;                 /*SDIO's extent GPIO32~39 interrupt*/
i2c_struct.h:249             uint32_t byte_num:      8;              /*Byte_num represent the number of data need to be send or data need to be received.*/
i2s_struct.h:261             uint32_t fifo_timeout:       8;
i2s_struct.h:308             uint32_t bad_cef_atten_para:       8;
i2s_struct.h:387             uint32_t clkm_div_num: 8;
rmt_struct.h:21                 uint32_t div_cnt:        8;             /*This register is used to configure the  frequency divider's factor in channel0-7.*/
rtc_cntl_struct.h:131             uint32_t ulp_cp_subtimer_prediv: 8;
sdmmc_struct.h:102             uint32_t div0: 8;   ///< 0: bypass, 1-255: divide clock by (2*div0).
sdmmc_struct.h:129             uint32_t response: 8;       ///< response timeout, in card output clock cycles
spi_struct.h:138             uint32_t usr_dummy_cyclelen: 8;                 /*The length in spi_clk cycles of dummy phase. The register value shall be (cycle_num-1).*/
spi_struct.h:226             uint32_t rdsta_dummy_cyclelen: 8;               /*In the slave mode it is the length in spi_clk cycles of dummy phase for read-status operations. The register value shall be (cycle_num-1).*/
spi_struct.h:235             uint32_t rdbuf_cmd_value: 8;                    /*In the slave mode it is the value of read-buffer command.*/
syscon_struct.h:30             volatile uint32_t xtal_tick:     8;
syscon_struct.h:37             volatile uint32_t pll_tick:     8;
syscon_struct.h:44             volatile uint32_t ck8m_tick:     8;
syscon_struct.h:80             volatile uint32_t rstb_wait:    8;
syscon_struct.h:97             volatile uint32_t apll_tick:     8;
uart_struct.h:143             uint32_t rxfifo_cnt: 8;                 /*(rx_mem_cnt rxfifo_cnt) stores the byte number of valid data in receiver's fifo. rx_mem_cnt register stores the 3 most significant bits  rxfifo_cnt stores the 8 least significant bits.*/
uart_struct.h:245             uint32_t xon_threshold:  8;             /*when the data amount in receiver's fifo is more than this register value， it will send a xoff char with uart_sw_flow_con_en set to 1.*/
uart_struct.h:297             uint32_t data:        8;                /*This register is used to configure the content of at_cmd char.*/
uhci_struct.h:251             uint32_t txfifo_timeout:       8;              /*This register stores the timeout value.when DMA takes more time than this register value to receive a data  it will produce  uhci_tx_hung_int interrupt.*/
uhci_struct.h:278             uint32_t seper_char:      8;                   /*This register stores the separator char  separator char is used to separate the data frame.*/
uhci_struct.h:287             uint32_t seq0:       8;                        /*This register stores the first substitute char used to replace the separate char.*/
uhci_struct.h:296             uint32_t seq1:       8;                        /*This register stores the flow control char to turn on the flow_control*/
uhci_struct.h:305             uint32_t seq2:       8;                        /*This register stores the flow_control char to turn off the flow_control*/
```

projectgus, 2017-05-11:
Thanks for this @rojer. There's something very unusual going on here!

* This bug doesn't happen at Debug optimatisation (-Og), only Release (-O2).

* Very simple code like `I2C0.command[x].byte_num = value` doesn't trigger it.

* The bug doesn't happen if you remove the write to 'val' at the top of the function:
```
void __attribute__ ((noinline)) bugmemore(i2c_dev_t *dev, int ci, uint16_t bn) {
  //dev->command[ci].val = 0;
  dev->command[ci].op_code = 1;
  dev->command[ci].ack_en = true;
  dev->command[ci].done = 1;
  dev->command[ci].byte_num = bn;  // doesn't boom
  dev->command[ci].ack_val = 0;
}
```

I don't 100% understand what's happening here. Writing the 'val' part of the union to zero seems to trigger the bug because gcc decides it now "knows" the values of the fields in the struct on the other side of the union, and can optimise accordingly. This seems like it may be near the root of the bug, because the entire union is part of a "volatile" struct so (AFAIK) gcc should never optimise on that assumption and should always generate a read-modify-write for each field. Possibly we need to sprinkle "volatile" somewhere else in the struct register definitions, or possibly it's a toolchain bug.

* The bug doesn't happen if i2c_dev_t isn't a pointer:
void __attribute__ ((noinline)) bugmemore(i2c_dev_t *dev, int ci, uint16_t bn) {
```
  dev = &I2C0;
  dev->command[ci].val = 0;
  dev->command[ci].op_code = 1;
  dev->command[ci].ack_en = true;
  dev->command[ci].done = 1;
  dev->command[ci].byte_num = bn;  // no more boom
  dev->command[ci].ack_val = 0;
}
```
... I have no idea why that is.

* While we're looking into this, something you can do is build the struct on the stack and then copy it to the register. This is actually a higher-performance option than writing each element directly, as gcc generates fewer read-write-modify cycles.
```
void __attribute__ ((noinline)) bugmemore(i2c_dev_t *dev, int ci, uint16_t bn) {
    const typeof(*dev->command) command = {
        .val = 0,
        .op_code = 1,
        .ack_en = true,
        .done = 1,
        .byte_num = bn,
        .ack_val = 0,
    };
    dev->command[ci].val = command.val;
}
```
It feels a bit odd to have to use the typeof() gcc-ism for this, maybe we should add explicit struct type names to the struct registers. Also, the C99-style struct initializer is also for performance as the type is still volatile, so will otherwise still issue read-modify-writes of the stack variable for each field write.

projectgus, 2017-05-11:
> Possibly we need to sprinkle "volatile" somewhere else in the struct register definitions, or possibly it's a toolchain bug.

Explicitly adding absurdly redundant "volatile"s everywhere in the type (which shouldn't change the type at all anyhow, AFAIK) doesn't change the compiler output:
```
  volatile union {
       volatile struct {
            volatile uint32_t byte_num:      8;              /*Byte_num represent the number of data need to be send or data need to be received.*/
            volatile uint32_t ack_en:        1;              /*ack_check_en  ack_exp and ack value are used to control  the ack bit.*/
            volatile uint32_t ack_exp:       1;              /*ack_check_en  ack_exp and ack value are used to control  the ack bit.*/
            volatile uint32_t ack_val:       1;              /*ack_check_en  ack_exp and ack value are used to control  the ack bit.*/
            volatile uint32_t op_code:       3;              /*op_code is the command  0：RSTART   1：WRITE  2：READ  3：STOP . 4:END.*/
            volatile uint32_t reserved14:   17;
            volatile uint32_t done:  1;                      /*When command0 is done in I2C Master mode  this bit changes to high level.*/
        };
        volatile uint32_t val;
    } volatile command[16];
```

So unless there's some intricacy of the C standard I don't understand (or I need my morning coffee before understanding), this is a gcc bug when writing to both sides of a volatile union.

projectgus, 2017-05-12:
Minimal test case:
```
typedef volatile struct {
  union {
	struct {
	  uint32_t byte2 : 8; // There can be other fields before this, it just needs to be type-aligned
	};
	uint32_t word;
  } a_reg[1];
} test_t;

void set_fields(test_t *test, int i)
{
  test->a_reg[0].word =0;
  test->a_reg[0].byte2=3;
}
```

Compiling with `-O2 -fstrict-volatile-bitfields` creates s8i on our toolchain, and a movb on amd64 w/ gcc 7.

Looks very similar to this bug: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=50521

Which was in my browser history already, so clearly I have bad memory for bugs I've already seen...

igrr, 2017-05-12:
Byte-wise access happens to work in some cases (SPI and UART FIFOs, for example), but as a general rule, peripheral registers only support 32-bit access. Behaviour for byte-wise access depends on the exact peripheral and buses, and may include generating a bus fault (Exception 3), value truncation, or some other undefined behavior.

In this case, we assumed that `-fstrict-volatile-bitfields` is enough to instruct the compiler to use the type of the field when choosing the write mode. There was another instance when this was happening in I2C driver, workaround was added in this commit: https://github.com/espressif/esp-idf/commit/94b91ee4af30f741ec8001331c358d2f596ed4ce.

projectgus, 2017-05-12:
> does something in the standard say that access in this case has to be by word?

C standard doesn't say anything about this. However, [gcc has -fstrict-volatile-bitfields](https://gcc.gnu.org/onlinedocs/gcc-5.2.0/gcc/Code-Gen-Options.html) which (should)[^] produce word accesses here. The default for this setting varies per-architecture, as some architectures (like ARM) explicitly say in their ABI that this kind of bitfield access is required so gcc enables it by default. For Xtensa it's off by default but we enable it in our compiler flags. (This isn't relevant to the bug, which seems to be non-architecture-specific, just gcc related trivia!)

There are a few -fstrict-volatile-bitfields patches floating around - there are the ones at the URL above, and one in this patch series here - https://gcc.gnu.org/ml/gcc-patches/2013-09/msg02057.html . Some combination of these will probably fix this, and we can probably apply them to our toolchain. Need a bit more time to investigate, first.

For now I'd suggest using the workaround with the struct initializer. As mentioned, this will give better performance than piecemeal volatile writes - and it's still pretty readable IMO.

[^] The bug report I linked above has a maintainer saying at one point that a bitfield in a volatile union may not meet gcc's defintion of a volatile bitfield, but this still doesn't seem like a "by design" thing to me.

Alvin1Zhang, 2019-03-22:
@rojer Sorry that get back to you so late, has your issue been resolved? Thanks.

rojer, 2019-03-22:
@Alvin1Zhang the particular issue about I2C register access has been resolved, but this bug is about generic issue of 8-bit stores to 32-bit peripheral registers. as @projectgus mentions in his comment here and @igrr says in https://github.com/espressif/esp-idf/issues/2413#issuecomment-422631305, this needs more investigation/work.
perhaps as a short-term measure, 8-bit wide struct fields on the byte boundaries could be disabled - e.g. `byte_num` renamed to `byte_num_bug597`, etc.

Alvin1Zhang, 2019-04-08:
@rojer Thanks for the updates.

Dazza0, 2021-09-23:
Closed on master branch by 874a720286b78e866684129ce18a6c948c8fa7fe and 14fe6dcaaf6d1bb9a1d1bb1d0e51a8fd184101e7 (for C++ workaround)

closed 2021-09-23 by Dazza0