espressif/esp-idf #598: components/driver/spi_master: RX data not delivered in half-duplex mode

devsaurus, 2017-05-13:
I'm testing the half-duplex mode of the spi_master driver and stumbled upon an issue where RX data is not delivered to the user application.

#### Test setup ####
W25Q16 flash chip connected to the ESP32:
  - SCLK at 19
  - /CS at 22
  - MOSI at 23
  - MISO at 22

Application shall retrieve manufacturer and device IDs. This is done by sending command 0x90 and address 0x000000 (24 bits). The device will return 2 bytes.

#### Problem description ####
When using half-duplex mode to first send 4 bytes for command & address and subsequently receive 2 bytes, the received data is always 0x00, 0x00. Even though the LA trace below shows that the flash chip provides 0xef, 0x14 at the correct bit slots.

[image]

#### Example ####
The following example project exercises 3 transactions with identical MOSI and MISO patterns (checked with LA).
1. Command and address fields are used to generate 4 bytes on MOSI. 2 bytes on MISO are received in half-duplex mode.
  Result OK: 0xef 0x14 reported by the test.
2. Data buffer is used to generate 6 bytes on MOSI. 6 bytes on MISO are received in full-duplex mode.
  Result OK: 0xex 0x14 reported by the test.
3. Data buffer is used to generate 4 bytes on MOSI, 2 bytes on MISO are received in half-duplex mode.
  Result not OK: 0x00 0x00 reported by the test.

```c
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "driver/spi_master.h"
#include "soc/gpio_struct.h"
#include "driver/gpio.h"

#define PIN_NUM_MISO 25
#define PIN_NUM_MOSI 23
#define PIN_NUM_CLK  19
#define PIN_NUM_CS   22

void app_main()
{
  esp_err_t ret;
  spi_bus_config_t buscfg={
    .miso_io_num=PIN_NUM_MISO,
    .mosi_io_num=PIN_NUM_MOSI,
    .sclk_io_num=PIN_NUM_CLK,
    .quadwp_io_num=-1,
    .quadhd_io_num=-1
  };
  //Initialize the SPI bus
  ret = spi_bus_initialize(HSPI_HOST, &buscfg, 1);
  assert(ret == ESP_OK);

  while (1) {

    {
      // ********************************************************************
      // Test case 1:
      // Use command and address, receive in halfduplex mode.
      //
      spi_device_handle_t dev;
      spi_device_interface_config_t devcfg = {
        .clock_speed_hz = 1000000,
        .mode = 0,
        .spics_io_num = PIN_NUM_CS,
        .queue_size = 1,
        .command_bits = 8,
        .address_bits = 24,
        .flags = SPI_DEVICE_HALFDUPLEX
      };
      //
      ret = spi_bus_add_device(HSPI_HOST, &devcfg, &dev);
      assert(ret == ESP_OK);

      spi_transaction_t trans = {
        .flags = SPI_TRANS_USE_RXDATA,
        .command = 0x90,
        .address = 0x000000,
        .length = 0,
        .rxlength = 2 * 8
      };
      //
      ret = spi_device_transmit(dev, &trans);
      assert(ret == ESP_OK);

      printf("Manufacturer ID: 0x%02x, Device ID: 0x%02x\n", trans.rx_data[0], trans.rx_data[1]);

      spi_bus_remove_device(dev);
      assert(ret == ESP_OK);
    }

    {
      // ********************************************************************
      // Test case 2:
      // Send MOSI with plain data, receive in fullduplex mode.
      //
      spi_device_handle_t dev;
      spi_device_interface_config_t devcfg = {
        .clock_speed_hz = 1000000,
        .mode = 0,
        .spics_io_num = PIN_NUM_CS,
        .queue_size = 1
      };
      //
      ret = spi_bus_add_device(HSPI_HOST, &devcfg, &dev);
      assert(ret == ESP_OK);

      uint8_t tx_data[] = {0x90, 0x00, 0x00, 0x00, 0x00, 0x00};
      uint8_t rx_data[6];
      spi_transaction_t trans = {
        .length = 6 * 8,
        .tx_buffer = tx_data,
        .rx_buffer = rx_data
      };
      //
      ret = spi_device_transmit(dev, &trans);
      assert(ret == ESP_OK);

      printf("Manufacturer ID: 0x%02x, Device ID: 0x%02x\n", rx_data[4], rx_data[5]);

      spi_bus_remove_device(dev);
      assert(ret == ESP_OK);
    }

    {
      // ********************************************************************
      // Test case 3:
      // Send MOSI with plain data, receive in halfduplex mode.
      //
      spi_device_handle_t dev;
      spi_device_interface_config_t devcfg = {
        .clock_speed_hz = 1000000,
        .mode = 0,
        .spics_io_num = PIN_NUM_CS,
        .queue_size = 1,
        .flags = SPI_DEVICE_HALFDUPLEX
      };
      //
      ret = spi_bus_add_device(HSPI_HOST, &devcfg, &dev);
      assert(ret == ESP_OK);

      spi_transaction_t trans = {
        .flags = SPI_TRANS_USE_RXDATA | SPI_TRANS_USE_TXDATA,
        .length = 4 * 8,
        .rxlength = 2 * 8,
        .tx_data = {0x90, 0x00, 0x00, 0x00}
      };
      //
      ret = spi_device_transmit(dev, &trans);
      assert(ret == ESP_OK);

      printf("Manufacturer ID: 0x%02x, Device ID: 0x%02x\n", trans.rx_data[0], trans.rx_data[1]);

      spi_bus_remove_device(dev);
      assert(ret == ESP_OK);
    }

    while (1);
  }
}
```

Spritetm, 2017-05-13:
Interesting. Will look into this after the weekend, thanks for the clear test case.

devsaurus, 2017-05-14:
Thanks for taking care, Jeroen!

If I remove the while at the end and let the 3 tests run in a loop, then every other iteration fails completely:
```
Manufacturer ID: 0xef, Device ID: 0x14     <--- iteration 1
Manufacturer ID: 0xef, Device ID: 0x14     <--- iteration 1
Manufacturer ID: 0x00, Device ID: 0x00     <--- iteration 1
Manufacturer ID: 0xef, Device ID: 0x00     <--- iteration 2
Manufacturer ID: 0xef, Device ID: 0x00     <--- iteration 2
Manufacturer ID: 0x00, Device ID: 0x00     <--- iteration 2
Manufacturer ID: 0xef, Device ID: 0x14     <--- iteration 3
Manufacturer ID: 0xef, Device ID: 0x14     <--- iteration 3
Manufacturer ID: 0x00, Device ID: 0x00     <--- iteration 3
Manufacturer ID: 0xef, Device ID: 0x00     <--- iteration 4
Manufacturer ID: 0xef, Device ID: 0x00     <--- iteration 4
Manufacturer ID: 0x00, Device ID: 0x00     <--- iteration 4
Manufacturer ID: 0xef, Device ID: 0x14     <--- iteration 5
Manufacturer ID: 0xef, Device ID: 0x14     <--- iteration 5
Manufacturer ID: 0x00, Device ID: 0x00     <--- iteration 5
Manufacturer ID: 0xef, Device ID: 0x00     <--- iteration 6
Manufacturer ID: 0xef, Device ID: 0x00     <--- iteration 6
Manufacturer ID: 0x00, Device ID: 0x00     <--- iteration 6
```

rsnbusa, 2017-05-15:
I confirm the same problem in half duplex. Logic analyzer does return the correct values but the data is not received in the buffer. 

MB85RS512T Fram chip. 

Logic Analyzer capture is perfect. Expected four 0x63 bytes as seen.
[image]

The data in the routine/code is incorrect. Received two 0x63 and two 0 bytes. Its random at best.

[image]

BTW, old library (without spi-slave,etc) does not have this error. Code was not changed when doing this test, except for the printf line.

[image]

Regards.

Robert

rsnbusa, 2017-05-23:
Hi Spritetm.
Is there a date to look forward to for an update. Im stuck on the previous esp-idf version cause I need the SPI Master but I need the wear-leveling filing system and it only works in version 2.0-2.1.

br

kazkojima, 2017-05-24:
I have a similar problem with the spi master after merging spi-slave support.
My app has multiple spi sensors and only one chip select is asserted with the recent version of idf.  Even for the device of which /cs is correctly asserted, the RX data is broken somewhere.
FYI, all works well again for me if the DMA is disabled when initializing spi:

    -    ret=spi_bus_initialize(VSPI_HOST, &buscfg, 1);
    +    ret=spi_bus_initialize(VSPI_HOST, &buscfg, 0);

Regards,
kaz

Spritetm, 2017-05-24:
Working on it right now... went down the rabbit hole a bit, though, running into some other kind of bug. Hope to have this figured out and fixed in a few days at max.

rsnbusa, 2017-05-24:
Thanks. Ill try disabling DMA and report

rsnbusa, 2017-05-24:
Actually it works with dma disabled. Performance would be another issue, but thanks.

br

rsnbusa, 2017-05-24:
Also, must not send more than 32 bytes in a transfer, read or write.

joicetm, 2017-06-27:
Hi spritetm, 
Any updates on the issue ? are you currently working on this.

ginkgm, 2017-08-03:
I'm afraid there's something wrong and cannot be fixed in a short time.
The issue occurs when DMA is enabled in half-duplex mode and use MISO phase after a MOSI phase.
To quickly get around this issue, you may try one of the following method:
1. use full-duplex mode instead
1. disable the DMA by set the last parameter to 0 in bus initialization function just as belows: `ret=spi_bus_initialize(VSPI_HOST, &buscfg, 0);`
this may forbid you from transmitting and receiving data longer than 32 bytes.
1. try to use command and address field to replace the MOSI phase. They should be used at the same time, so the TX length is restricted to 1+1 bytes at least and 2+8 bytes at most.

negativekelvin, 2017-08-03:
@ginkgm so the fact that this was reported working in earlier versions was just coincidence due to the fact DMA was not enabled for small transfers ?

ginkgm, 2017-08-07:
@negativekelvin no, it was not a coincidence. As you may have seen, the code in IDF v2.0 always use memcpy rather than DMA to get data shorter than 32 bytes, though DMA channel is assigned in the initialization process. That's the difference between IDF 2.1 and earlier version.

We're going to add some warnings and asserts to help users get around the issue temporarily.

johanvdp, 2017-08-20:
I struggled some time trying countless variants of device and transaction settings, but ended up googling, and finding this issue. I think it describes my problem.
I can not find any way to read more than 32 bytes from sram, I am using a command and address fragment  as suggested above, but no useful data is transferred. My logic analyser does show the correct data to be on the bus, but the data does not end up in the (DMA) buffers.
Not being able to read SPI using DMA looks like a serious bug.

ginkgm, 2017-08-22:
@johanvdp  
Some fixs for RX DMA is uploaded to the master branch (commit ed1d084aeae and before). Though the RX DMA in half-duplex mode with both MOSI and MISO phase issue will not be solved very soon, the update in IDF already helps some users to solve their problem.

I hope you can have a try and come back if the problem still exists.

johanvdp, 2017-08-25:
@ginkgm I am glad to report the SPI read using DMA now works in my case (I am using both command and address fields, and transfer data in one direction only. I did not test SDI or SQI transfers but assume that these do not work as these transfers are half-duplex by nature.

ginkgm, 2017-08-28:
glad to help you :)

lakshmidhanaraj, 2017-10-28:
@johanvdp @ginkgm   Please share the spi master and slave code for SPI half duplex using DMA.I need to transfer more than 32 bytes.

johanvdp, 2017-11-01:
The project is not complete, but this part works
https://github.com/johanvdp/net-radio/blob/master/components/mem/spi_mem.c

closed 2018-02-12 by igrr

tve, 2020-06-25:
FYI, I'm seeing problems that seem to be related to this in ESP-IDF v4.0.1 with multiple SPI devices. Specifically, I have one device where I use full-duplex transactions and one where I use half duplex, both on the same SPI bus (hspi or vspi, makes no difference). If I only use the full-duplex device it works great, the moment I perform one transaction on the half duplex device the full-duplex one ceases to receive the results of its transactions (read fails). I see the correct signals in a logic analyzer but the rx buffer is messed-up. Sometimes the first 4 bytes appear, sometimes a prefix is zeroed and then the first 4 bytes, but never do I get the correct data.
Unfortunately I don't have simple test case to publish due to using MicroPython and special C bindings.

zengdapang, 2024-05-23:
esp32c6 全双工仅支持 1bit , QIO使用半双工 总失败...