espressif/esp-idf #599: <sys/time.h> implicit declaration of function 'timeradd'

Beck-Sisyphus, 2017-05-14:
Hi espressif,

In recent development on system time, I used Linux similar command for struct timeval, but get the following error.

> error: implicit declaration of function 'timeradd' [-Werror=implicit-function-declaration]
     timeradd(t_start, timeout, t_timeout);

My program looks like:

`#include <sys/time.h>

    struct timeval t_start;
    gettimeofday(&t_start, NULL);
    struct timeval timeout = { .tv_sec = 30, .tv_usec = 0,};
    struct timeval t_timeout;
    timeradd(t_start, timeout, t_timeout);`

I have found the library at esp-idf/components/newlib/include/sys/time.h, and timeval works fine. 

Thanks,

Beck

Beck-Sisyphus, 2017-05-14:
I guess it simply proves that this function is not POSIX complaint. Rewrite my own functions.

closed 2017-05-14 by Beck-Sisyphus

Beck-Sisyphus, 2017-05-14:
void * memset ( void * ptr, int value, size_t num ); in #include <stdio.h> also give me
 implicit declaration of function

igrr, 2017-05-14:
memset should be in string.h, not stdio.